espressif/esptool #502: Catalina and "Timed out waiting for packet header"

GermanSheepDog, 2020-01-06:
Hello,
I used M5Stack and M5StickC at my Mac with OS 10.14. Now I changed to Catalina (10.15) and I installed the current FTDI driver and the current CP2104 (the FTDI driver is for M5StickC, CP2104 driver is for M5Stack Core).

The FTDI driver didn't work. If I flash the ESP32 I got: 
'''
A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header.
'''
My workaround is "Connect G0 pin to GND". 

The CP2104 at M5Stack Core often worked, but sometimes I got the same error and I then I connect G0 pin to GND.

My short test showed the problem:
$IDF_PATH/components/esptool_py/esptool/esptool.py chip_id
-> Here I got the "Timed out waiting for packet header".

Can anyone help me to program without connecting G0 pin to GND?

klebermagno, 2020-01-13:
Same problem here.

GermanSheepDog, 2020-01-16:
Does anyone else have this problem too?

rudiniemeijer, 2020-01-17:
There is a known timing problem with macOS X Catalina and esptool.py. Espressif knows about it. The workaround entails connecting pin G0 to ground while using esptool.py.

See [this article](http://www.rudiniemeijer.nl/workaround-voor-probleem-met-macos-10-15-niet-compatibel-met-esptool-py-en-m5stickc) (Dutch) and the discussion [here](http://community.m5stack.com/topic/1369/m5stickc-usb-driver-and-catalina).

GermanSheepDog, 2020-01-17:
Hello @rudiniemeijer,
thank you gor the Information. I didn‘t understand dutch, but I know the article: I am the first author. Now I am waiting for a software solution.

klebermagno, 2020-01-27:
For M5Stickc Ok
But I need use in esp32 TF Module in my Sipeed Maix Bit kit.

projectgus, 2020-02-21:
Hi @rudiniemeijer @GermanSheepDog ,

Sorry for the slow reply on this one. This seems to an issue between the CH552 programmable USB microcontroller used by M5Stack for the serial interface, and the relevant driver in Mac OS X Catalina. The [serial control signals used by esptool.py for auto-reset](https://github.com/espressif/esptool/wiki/ESP32-Boot-Mode-Selection#automatic-bootloader) were not getting through to the hardware.

According to M5Stack the solution is to update the firmware on the CCH552 chip:
http://community.m5stack.com/topic/1591/m5stickc-and-atom-on-macos-platform-can-t-upload-programs-solustion

I don't think there's anything else we can do about this from esptool.py, it needs to be fixed at this lower layer.

hallard, 2020-02-23:
I confirm, I'm unable to upload anything on catalina ESP8266/ESP8285 (not tried on ESP32) but whatever chip I'm using CH340, FTDI, CP2102 so yes looks like underlying layer but working on 3 mac and all have the same issue and using win app in parallel does the same thing. It's really a pity I need to go on windows laptop to flash ;-(

nodofilm, 2020-05-22:
I too am experiencing this issue with the CP2102. Any updates?

kryptonn, 2020-05-26:
Same Here with ESP32 PICO, using official and even git head version of esptool in command line.
I got the error even with Arduino IDE and platform.io
With my older mac and older OS (Mojave 10.14.6) it works fine.
Any news ?

GermanSheepDog, 2020-05-27:
@kryptonn, @nodofilm : Did you read this [https://forum.m5stack.com/topic/1591/m5stickc-and-atom-on-macos-catalina-can-t-upload-programs-solution](url)

nodofilm, 2020-05-27:
The solution on our end was not Catalina actually. It was a Wacom tablet driver. They also use an SiLabs UART to Serial chip and they modified the SiLabs driver. It looks like "SiLabsUSBDriver64.kext" and must be removed.

kryptonn, 2020-05-29:
> @kryptonn, @nodofilm : Did you read this [https://forum.m5stack.com/topic/1591/m5stickc-and-atom-on-macos-catalina-can-t-upload-programs-solution](url). Important: You must log in on M5Stack-Website to see the information.
@GermanSheepDog As it is said it's only for M5stick and my cause dommage, so is it applicable for an ESP32 PICO KIT ?

projectgus, 2020-08-28:
Closing as the CH552 issue with M5Stack seems to be resolved.

@kryptonn Sorry for not noticing your other replies here. The ESP-PICO-KIT uses a CP2102N USB/serial chip which is made by a different manufacturer and uses a different driver. Suggest checking that you have the latest [macOS CP2102 driver from Silicon Labs installed](https://www.silabs.com/products/development-tools/software/usb-to-uart-bridge-vcp-drivers) and no errors visible in System Preferences relating to driver signing, etc. If it's still not working then please open a new issue with the details.

closed 2020-08-28 by projectgus

edwinm, 2022-12-03:
> The solution on our end was not Catalina actually. It was a Wacom tablet driver. They also use an SiLabs UART to Serial chip and they modified the SiLabs driver. It looks like "SiLabsUSBDriver64.kext" and must be removed.

This should be illegal! 😱