espressif/esptool #503: Arduino IDE - "Timed out waiting for packet header" SPIFFS Upload failed!

BarryO2905, 2020-01-08:
----------------------------- Delete below -----------------------------

# Operating system
Windows 10 Pro
# Python version

Can run `python -V` to check this.

# What Chip

ESP8266-01 module

# What development board or other hardware is the chip attached to

Using Arduino IDE to code

# Is anything else attached to the development board, except for the serial flasher connections?

nothing yet just trying to upload Sketch Data, SPIFF

# Are you running esptool.py from an IDE such as Arduino or Eclipse?
ARDUINO esptool.py V2.8

# Full esptool.py command line that was run:

not sure of this 

# Full output from esptool.py

[SPIFFS] data    : C:\Users\Barry\Documents\Arduino\ESP_Chart_Web_Server\data
[SPIFFS] size    : 64
[SPIFFS] page    : 256
[SPIFFS] block   : 4096
/index.html

[SPIFFS] upload  : C:\Users\Barry\AppData\Local\Temp\arduino_build_516271/ESP_Chart_Web_Server.spiffs.bin
[SPIFFS] address  : 0xEB000
[SPIFFS] reset    : --before default_reset --after hard_reset
[SPIFFS] port     : COM6
[SPIFFS] speed    : 115200
[SPIFFS] python   : C:\Users\Barry\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1\python3.exe
[SPIFFS] uploader : C:\Users\Barry\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\tools\upload.py

esptool.py v2.8

Serial port COM6

Connecting.....

Chip is ESP8266EX

Features: WiFi

Crystal is 26MHz

MAC: bc:dd:c2:24:9a:e7

Uploading stub...

Running stub...

Stub running...

Configuring flash size...

Auto-detected Flash size: 1MB

Compressed 65536 bytes to 1749...

Writing at 0x000eb000... (100 %)
Wrote 65536 bytes (1749 compressed) at 0x000eb000 in 0.2 seconds (effective 2979.1 kbit/s)...

Traceback (most recent call last):

  File "C:\Users\Barry\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\tools\upload.py", line 65, in <module>

    esptool.main(cmdline)

  File "C:/Users/Barry/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/esptool\esptool.py", line 2938, in main

    operation_func(esp, args)

  File "C:/Users/Barry/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/esptool\esptool.py", line 2398, in write_flash

    res = esp.flash_md5sum(address, uncsize)

  File "C:/Users/Barry/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/esptool\esptool.py", line 104, in inner

    return func(*args, **kwargs)

  File "C:/Users/Barry/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/esptool\esptool.py", line 691, in flash_md5sum

    timeout=timeout)

  File "C:/Users/Barry/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/esptool\esptool.py", line 369, in check_command

    val, data = self.command(op, data, chk, timeout=timeout)

  File "C:/Users/Barry/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/esptool\esptool.py", line 347, in command

    p = self.read()

  File "C:/Users/Barry/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/esptool\esptool.py", line 292, in read

    return next(self._slip_reader)

  File "C:/Users/Barry/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/esptool\esptool.py", line 2045, in slip_reader

    raise FatalError("Timed out waiting for packet %s" % waiting_for)

esptool.FatalError: Timed out waiting for packet header

SPIFFS Upload failed!

# Do you have any other information from investigating this?

iv tried the connections alll seem ok,  connection for programming in pic.
[image]

# Is there any other information you can think of which will help us reproduce this problem?
i tried several times and keeps failing at same place, 

File "C:/Users/Barry/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/esptool\esptool.py", line 2045, in slip_reader

    raise FatalError("Timed out waiting for packet %s" % waiting_for)

ive noticed that its .............../2.6.3/tools/esptool.py

does this make a difference

projectgus, 2020-02-10:
Hi @BarryO2905 ,

Sorry for the slow reply. Are you sure your module has a stable enough 3.3V power source, with enough capacitance close to the ESP8266 chip? Power instability can cause this kind of failure midway through flashing.

BarryO2905, 2020-02-11:
Hi @projectgus 
Thanks for getting back to me, I’m flashing though Arduino IDE and using Arduino uno connected directly so I have no caps in the set up, I’ll give it a go again soon ( working on other parts of project) and keep an eye on the 3.3 v from Arduino... what do you suggest for capacitor values , type and placement,  

Regards
Barry

projectgus, 2020-02-11:
> I’m flashing though Arduino IDE and using Arduino uno connected directly so I have no caps in the set up,

Ah, this is useful information that was not in the original report. So you're using the Arduino Uno as a serial interface?

The onboard 3.3V regulator of an Arduino Uno is not powerful enough to power the ESP8266, you'll probably need an external 3.3V regular.

BarryO2905, 2020-02-19:
Thank you, I have one ordered and I’ll give it a go over the weekend, thanks for your help and advice

AKTanara, 2020-07-19:
Hi,
I have EXACTLY the same problem.
Same HW, same esptool.py v2.8 &...
Have you found any solutions?

radimkarnis, 2020-12-09:
Hello @BarryO2905,

Have you been able to flash the ESP8266 while powering it with an external 3.3V regulator?

BarryO2905, 2020-12-10:
Hi Radim Karniš

Yes I managed to flash it with an external voltage regulator, i used an
LM367, 3.3V regulator .

regards
barry

radimkarnis, 2020-12-11:
I'm glad to hear that! Thank you for the update. I'm therefore closing this issue, feel free to reopen if any further discussion is needed.

Radim

closed 2020-12-11 by radimkarnis