espressif/esptool #505: ESP32-WROVER-B  Flash Read Err

aliraza1944, 2020-01-21:
# Operating system
Win 10
# Python version
2.7.15

# What Chip
ESP32-WROVER-B

# What development board or other hardware is the chip attached to
Custom developed PCB. The design is based upon the Ethernet kit 1.1 reference design.

If your hardware is custom or unusual, please attach a photo to the issue.
https://imgur.com/KpE0OH4

# Is anything else attached to the development board, except for the serial flasher connections?
Yes, but pins 6 -11 are unconnected. MTDI is unconnected. As per the datasheet it shows that internally its NC too due to the 3.3V flash component.

# Are you running esptool.py from an IDE such as Arduino or Eclipse?
Using command prompt

# What else I have observed.
Probably due to the continuous restarts gpio5 pin is pulled down if I manually pull it up with a wire it goes into flash mode but then I get this.
```
 make -j12 flash monitor ESPPORT=COM5 ESPBAUD=115200
Toolchain path: /opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a5
Compiler version: 5.2.0
App "lanecontrol" version: 1.3.0
Python requirements from C:/esp/esp-idf/requirements.txt are satisfied.

Flashing binaries to serial port COM5 (app at offset 0x10000)...
esptool.py v2.8-dev
Serial port COM5
Connecting....
Chip is ESP32D0WDQ5 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: c8:2b:96:88:c6:f8
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Compressed 24656 bytes to 14754...
Wrote 24656 bytes (14754 compressed) at 0x00001000 in 1.3 seconds (effective 151.0 kbit/s)...

A fatal error occurred: Timed out waiting for packet header!
make: *** [/c/esp/esp-idf/components/esptool_py/Makefile.projbuild:67: flash] Error 2
```

I am also able to erase the flash using "make erase_flash".

# What have I tried
- Different baud rates.
- Manually pulling down MTDI just to be sure.
- Made sure via oscilloscope that GPIO0 is not being effected by the clock output from the PHY chip crystal

# Efuse Summary output

```
$ ./espefuse.py --port COM5 summary
espefuse.py v2.8-dev
Connecting........_____.
EFUSE_NAME             Description = [Meaningful Value] [Readable/Writeable] (Hex Value)
----------------------------------------------------------------------------------------
Security fuses:
FLASH_CRYPT_CNT        Flash encryption mode counter                     = 0 R/W (0x0)
FLASH_CRYPT_CONFIG     Flash encryption config (key tweak bits)          = 0 R/W (0x0)
CONSOLE_DEBUG_DISABLE  Disable ROM BASIC interpreter fallback            = 1 R/W (0x1)
ABS_DONE_0             secure boot enabled for bootloader                = 0 R/W (0x0)
ABS_DONE_1             secure boot abstract 1 locked                     = 0 R/W (0x0)
JTAG_DISABLE           Disable JTAG                                      = 0 R/W (0x0)
DISABLE_DL_ENCRYPT     Disable flash encryption in UART bootloader       = 0 R/W (0x0)
DISABLE_DL_DECRYPT     Disable flash decryption in UART bootloader       = 0 R/W (0x0)
DISABLE_DL_CACHE       Disable flash cache in UART bootloader            = 0 R/W (0x0)
BLK1                   Flash encryption key
  = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W
BLK2                   Secure boot key
  = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W
BLK3                   Variable Block 3
  = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W

Efuse fuses:
WR_DIS                 Efuse write disable mask                          = 0 R/W (0x0)
RD_DIS                 Efuse read disablemask                            = 0 R/W (0x0)
CODING_SCHEME          Efuse variable block length scheme                = 0 R/W (0x0)
KEY_STATUS             Usage of efuse block 3 (reserved)                 = 0 R/W (0x0)

Config fuses:
XPD_SDIO_FORCE         Ignore MTDI pin (GPIO12) for VDD_SDIO on reset    = 0 R/W (0x0)
XPD_SDIO_REG           If XPD_SDIO_FORCE, enable VDD_SDIO reg on reset   = 0 R/W (0x0)
XPD_SDIO_TIEH          If XPD_SDIO_FORCE & XPD_SDIO_REG, 1=3.3V 0=1.8V   = 0 R/W (0x0)
CLK8M_FREQ             8MHz clock freq override                          = 50 R/W (0x32)
SPI_PAD_CONFIG_CLK     Override SD_CLK pad (GPIO6/SPICLK)                = 0 R/W (0x0)
SPI_PAD_CONFIG_Q       Override SD_DATA_0 pad (GPIO7/SPIQ)               = 0 R/W (0x0)
SPI_PAD_CONFIG_D       Override SD_DATA_1 pad (GPIO8/SPID)               = 0 R/W (0x0)
SPI_PAD_CONFIG_HD      Override SD_DATA_2 pad (GPIO9/SPIHD)              = 0 R/W (0x0)
SPI_PAD_CONFIG_CS0     Override SD_CMD pad (GPIO11/SPICS0)               = 0 R/W (0x0)
DISABLE_SDIO_HOST      Disable SDIO host                                 = 0 R/W (0x0)

Identity fuses:
MAC                    Factory MAC Address
  = c8:2b:96:88:c6:f8 (CRC b1 OK) R/W
CHIP_VER_REV1          Silicon Revision 1                                = 1 R/W (0x1)
CHIP_VER_REV2          Silicon Revision 2                                = 0 R/W (0x0)
CHIP_VERSION           Reserved for future chip versions                 = 2 R/W (0x2)
CHIP_PACKAGE           Chip package identifier                           = 1 R/W (0x1)

Calibration fuses:
BLK3_PART_RESERVE      BLOCK3 partially served for ADC calibration data  = 0 R/W (0x0)
ADC_VREF               Voltage reference calibration                     = 1107 R/W (0x1)

Flash voltage (VDD_SDIO) determined by GPIO12 on reset (High for 1.8V, Low/NC for 3.3V).
```

# ESPEFUSE Dump

```
$ ./espefuse.py --port COM5 dump
espefuse.py v2.8-dev
Connecting........_____....._
EFUSE block 0:
00000000 9688c6f8 00b1c82b 0000a200 00000132 00000000 00000004
EFUSE block 1:
00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
EFUSE block 2:
00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
EFUSE block 3:
00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
```

# ESPTool Flash id output
```
$ ./esptool.py flash_id
esptool.py v2.8-dev
Found 1 serial ports
Serial port COM5
Connecting........_____....._____.
Detecting chip type... ESP32
Chip is ESP32D0WDQ5 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: c8:2b:96:88:c6:f8
Uploading stub...
Running stub...
Stub running...
Manufacturer: ff
Device: ffff
Detected flash size: Unknown
Hard resetting via RTS pin...
```

# Note:

In all the above cases, where I am able to get serial output from the board. I have to manually pull up GPIO5 using a wire.

aliraza1944, 2020-01-21:
@projectgus I have follow a lot of your responses on issues. Kindly take a look at this one too.

projectgus, 2020-01-21:
Hi @aliraza1944 ,

Thanks for the very detailed bug report. All of the esptool output suggests that the flash chip in the module is not responding to any requests.

> Warning: Could not auto-detect Flash size (FlashID=0x0, SizeID=0x0), defaulting to 4MB

> Manufacturer: ff
> Device: ffff
> Detected flash size: Unknown

Everything else looks normal and correct to me. 

You mention that nothing is connected to the pins of the module that are shared with flash. Can you double-check for shorts between any of the module pads, in particular IOs 6-11?

123swk123, 2020-01-21:
I had exactly same issue with custom board using 128mb flash and esp-wroom-32. Root cause was flash qspi pads had some shorts which occurred during soldering. So I had to desolder the module from the board removed all solders from the pad using wick and then resoldered very carefully then the shorts are gone and it worked like a charm.

aliraza1944, 2020-01-21:
@projectgus I checked the pins 6-11 for shorts between them and they all seem soldered fine. Also double checked EN pin and it is being pulled up @ 3.3V. 

I have noticed one strange thing and that is the MTDI pin is low at startup as it should be but when I press the Boot button it gets pulled up to 3.3V (probably the continuous resets due to the flash align with the boot button press and put the device in boot mode). If I do the same thing on the ethernet dev kit by espressif, MTDI remains low whether the device is in normal or boot mode.

I have access to two different boards of the same production batch and both are causing the same issue. Is there a chance that the SPI flash on board the module is actually fried.

Also the MTDO pin is being manually pulled up by me too but I guess that shouldn't effect as the default state for MTDO is pull up.

My next steps involve taking the metal jacket of ESP32-WRover-B module and inspecting the SPI and desoldering the module and replacing it with the ESP module from the ethernet devkit.

Another difference between the module on my custom PCB and ethernet devkit is that the custom PCB ESPWrover has a marking "XX0H64" whereas the ethernet devkit module has marking "XX0H32".
Let me know if there is anything else I can try.

@123swk123 I checked for shorts but everything seems fine.

aliraza1944, 2020-01-21:
Hi everyone, I was able to fix the issue. As expected it was a hardware issue. The ethernet devkit reference designs says to leave GPIO2 NC which I did. If I manually drive the pin low during the upload process the device works.

closed 2020-01-21 by aliraza1944