espressif/esptool #509: "Timed out waiting for packet header" - fails at any -b baudrate

gandsnut, 2020-01-29:
Trying to read firmware off Sonoff device via an FTDI USB-to-serial interface.  Lubuntu 18.04
Attempted to  use both 2.x and 3.x USB ports, and 2 cables.  I have tried the suggested variety of "-b" baud rates from documentation (115200, 921600, 230400, etc).

output from esptool.py:

gandsnut@haserver:/extra/sonoff_specific$ /usr/share/esptool/esptool.py -p /dev/ttyUSB0 -b 12000000 read_flash 0 0x100000 sonoff-original-firmware.bin
esptool.py v2.1
Connecting........_____....._____....._____....._____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header

hwinfo:

04: USB 00.0: 0700 Serial controller
  [Created at usb.122]
  Unique ID: UfPf.bzv2XOUTxN6
  Parent ID: 2XnU._63LoM0erv2
  SysFS ID: /devices/pci0000:00/0000:00:10.0/usb5/5-1/5-1:1.0
  SysFS BusID: 5-1:1.0
  Hardware Class: unknown
  Model: "Future Technology Devices International FT232 Serial (UART) IC"
  Hotplug: USB
  Vendor: usb 0x0403 "Future Technology Devices International, Ltd"
  Device: usb 0x6001 "FT232 Serial (UART) IC"
  Revision: "6.00"
  Serial ID: "A916MHVD"
  Driver: "ftdi_sio"
  Driver Modules: "usbserial", "ftdi_sio"
  Device File: /dev/ttyUSB0
  Device Files: /dev/ttyUSB0, /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_A916MHVD-if00-port0, /dev/serial/by-path/pci-0000:00:10.0-usb-0:1:1.0-port0
  Device Number: char 188:0
  Tags: mouse, modem
  Speed: 12 Mbps
  Module Alias: "usb:v0403p6001d0600dc00dsc00dp00icFFiscFFipFFin00"
  Driver Info #0:
    Driver Status: ftdi_sio is active
    Driver Activation Cmd: "modprobe ftdi_sio"
  Config Status: cfg=new, avail=yes, need=no, active=unknown
  Attached to: #17 (Hub)

Thanks
rjm@gandsnut.net

Twilight-Logic, 2020-01-30:
Similar issue here. I have a LoLin NodeMCU V3 ESP8266 board. The ESP8266 module itself is  marked ESP8266MOD.

Running ESP8266 community library in Ardion IDE I get the following:

```
Executable segment sizes:
IROM   : 267496          - code in flash         (default or ICACHE_FLASH_ATTR) 
IRAM   : 26900   / 32768 - code in IRAM          (ICACHE_RAM_ATTR, ISRs...) 
DATA   : 1256  )         - initialized variables (global, static) in RAM/HEAP 
RODATA : 17180 ) / 81920 - constants             (global, static) in RAM/HEAP 
BSS    : 33424 )         - zeroed variables      (global, static) in RAM/HEAP 
Sketch uses 312832 bytes (62%) of program storage space. Maximum is 499696 bytes.
Global variables use 51860 bytes (63%) of dynamic memory, leaving 30060 bytes for local variables. Maximum is 81920 bytes.
esptool.py v2.7
Serial port /dev/ttyUSB2
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: cc:50:e3:55:db:52
Uploading stub...
Running stub...
Stub running...
Manufacturer: 20
Device: 4016
Detected flash size: 4MB
Hard resetting via RTS pin...
esptool.py v2.7
Serial port /dev/ttyUSB2
Connecting........_____....._____....._____....._____....._____....._____....._____
Traceback (most recent call last):
  File "/home/johnc/.arduino15/packages/esp8266/hardware/esp8266/2.6.0/tools/upload.py", line 25, in <module>
    esptool.main(fakeargs)
  File "/home/johnc/.arduino15/packages/esp8266/hardware/esp8266/2.6.0/tools/esptool/esptool.py", line 2841, in main
    esp.connect(args.before)
  File "/home/johnc/.arduino15/packages/esp8266/hardware/esp8266/2.6.0/tools/esptool/esptool.py", line 483, in connect
    raise FatalError('Failed to connect to %s: %s' % (self.CHIP_NAME, last_error))
esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header
esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header
```

Library version 2.6.3 which comes with esptool.py version 2.8 has a similar problem with the same terminating error.

Reverting back to library 2.5.2 which comes with esptool.py version 2.6 works perfectly fine.

There seems to be some issue with esptool.py 2.7 and 2.8 that is preventing successful uploads.

Twilight-Logic, 2020-01-31:
Digging a little further I discovered that esptool.py version 2.7, 2.8 and 3.0 work fine when called from the command line with something like:

./esptool.py --port /dev/ttyUSB2 write_flash 0x000000 mybinary.bin

The problem arises only when esptool.py is called from the Arduino IDE with ESP8266 library versions above 2.5.2. I guess this means that the problem is not with esptool.py, but with the calling application, which in this case is either the Arduino IDE, or the ESP8266 library plugin.

gandsnut, 2020-01-31:
@Twilight-Logic :  Well, since I'm a __rank__noob, I have to ask:  ok, so what's the best way to deal with this?  Should I take this to an Arduino IDE forum, or one for the ESP8266 library (presumably here on github)?  And for now, once the IDE fails with the observed error, can I use your suggested command line - and if so, can you suggest where I'd find the .bin (in the IDE)?  Appreciated!

Twilight-Logic, 2020-01-31:
gandsnut, I have no connection with the esptool.py project, but from your comment I presume you also had the problem within the Arduino IDE?

I have opened a thread on the Arduino forum about this:
https://forum.arduino.cc/index.php?topic=662043.msg4459413#msg4459413

Please feel free to comment or add any further information to this. I have also opened an issue on the ESP8266 library GitHub:

https://github.com/esp8266/Arduino/issues/7050

I'm not sure if its an Arduino or a ESP8266 library issue, but hopefully by bringing attention to it in both places will increase the chances of someone having a look at it.

To export the binary from the Arduino IDE got to Sketch | Export compiled binary. The exported binary will be written to the same directory/folder as your sketch as a .bin file. With some boards you get two versions with and without bootloader, but with the ESP8266 you get just one .bin file. Easiest thing is to just simplify the name of the binary and place it in the same folder as the esptool.py program. You can dig around for the copy of esptool.py in the program arduino hierarchy (for my Linux machine it was in /home/user/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool) but it might be simpler to just pull the current version from the Github!

Twilight-Logic, 2020-02-01:
After some further investigation I may have found the root of the problem. Looking at the output in the Arduino IDE after the upload, it shows that the following command line is being used:

$ python upload.py --chip esp8266 --port /dev/ttyUSB0 --baud 115200 --before no_reset --after soft_reset write_flash 0x0 AR488-wifi.bin

Note that for clarity I have removed the full path to the python and tools directory in the esp8266 library hierarchy. Running with the above command line in terminal duplicates the error condition. Since the error output showed and unexpected reset at the start of the upload, I ran the command again but without the  '--before no_reset' parameter:

$ python upload.py --chip esp8266 --port /dev/ttyUSB0 --baud 115200 --before no_reset --after soft_reset write_flash 0x0 AR488-wifi.bin

This time the upload worked. It seems that this parameter is not used in the version 2.5.2 call which looks like this:

python upload.py --chip esp8266 --port /dev/ttyUSB0 --baud 115200 --trace version --end --chip esp8266 --port /dev/ttyUSB0 --baud 115200 --trace write_flash 0x0 /tmp/arduino_build_299222/AR488-wifi.ino.bin --end

Nor does this parameter appear in the version 2.5.2 boards.txt.py file as it does with the later version:

```
$ grep -n "no_reset" boards.txt.py
305:            ( '.menu.ResetMethod.v1.upload.resetmethod', '--before no_reset --after soft_reset' ),
1017:        ( '.menu.ResetMethod.ck.upload.resetmethod', '--before no_reset --after soft_reset' ),
1022:        ( '.menu.ResetMethod.nodtr_nosync.upload.resetmethod', '--before no_reset_no_sync --after soft_reset' ),
1028:        ( '.upload.resetmethod', '--before no_reset --after soft_reset' ),
1036:        ( '.upload.resetmethod', '--before no_reset --after soft_reset' ),
1044:        ( '.upload.resetmethod', '--before no_reset_no_sync --after soft_reset' ),

```
Perhaps this might provide a bit of a clue as to what is going on?

gandsnut, 2020-02-01:
Twilight-Logic:   Ok, just read your 2/1/2020 email containing:

"I ran the command again but without the '--before no_reset' parameter:"   which you say made the upload succeed.  Way cool.

Given this new info, is the (temporary fix) method to find in the Arduino IDE where it sets up each executed line of compilation, then edit out the "--before no_reset" usage?  I don't know if that's even possible... 

Appreciated (and thanks for the work-around suggestions).

Twilight-Logic, 2020-02-01:
Looks like something is getting out of sync during the update process. I have just followed a suggestion to delete the ~/.arduino15 directory (I renamed mine) and install from scratch. While this will loose your configured preferences and any additional installed boards, it doesn't take long to set up. At minimum you have to update the AVR library again and install the ESP8266 library. After that the ESP8266 upload worked. If you have any other categories of boards, e.g. Teensy, STM etc then there will likely have to be re-installed as well. Doesn't really explain why and what went wrong but at least it does provide a fix.

projectgus, 2020-02-10:
Hi @Twilight-Logic ,

It looks like your problem has been resolved by fixing the bad update of the arduino environment, yes? This is good to hear.

Hi @gandsnut,

Your problem looks different, although the error message is the same.

I would start with the basics: double-check all the connections, check you're correctly setting GPIO0 low to put the Sonoff into bootloader mode (and then re-powering it to reset it), and that TX & RX pins aren't accidentally swapped when connecting to the FTDI board.

Twilight-Logic, 2020-02-10:
> Hi @Twilight-Logic ,
> 
> It looks like your problem has been resolved by fixing the bad update of the arduino environment, yes? This is good to hear.

Yes, seems to be fixed for me. Thanks.

gandsnut, 2020-02-10:
@projectgus 

I've been diligent in wiring and using reset button to establish bootloader mode.  Many lay fault with cheap foreign FTDI boards;  apparently the '340' variety are more aggravating.  I have yet to try  @Twilight-Logic suggestion of reverting to 2.5.2 of the ESP8266 library, though I did try his other suggestion to rename the .arduino15 dir tree and reinstalling.  One wonders what repercussions come from that...  I did find esptool.py for both 8266 & 32, so I can try the other suggestion to invoke it directly.  Somewhere on the 'Net I saw a 'esptool.ck' - written in 'C'?  I don't know.  I've spent so many hours on this I'm a bit burnt out.

projectgus, 2020-02-20:
Hi @gandsnut ,

There is an esptool-ck but if esptool.py can't connect to the chip at all then the problem is almost certainly in the serial port config, the USB/serial adapter board, or the hardware setup on the Sonoff itself.

luxarts, 2020-03-12:
Hi, people!

I had the same problem using Arduino IDE 1.8.12 and ESP8266 2.6.3 library. As you mentioned the last version of the library that worked is 2.5.2, which uses esptool v2.6. So I copied the esptool folder from 2.6, installed the ESP8266 2.6.3 library and replaced the esptool version with the 2.6.

1. Install ESP8266 version 2.5.2 from the board manager
2. Go to the folder where the boards are stored and go to `esp8266/tools`
3. Copy the `esptool` folder to safe directory
4. Install ESP8266 version 2.6.3 from the board manager
5. Go to `esp8266/tools`, delete the `esptool` folder and paste the previous copied one

You should be able to upload sketches using Arduino IDE again.

I'll take a look in the next weeks to the esptool to search for the change that broke the upload.

Hope you find it useful!

scroggyg, 2020-04-12:
I am having a problem that I cannot change the baudrate, it is stuck at 460800, I used idf.py menuconfig, saved the changes, and it always stays at 460800. It is changing it in sdkconfig.

projectgus, 2020-04-14:
@scroggyg This sounds like an issue with ESP-IDF and not related to esptool at all. Is it possible you set the environment variable `IDF_MONITOR_BAUD` or `MONITORBAUD` (old name)? This will override the sdkconfig setting.

(If the above doesn't fix it, please [open a new issue on ESP-IDF](https://github.com/espressif/esp-idf/issues/) rather than replying to this unrelated discussion.)

DuncanWilliamGibbons, 2020-06-09:
> Hi, people!
>
> I had the same problem using Arduino IDE 1.8.12 and ESP8266 2.6.3 library. As you mentioned the last version of the library that worked is 2.5.2, which uses esptool v2.6. So I copied [...]

Gave this a try, but still getting the esptool error

radimkarnis, 2020-12-09:
Hello @DuncanWilliamGibbons, 

Were you able to successfully upload a sketch? Are there any more steps you tried to resolve the esptool error?

gandsnut, 2020-12-12:
I've been unable to continue testing this, having temporarily lost the space where I was working.  Will have to re-establish my tools and bench at some point in the future.  Thanks.

radimkarnis, 2020-12-12:
Thank you for the update. I am closing this issue for now, as it seems it is not relevant anymore. Feel free to reopen anytime in the future if you feel further discussion is needed.

closed 2020-12-12 by radimkarnis