espressif/esptool #510: Getting A fatal error occurred: MD5 of file does not match data in flash! message

3xtr3m3d, 2020-01-30:
Hi, I'm Trying to upload new sketch to ESP01 module with 4MB flash using Arduino IDE and i get message

```
esptool.py v2.8
Serial port COM4
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 84:0d:8e:83:90:de
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 362496 bytes to 257370...

Writing at 0x00000000... (6 %)
Writing at 0x00004000... (12 %)
Writing at 0x00008000... (18 %)
Writing at 0x0000c000... (25 %)
Writing at 0x00010000... (31 %)
Writing at 0x00014000... (37 %)
Writing at 0x00018000... (43 %)
Writing at 0x0001c000... (50 %)
Writing at 0x00020000... (56 %)
Writing at 0x00024000... (62 %)
Writing at 0x00028000... (68 %)
Writing at 0x0002c000... (75 %)
Writing at 0x00030000... (81 %)
Writing at 0x00034000... (87 %)
Writing at 0x00038000... (93 %)
Writing at 0x0003c000... (100 %)
Wrote 362496 bytes (257370 compressed) at 0x00000000 in 22.8 seconds (effective 127.3 kbit/s)...
File  md5: 2823c8a64862c8fd1465115abeded515
Flash md5: 825451623e307da01ae4c83c868bd246
MD5 of 0xFF is 27846d141758ea4ad678a3f5bfa1b9b8
Traceback (most recent call last):
  File "C:\Users\exd\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3/tools/upload.py", line 65, in <module>
    esptool.main(cmdline)
  File "C:/Users/exd/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/esptool\esptool.py", line 2938, in main
    operation_func(esp, args)
  File "C:/Users/exd/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/esptool\esptool.py", line 2403, in write_flash
    raise FatalError("MD5 of file does not match data in flash!")
esptool.FatalError: MD5 of file does not match data in flash!
esptool.FatalError: MD5 of file does not match data in flash!
```
Then i tried esptool latest and version 2.3.1 and get same results
```

D:\Backup\esptool-2.3.1>esptool.py --port COM4 write_flash 0x1000  C:\Users\exd\Documents\Arduino\CellMonitor-master-node-final-v2\CellMonitor-master-node-final-v2.ino.generic.bin
esptool.py v2.3.1
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 362496 bytes to 257370...
Wrote 362496 bytes (257370 compressed) at 0x00001000 in 22.7 seconds (effective 127.5 kbit/s)...
File  md5: 2823c8a64862c8fd1465115abeded515
Flash md5: 2e48fe48802d4e985f3c62d1874c1c67
MD5 of 0xFF is 27846d141758ea4ad678a3f5bfa1b9b8

A fatal error occurred: MD5 of file does not match data in flash!
```
also tried 

`esptool.py --port COM4 erase_flash`

but result is the same.  is there anyway i can get this working again?

Regards

projectgus, 2020-02-10:
Hi @3xtr3m3d ,

Is it possible to identify the flash chip model you have, please? You can hopefully read it off the module, if this doesn't work then `esptool.py flash_id` may provide a clue.

There are a few flash chips in the market that don't respond to standard programming commands, #391 has another such example.

Angus

projectgus, 2020-08-28:
Closing due to lack of any update.

closed 2020-08-28 by projectgus