espressif/esptool #511: Sonoff 4CH pro R2 - Problems writing addresses above 0x80000

Petz66, 2020-02-01:
I was trying to update 5 different sonoff 4ch pro r2 devices writing tasmota fw.

For 2 of them I did't get any issues.
For the other 3 devices I always got after programming the: "MD5 of file does not match data in flash!" error.

Reading back from flash and comparing with original bin file I could see that I had some adresses that kept 0xff instead of the programmed values.

All those locations were at addresses above 0x80000.
I performed different attemp of programming and related comparison.

The behaviour is similar for all the devices. 
The failing locations are not always the same (failing addresses in a write attempt get correct values after following writes, while other addresses get failing contents)

I also tried to write a small (100H bytes) bin file containing a fixed pattern (0x55) starting from address 0x80000 and still get some unprogrammed bytes (always a sequence of 4 x 0xFF bytes)

I tried using different programmers.
I used FTDI, CH340G and also tried to get 3.3V from a breadboard power supply.

In so many attempts I never had problems for addresses below 0x80000
There is some sort of protection wrtiting in the 2nd half of flash ?
How can I overcome these problems ?

Thanks for the help.

projectgus, 2020-02-10:
Hi @Petz66 ,

If you run `esptool.py flash_id` on the working & non-working models, do you get different flash chip IDs?

Is it possible the non-working models are actually only 512KB flash size? (Most flash chips will "wrap around" the addresses if you go past the end, so if that's the case you'll see the same data aliased at 0x0 and 0x80000, possibly corrupted if it's been written twice.)

Petz66, 2020-02-16:
Hi @projectgus,

thanks for the suggestions.

Yes, esptool.py recognize the ESP8285

Detecting chip type... ESP8266
Chip is ESP8285
Features: WiFi, Embedded Flash
Crystal is 26MHz
MAC: 2c:f4:32:a7:fe:fc
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 1MB

In my recent attempts I see that the failing locations was always the same per device. 
Changing device also change the set of unprogrammable locations (but still these failings sets are at addresses above 0x80000),
May those location be somehow "broken" ?

projectgus, 2020-02-20:
Hi @Petz66,

Wow, the result of the "all 0x20" certainly shows a lot of strange corruption.

There is no feature of the flash chip in the ESP8285 which would cause it to do this. It's maybe possible that if the chip is browning out due to some power supply stability problem then it might be causing random corruption when trying to internally write the flash. I noticed you said you tried different power supplies, but maybe also check the leads are as short as possible, all connections are firmly made, add capacitor on the Sonoff board if you can, etc.

If you're sure the power supply is 100% correct then I'm afraid the only other explanation is the hardware is somehow faulty or damaged.

radimkarnis, 2020-12-09:
Hello,
I’m closing this issue because it has been inactive for a few months and it seems like it's not relevant anymore. Please reopen if further discussion is needed.

closed 2020-12-09 by radimkarnis