espressif/esptool #512: Timed out waiting for packet header, after: Wrote 3072 bytes (103 compressed) at 0x00008000

psommerfeld, 2020-02-02:
----------------------------- Delete below -----------------------------

----------------------------- Delete above -----------------------------

# Operating system
Win 10

# Python version
3.7.3
# What Chip

ESP32-CAM + FT232RL

# What development board or other hardware is the chip attached to

none

# Is anything else attached to the development board, except for the serial flasher connections?

no

# Are you running esptool.py from an IDE such as Arduino or Eclipse?

No IDE, Windows command prompt

# Full esptool.py command line that was run:

python esptool.py --chip esp32 --before=default_reset --baud=115200 --after=hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 2MB 0x8000 partition_table/partition-table.bin 0x1000 bootloader/bootloader.bin 0x10000 hello-world.bin

# Full output from esptool.py
esptool.py v3.0-dev
Found 1 serial ports
Serial port COM3
Connecting........_____.
Chip is ESP32D0WDQ5 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 84:0d:8e:d0:6e:58
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Compressed 3072 bytes to 103...
Wrote 3072 bytes (103 compressed) at 0x00008000 in 0.0 seconds (effective 1485.2 kbit/s)...

A fatal error occurred: Timed out waiting for packet header

# Do you have any other information from investigating this?
I also have press the RESET button during the initial Connecting, otherwise it will not proceed. Makes me wonder if this is related.

# Is there any other information you can think of which will help us reproduce this problem?

psommerfeld, 2020-02-05:
I tried another ESP32-CAM and it works. I wonder what piece of HW failed.

projectgus, 2020-02-10:
Hi @psommerfeld ,

This could be an issue with power stability, or with signal integrity. May be as simple as a "cold solder joint" on one of the pads which provides intermittent connectivity.

I'm going to close that it seems that it's not an esptool.py issue, if swapping out the ESP32-CAM board fixes it.

closed 2020-02-10 by projectgus

jneeven, 2020-09-19:
> I tried another ESP32-CAM and it works. I wonder what piece of HW failed.

I also have this problem with just one of my boards... Did you ever find the problem?

psommerfeld, 2020-09-19:
No I don't think I found out what was wrong with that hardware.