espressif/esptool #513: Add option to swap and invert DTR/RTS

nitro2k01, 2020-02-09:
I have a custom programming jig with a USB-serial chip for multiple hardware types (STM32, AVR and now ESP32). Unfortunately for the hardware I'm developing atm, I've swapped the lines for EN and GPIO0 signals on the target hardware. For this reason, it would be useful to have the option to swap those signals in software. I've already implemented this change myself, and if there's interest I can submit a PR. This PR would also include options for inverting the both of the signals, to enable all possibilities one might need, for example if a signal is inverted in the EEPROM settings of an FTDI chip.

projectgus, 2020-02-10:
Hi @nitro2k01,

Thanks for offering to send a PR for this.

esptool.py is already dangerously close to a crippling case of "option-itis" (so many options it's hard for users to know what they all do, and hard for developers to test them all exhaustively). At this stage I'd prefer for one-off custom pieces of hardware to hack up esptool.py as necessary, rather than provide a lot of command line options to accomodate them at the cost of further option-itis and code complexity. 

Something I would be happy to accept is a PR that abstracted out the functions that set DTR & RTS, so that they're easy for someone else to override and replace with whatever custom behaviour is needed.

nitro2k01, 2020-02-10:
To clarify, in my case, these are prototype board, and I'm considering it a bug that will be fixed in the next board revision. If the use case was to support custom hardware long-term, I agree that a local solution would be preferable. But the use case I'm imagining for these switches is more like as a temporary solution. Imagine: You need to reprogram an ESP module on a board for whatever reason and hand solder wires to your dubious Chinese serial adapter. You're unsure which wire you soldered where. Wouldn't it be nice if you could try swapping the lines real quick without turning on the soldering iron again? (And without a deep dive into the code.) The options would exist as a quick troubleshooting hack.

I also think the relatively small scope of the change would make it low risk. The change can be wrapped inside the existing functions. And there's a systematic way to test the code even without access to special hardware: enable any of the options (except inverting EN since it toggles anyway) and confirm that the flashing fails when it previously worked.

With regard to the last paragraph, isn't that sort of what `_connect_attempt` is, meaning the proper course of action in that case would be to subclass `ESPLoader` or one of its subclasses and replace `_connect_attempt`?

projectgus, 2020-02-11:
> I also think the relatively small scope of the change would make it low risk.

I agree this one change by itself is probably low risk. It's not so much the risk of regressions, but adding untestable features for niche use cases is ultimately not sustainable. Eventually the code becomes untestable and unmaintainable due to its own weight.

> I also think the relatively small scope of the change would make it low risk. The change can be wrapped inside the existing functions. And there's a systematic way to test the code even without [...]

Testing that changing the mode breaks something is not the same as testing that the alternative mode works as specified.

esptool actually has a set of automated tests for most common functions: https://github.com/espressif/esptool/blob/master/test/test_esptool.py

> With regard to the last paragraph, isn't that sort of what _connect_attempt is, meaning the proper course of action in that case would be to subclass ESPLoader or one of its subclasses and replace _connect_attempt?

Currently, yes that's a viable approach. You could make a wrapper script which either subclassed the loader, or monkey-patched the function. Let me know if you want me to mock up an example of how to do this.

What I was suggestingis actually something even more factored where there is a function like `set_hardware_mode(self, mode)` and the mode is "reset", "normal boot", or "enter bootloader mode" and then _connect_attempt can call these in sequence, or possibly even three functions `set_hardware_reset()`, `set_hardware_normal_boot()`, `set_hardware_bootloader_mode()`.

radimkarnis, 2020-12-09:
Hello,
I’m closing this issue because it has been inactive for a few months and it seems like it's not relevant anymore. Please reopen if further discussion is needed.

closed 2020-12-09 by radimkarnis