espressif/esptool #514: elf2image fails to produce bootable image

readmodifywrite, 2020-02-10:
# Operating system
MacOS 10.14.6

# Python version
Python 2.7.16

# What Chip
ESP-12

# What development board or other hardware is the chip attached to
Adafruit Feather Huzzah

# Is anything else attached to the development board, except for the serial flasher connections?
Nope.

# Are you running esptool.py from an IDE such as Arduino or Eclipse?
Running from bash terminal.

# Full esptool.py command line that was run:
`python esptool.py elf2image -o image.bin -ff 40m -fm dio -fs 4MB main.elf`

If I process the same elf image with esptool-ck, the image boots and runs correctly:
`./esptool -eo main.elf -bo firmware.bin -bm dio -bf 40 -bz 4M -bs .text -bs .data -bs .rodata -bc -ec -eo main.elf -es .irom0.text firmware.bin.irom0text.bin -ec -v`

esptool-py's elf2image output fails to boot after the first segment:
```
ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x3ffe8000, len 1328, room 16 
tail 0
chksum 0xc4
load 0x3ffe852f, len 8, room 8 
```
Boot process halts at this point.

esptool-ck's boot output:

```
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 25324, room 16 
tail 12
chksum 0x10
ho 0 tail 12 room 4
load 0x3ffe8000, len 1328, room 12 
tail 4
chksum 0x3b
load 0x3ffe8540, len 2808, room 4 
tail 4
chksum 0xfa
csum 0xfa
```
And the program runs correctly.

image_info yields some differences:
```
esptool-py:
Image version: 1
Entry point: 40100004
4 segments

Segment 1: len 0x00530 load 0x3ffe8000 file_offs 0x00000008 [DRAM]
Segment 2: len 0x00008 load 0x3ffe852f file_offs 0x00000540 [DRAM]
Segment 3: len 0x00af8 load 0x3ffe8540 file_offs 0x00000550 [DRAM]
Segment 4: len 0x062ec load 0x40100000 file_offs 0x00001050 [IRAM]
Checksum: fa (valid)

```
esptool-ck:
```
Image version: 1
Entry point: 40100004
3 segments

Segment 1: len 0x062ec load 0x40100000 file_offs 0x00000008 [IRAM]
Segment 2: len 0x00530 load 0x3ffe8000 file_offs 0x000062fc [DRAM]
Segment 3: len 0x00af8 load 0x3ffe8540 file_offs 0x00006834 [DRAM]
Checksum: fa (valid)
```

To summarize:
The actual ELF image itself works fine.  I'm using esptool-py to load both variants.  The only difference is the elf2image processing.

projectgus, 2020-02-20:
Hi @readmodifywrite,

It looks like the ELF has an additional section which isn't being processed by `esptool-ck` (because each section name is given on the command line, this is not too surprising). 

Where was `main.elf` generated from? Is it possible to attach it to the issue?

(If the ELF contents are sensitive in some way then that's OK, but could you please post the output of `xtensa-lx106-elf-objdump -h main.elf` instead?)

readmodifywrite, 2020-02-25:
[main.elf.zip](https://github.com/espressif/esptool/files/4248215/main.elf.zip)

main.elf attached (via zip file).

I'm using a custom build system, so I've been reverse engineering the Arduino and PlatformIO builds.

readmodifywrite, 2020-02-25:
Ok, so I'm seeing a ton of other weirdness, and I'm starting to suspect I have a strange compiler issue instead.  I'm not sure why I was seeing a disparity between both esptools, but now I'm seeing an issue where images from both tools will fail to load if I include a specific variable in a build, and both work if I remove it.  I've been using a patched compiler to get the mforce-l32 option, and it's looking like that is causing some issues.  If I go back to the stock compiler from the SDK, things seem to work ok with both esptools.

So, tl;dr, I suspect I might have a different problem that was just manifesting in a convoluted way. 

I'm going to close this issue for now, and re-open if I get some more conclusive evidence there's an actual issue with esptool.  For now, it seems like I have a different problem.

Thanks!