espressif/esptool #522: Why need to separate IROM data (esp8266 --version=1)?

r44083, 2020-04-04:
**Is your feature request related to a problem? Please describe.**
There is a strange logic to separate IROM data and everything else when creating binary firmware file from .elf:
```python
class ESP8266ROMFirmwareImage(BaseFirmwareImage):
...
    def save(self, basename):
        """ Save a set of V1 images for flashing. Parameter is a base filename. """
        # IROM data goes in its own plain binary file
...
        # everything but IROM goes at 0x00000 in an image file
...
```
As a result, ```esptool elf2image --version=1``` generate two .bin files instead of one solid binary.
What is the purpose for such logic?

**Describe the solution you'd like**
Don't split IROM data in separate binary file. ```--version=1``` should generate one .bin file which should be loaded to 0x40200000.

**Describe alternatives you've considered**
-

**Additional context**
-

projectgus, 2020-04-09:
Hi @r44083 ,

The very first versions of the ESP8266 SDK worked this way, part of the .bin file was loaded to RAM out of the .bin file (first file) but the part mapped from the flash cache was flashed to a totally separate address in flash, not part of the .bin file. This option exists to support that.

Any modern version of the ESP8266 SDK (since at least 2016, probably earlier) will use the `--version=2` .bin file format.

However, I notice now that the default `esptool.py elf2image` will use `--version=1` if not specified. esptool v3.0 is coming up, so we can change this to 2 - or possibly even remove "version 1" support for ESP8266 entirely, as I don't think anyone is using it any more.

projectgus, 2020-04-09:
I see now from https://github.com/espressif/ESP8266_RTOS_SDK/issues/858 that you're looking to bypass the bootloader stage for an RTOS SDK app that boots directly from ROM.

Unfortunately, I don't think that the RTOS SDK app will boot correctly if it doesn't have a bootloader to initialize it, but I don't work on this SDK so I can't say for sure - will wait for a colleague to answer.

r44083, 2020-04-09:
Currently, I already have built ESP8266_RTOS_SDK app which doesn't require software bootloader.
It requires only the following diff in components/esp8266/source/startup.c:
```diff
-void call_user_start(size_t start_addr)
+__attribute__((section(".iram1"))) void call_user_start(size_t start_addr)
{
+    Cache_Read_Disable();
+    Cache_Read_Enable(0, 0, 0);
+    start_addr = 0;
```

```diff
    phy_get_bb_evm();

-   assert(nvs_flash_init() == 0);
+   //assert(nvs_flash_init() == 0);
    assert(wifi_nvs_init() == 0);
```
It works OK.

But the issue is that esptool elf2image --version=1 generates two separate binaries:
image.bin-0x00000
image.bin-0x10000

Whereas 0x10000 is start of ```irom0_0_seg``` (named ```iram0_2_seg``` in latest SDK v3.3-rc1).

projectgus, 2020-04-20:
Hi @r44083 ,

> I already have built ESP8266_RTOS_SDK (v3.2 and v3.3-rc1) app which doesn't require software bootloader. [...] It requires only the following diff

I don't mean to be rude, but if it requires modifying the SDK internals then it's not a supported configuration. It's possible that the SDK developers can be convinced to support it, you'll need to talk to them.

Note that by relying on the ROM bootloader this way you lose the ability to do OTA updates, and also there is no way to verify the app before booting it.

>generates two separate binaries

Yes, this is by design to support old versions of the SDK.

The first file `image.bin-0x00000` is the ".bin" file with all of the segments which are copied into RAM (DRAM & IRAM) on boot. In the supported configuration of RTOS SDK, this is the bootloader and this file contains every segment from the ELF file as there are no segments mapped via a flash cache address ("IROM"). This file has an internal structure and a checksum.

The `image.bin-0x10000` contains the flash mapped part of the app ("IROM"), which needs to be flashed to the address specified in the flash chip (this is the "App Offset" in the RTOS SDK, can be changed if you move the partition table), and is then mapped via the flash cache. This file doesn't have any metadata at all - it's just the raw binary contents of `iram0_2_seg` from the ELF file. It also has no checksum or other validation.

The reason why the second file isn't a segment in the first file is that the ESP8266 ROM bootloader doesn't know how to configure the flash cache.

Since approx 2014/2015 the ESP8266 SDK doesn't build firmware this way, and uses a software bootloader at offset 0x0 instead. This lets the whole structure of the firmware be treated as a single file with headers for each segment, metadata, etc.

If you really want to use the ROM bootloader to load the app and then enable the flash cache during app startup then you can still do this, but it's not currently supported.

You can combined the two files into one file now by padding `image.bin-0x00000` to length 0x10000 and then appending `image.bin-0x10000` to it. This gives you a "single file" - where flashing it to offset 0x0 the result is the same as flashing both files. Note that only the first part of the "joined file" can be parsed by any tools or the ESP8266 ROM bootloader, the second part is not readable by these because it's raw binary data with no structure.

projectgus, 2020-08-28:
Closing as the question is answered, and we can't change this mode's behaviour without causing compatibility issues.

closed 2020-08-28 by projectgus