espressif/esptool #525: [ESP32-S2]A fatal error occurred: Invalid head of packet (0x50)

imliubo, 2020-04-19:
# Operating system
Windows
# Python version
python3.7
# What Chip
ESP32-S2
# What development board or other hardware is the chip attached to
Designer by myself.
USB=>UART
[image]
# Is anything else attached to the development board, except for the serial flasher connections?

NO.

# Are you running esptool.py from an IDE such as Arduino or Eclipse?

No IDE, Windows command prompt

# Full esptool.py command line that was run:
```bash
idf.py -p COM8 flash -b 115200
```
# Full output from esptool.py
```bash
Adding "flash"'s dependency "all" to list of commands with default set of options.
Executing action: all (aliases: build)
Running ninja in directory g:\espressif\esp32\esp-idf-v4.1-beta1\examples\get-started\blink\build
Executing "ninja all"...
[1/3] Performing build step for 'bootloader'
ninja: no work to do.
Executing action: flash
Running esptool.py in directory g:\espressif\esp32\esp-idf-v4.1-beta1\examples\get-started\blink\build
Executing "G:\espressif\tools\.espressif\python_env\idf4.1_py2.7_env\Scripts\python.exe G:\espressif\esp32\esp-idf-v4.1-beta1\components/esptool_py/esptool/esptool.py -p COM8 -b 115200 --before default_reset --after hard_reset --chip esp32s2beta write_flash @flash_project_args"...
esptool.py -p COM8 -b 115200 --before default_reset --after hard_reset --chip esp32s2beta write_flash --flash_mode dio --flash_freq 40m --flash_size 4MB 0x8000 partition_table/partition-table.bin 0x1000 bootloader/bootloader.bin 0x10000 blink.bin
esptool.py v2.9-dev
Serial port COM8
Connecting........__
Chip is ESP32S2 Beta
Features: Engineering Sample
Crystal is 40MHz
MAC: 7c:df:a1:00:3b:e2
Uploading stub...
Running stub...

A fatal error occurred: Invalid head of packet (0x50)
esptool.py failed with exit code 2
```

# Do you have any other information from investigating this?

Always fail.

# Is there any other information you can think of which will help us reproduce this problem?

Windows 10. USB->UART chip is CP2102.

projectgus, 2020-04-20:
Hi @imliubo ,

If you have an ESP32-S2 and not the "ESP32-S2 Beta" chip then you'll need to update to esptool v3.0-dev. You can get this from the master branch, or if you're using ESP-IDF then you can update to the latest master branch version of ESP-IDF.

Angus

imliubo, 2020-04-20:
Hi @projectgus ,
Thank you for your help, yes! According to your guidance, I have successfully solved the problem. The reason I downloaded the release version is because pulling the main version is very time-consuming and will fail (GFW ...). In the release version, I only found that v4.1-beta1 supports ESP32S2. However, I have successfully downloaded the master version through some special methods. It is recommended to release another version for ESP32S2.
[image]

Have a nice day!I will close this issue.

IAMLIUBO

closed 2020-04-20 by imliubo

projectgus, 2020-04-20:
Hi @imliubo ,

Glad you found a solution. Just to make sure everything is understood, and for anyone else who finds this issue:

> In the release version, I only found that v4.1-beta1 supports ESP32S2

ESP-IDF v4.1 will support *ESP32-S2 Beta* only.  This is a different chip to the ESP32-S2 final hardware, and not compatible.

ESP-IDF v4.2 will support ESP32-S2 final chip. This is currently available on the ESP-IDF master branch only. [The docs for this version, including the getting started guide, are here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/).

esptool is versioned separately to ESP-IDF. esptool.py v2.9-dev is a development version that supports ESP32-S2 Beta, but esptool.py v3.0-dev or newer is required for ESP32-S2.

If you get a version of ESP-IDF that supports ESP32-S2, it will come with a version of esptool that supports ESP32-S2.

imliubo, 2020-04-21:
Thank you very much for your guidance.I initially thought that the ESP32 S2 beta is the same as the ESP32 S2, which really didn't pay much attention to.