espressif/esptool #526: Sonoff 4CH PRO R2 flash issue

PetrVyhlidal, 2020-04-24:
# Operating system
Windows

# Python version
Python 3.7.4
Can run `python -V` to check this.

# What Chip
Chip is ESP8285

# What development board or other hardware is the chip attached to
USB na UART TTL - CP2102

If your hardware is custom or unusual, please attach a photo to the issue.
Sonoff 4CH PRO R2

# Is anything else attached to the development board, except for the serial flasher connections?

# Are you running esptool.py from an IDE such as Arduino or Eclipse?
No IDE, Windows command prompt

esptool.py --port COM8 erase_flash 
esptool.py --port COM8 write_flash -fs 1MB -fm dout 0x0 tasmota-CZ.bin

# Full esptool.py command line that was run:
esptool.py --port COM8 write_flash -fs 1MB -fm dout 0x0 tasmota-CZ.bin

# Full output from esptool.py

c:\Users\petr.vyhlidal\AppData\Local\Programs\Python\Python37-32\Lib\site-packages>esptool.py --port COM8 write_flash -fs 1MB -fm dout 0x0 tasmota-CZ.bin
esptool.py v2.8
Serial port COM8
Connecting....
Detecting chip type... ESP8266
Chip is ESP8285
Features: WiFi, Embedded Flash
Crystal is 26MHz
MAC: 60:01:94:be:29:f5
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Compressed 592608 bytes to 411442...
Wrote 592608 bytes (411442 compressed) at 0x00000000 in 37.8 seconds (effective 125.4 kbit/s)...
File  md5: a5f679f725bd12d04a7056378224434a
Flash md5: 67faba40ae284c860babb0e9488ff484
MD5 of 0xFF is b0083667dca081dc853efa022040e583

A fatal error occurred: MD5 of file does not match data in flash!

# Do you have any other information from investigating this?
First flash succeeded.
Next runs failed

# Is there any other information you can think of which will help us reproduce this problem?

N/A

projectgus, 2020-04-27:
Hi @PetrVyhlidal ,

Thanks for providing all these details.

* Could you please run the command `esptool.py -p PORT flash_id` and post the output?
* The `tasmota-CZ.bin` firmware doesn't contain any sensitive information, right? If so, and if the Sonoff flash probably doesn't contain anything sensitive, could you please run `esptool.py -p PORT read_flash 0x0 0x100000 readback.bin` and then attach the contents of `readback.bin` and `tasmota-CZ.bin` to this issue? If it's possible the binaries may contain some personal info then please don't do this (but if the Sonoff is new and Tasmota firmware is generic, there shouldn't be anything personal on there).

PetrVyhlidal, 2020-04-29:
Hi.
Here is output:

c:\Users\petr.vyhlidal\AppData\Local\Programs\Python\Python37-32\Scripts\esphome\Scripts>esptool.exe -p COM8 flash_id
esptool.py v2.7
Serial port COM8
Connecting....
Detecting chip type... ESP8266
Chip is ESP8285
Features: WiFi, Embedded Flash
Crystal is 26MHz
MAC: 60:01:94:be:29:f5
Uploading stub...
Running stub...
Stub running...
Manufacturer: 51
Device: 4014
Detected flash size: 1MB
Hard resetting via RTS pin...

PetrVyhlidal, 2020-04-29:
And here is content of device (and log):

c:\Users\petr.vyhlidal\AppData\Local\Programs\Python\Python37-32\Scripts\esphome\Scripts>esptool.exe -p COM8 read_flash 0x0 0x100000 readback.bin
esptool.py v2.7
Serial port COM8
Connecting....
Detecting chip type... ESP8266
Chip is ESP8285
Features: WiFi, Embedded Flash
Crystal is 26MHz
MAC: 60:01:94:be:29:f5
Uploading stub...
Running stub...
Stub running...
1048576 (100 %)
1048576 (100 %)
Read 1048576 bytes at 0x0 in 95.5 seconds (87.8 kbit/s)...
Hard resetting via RTS pin...

[readback.zip](https://github.com/espressif/esptool/files/4550150/readback.zip)
[tasmota-CZ.zip](https://github.com/espressif/esptool/files/4550151/tasmota-CZ.zip)

PetrVyhlidal, 2020-05-16:
> Hi @PetrVyhlidal ,
>
> Thanks for providing all these details.
>
>     * Could you please run the command `esptool.py -p PORT flash_id` and post the output?
>
>     * The `tasmota-CZ.bin` firmware doesn't [...]

Hello, 
please, could you try to look on attached info in previous posts ?
Thanks a lot !!

Petr

projectgus, 2020-05-21:
Hi @PetrVyhlidal ,

Thanks for the update and sorry for the slow turnaround.

Looking at the dump, it looks like a lot of individual bits in the flash have not programmed correctly. Usually this means there is a power supply problem. How is the 3.3V power source for the ESP8285 chip in the SONOFF being provided? If it's coming from the USB/serial adapter then probably this is the root cause, the adapter doesn't supply enough current to keep the voltage stable while flashing.

Some more details can be found here:
https://github.com/espressif/esptool#insufficient-power

radimkarnis, 2020-12-09:
Hello @PetrVyhlidal,

Have you been able to perform a successful flash with a different power supply?

PetrVyhlidal, 2020-12-09:
HelloNo, i was not able. Finally found, that ESP chip was brokenSent from Android device

radimkarnis, 2020-12-09:
Thank you for your reply, I'm sorry to hear that. If you have any other questions, feel free to ping me. I am now closing this issue.

closed 2020-12-09 by radimkarnis