espressif/esptool #527: ESP01:  Failed to connect to Espressif device: Timed out waiting for packet header

Robino, 2020-04-26:
Good Morning, 

I have a strange error launching on my raspbian on RPi4, esptool to have information on my ESP01.
I can make AT Commands with miniterm for exemple without any issues (returns informations)

#523  Operating system
Linux raspberrypi 4.19.97-v7l+

# Python version
Python 3.7.3
Can run `python -V` to check this.

# What Chip

ESP01 may 2016 / 
AT+GMR
AT version:1.1.0.0(May 11 2016 18:09:56)
SDK version:1.5.4(baaeaebb)
compile time:May 20 2016 15:08:19

# What development board or other hardware is the chip attached to

The chip is used as is, direct connection without any other hardware attached.

# Is anything else attached to the development board, except for the serial flasher connections?

I have attached to the board raspberry the GPIO 4,17,22,27 (I try removing all and it is the same

# Are you running esptool.py from an IDE such as Arduino or Eclipse?

I launch the script directly from shell command

# Full esptool.py command line that was run:

esptool.py read_mac

# Full output from esptool.py

esptool.py v2.8
Found 1 serial ports
Serial port /dev/ttyAMA0
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.001 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.006 Read 1 bytes: c0
TRACE +0.000 Read 15 bytes: 08240707122055555555555555555502 | Offline | ttyAMA0                                                                                                   
TRACE +0.001 Read 1 bytes: 55
TRACE +0.000 Read 15 bytes: 555555555555555555555555555555
TRACE +0.001 Read 1 bytes: 55
TRACE +0.000 Read 7 bytes: 555555555555c0
TRACE +0.000 Received full packet: 
    0824070712205555 5555555555555555 | .$... UUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    555555555555                      | UUUUUU
TRACE +0.101 Timed out waiting for packet header
.TRACE +0.051 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.001 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.006 Read 1 bytes: c0
TRACE +0.000 Read 15 bytes: 082407071220555555555555555555
TRACE +0.001 Read 1 bytes: 55
TRACE +0.000 Read 15 bytes: 555555555555555555555555555555
TRACE +0.001 Read 1 bytes: 55
TRACE +0.000 Read 7 bytes: 555555555555c0
TRACE +0.000 Received full packet: 
    0824070712205555 5555555555555555 | .$... UUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    555555555555                      | UUUUUU
TRACE +0.101 Timed out waiting for packet header
.TRACE +0.051 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.001 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.006 Read 1 bytes: c0
TRACE +0.000 Read 15 bytes: 082407071220555555555555555555
TRACE +0.001 Read 1 bytes: 55
TRACE +0.000 Read 15 bytes: 555555555555555555555555555555
TRACE +0.001 Read 1 bytes: 55
TRACE +0.000 Read 7 bytes: 555555555555c0
TRACE +0.000 Received full packet: 
    0824070712205555 5555555555555555 | .$... UUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    555555555555                      | UUUUUU
TRACE +0.101 Timed out waiting for packet header
.TRACE +0.051 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.001 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.006 Read 1 bytes: c0
TRACE +0.001 Read 15 bytes: 082407071220555555555555555555
TRACE +0.001 Read 1 bytes: 55
TRACE +0.001 Read 15 bytes: 555555555555555555555555555555
TRACE +0.001 Read 8 bytes: 55555555555555c0
TRACE +0.001 Received full packet: 
    0824070712205555 5555555555555555 | .$... UUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    555555555555                      | UUUUUU
TRACE +0.101 Timed out waiting for packet header
.TRACE +0.051 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.001 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.007 Read 1 bytes: c0
TRACE +0.001 Read 15 bytes: 082407071220555555555555555555
TRACE +0.001 Read 16 bytes: 55555555555555555555555555555555
TRACE +0.001 Read 8 bytes: 55555555555555c0
TRACE +0.000 Received full packet: 
    0824070712205555 5555555555555555 | .$... UUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    555555555555                      | UUUUUU
TRACE +0.101 Timed out waiting for packet header
.^C
Traceback (most recent call last):
  File "/usr/local/bin/esptool.py", line 3201, in <module>
    _main()
  File "/usr/local/bin/esptool.py", line 3194, in _main
    main()
  File "/usr/local/bin/esptool.py", line 2883, in main
    esp = ESPLoader.detect_chip(each_port, initial_baud, args.before, args.trace)
  File "/usr/local/bin/esptool.py", line 274, in detect_chip
    detect_port.connect(connect_mode)
  File "/usr/local/bin/esptool.py", line 478, in connect
    last_error = self._connect_attempt(mode=mode, esp32r0_delay=True)
  File "/usr/local/bin/esptool.py", line 447, in _connect_attempt
    time.sleep(0.4)  # allow watchdog reset to occur
KeyboardInterrupt

-----

esptool.py v2.8
Found 1 serial ports
Serial port /dev/ttyAMA0
Connecting........_____....._____....._____....._____....._____....._____....._____
/dev/ttyAMA0 failed to connect: Failed to connect to Espressif device: Timed out waiting for packet header

# Do you have any other information from investigating this?

No, just when I'm using the program miniterm I can access without any issue with AT commands.

Robino, 2020-04-26:
I made a new test using minicom 2.7.1 version.
Commands working:
- AT ==> ok
- AT+GMR ==> ok
- AT+UART_CUR ==> NO (ERROR)
- AT+SYSADC ==> NO (ERROR).

I hope it can help

projectgus, 2020-04-27:
Hi @Robino ,

In order to talk to esptool, the ESP-01 has to boot into a different mode.

Usually this is done by connecting GPIO0 to Ground when turning on the chip. If you don't ground GPIO0 then it boots the firmware from flash, which in your case is the AT command firmware.

For more details, see here:
https://github.com/espressif/esptool/wiki/ESP8266-Boot-Mode-Selection

Robino, 2020-04-27:
Hi @projectgus 

Wahoo ! A big thank it helps me a lot, it works perfectly now.

Have a good day.

closed 2020-04-27 by Robino