espressif/esptool #528: Ubuntu 20.04 ImportError: No module named serial

Duino999, 2020-05-10:
Ubuntu 20.04
Python 2.7.18rc1
ESP32-WROOM32
MH ET LIVE Mini
Running esptool.py from Arduino IDE 1.8.10
board installed ESP32 by Espressif Systems version 1.04
esptool version 2.8

Error when compiling from Arduino IDE:
.arduino15/packages/esp32/tools/esptool_py/2.6.1/esptool.py", line 38, in <module>
    import serial
ImportError: No module named serial

It was working fine with Ubuntu 19.04 but not with the new version.

Try pip3 uninstall serial the pip3 then install pyserial, but no difference.

Duino999, 2020-05-10:
If I'm using python3 for esptool.py execution is OK. but with python esptool.py its the same error.

Duino999, 2020-05-10:
I found the following solution on 

https://askubuntu.com/questions/320996/how-to-make-python-program-command-execute-python-3

On Ubutu 20.04 the following command will fix the issue

sudo apt install python-is-python3

On top of that, you can prevent Python 2 from being installed as a dependency of something in the future with apt-mark hold:

sudo apt-mark hold python2 python2-minimal python2.7 python2.7-minimal libpython2-stdlib 
libpython2.7-minimal libpython2.7-stdlib

I also install pyserial pip3 install pyserial.

esptool is the version 2.8

Hope it could be usefull.

Duino999, 2020-05-10:
Finally this the link to the solution.

https://askubuntu.com/a/1031733/1080482

kilnageer2, 2020-05-14:
Thanks Duino999. I just had the same issue.

edoardottt, 2020-06-09:
I had the same issue and [this](https://askubuntu.com/a/1031733/1080482) didn't solve the problem.

projectgus, 2020-08-28:
Closing as this was an issue with Ubuntu 20.04 + Arduino ESP8266 bundling of esptool (see linked issue above), rather than esptool itself.

closed 2020-08-28 by projectgus

yorkfrog, 2021-05-14:
@cyzen that link partially worked but I also needed to install `pyserial` after `python-is-python3`.  The link below explains the full process of how to solve this error and some extra ESP32 setup trouble shooting.
 http://arduino-er.blogspot.com/2020/06/install-esp32esp8266-to-arduino-ide-on.html?m=1

rundekugel, 2021-06-20:
Had the same issue. 
You can change your system or just change the tool. Edit 1st line:

    #!/usr/bin/env python3

So it'll be started with python3, no matter, which python is your default.

If you use it sparely, you can also do this:
    python3 esptool.py ...

JKSAR, 2021-08-14:
> @cyzen that link partially worked but I also needed to install `pyserial` after `python-is-python3`. The link below explains the full process of how to solve this error and some extra ESP32 setup [...]

If you are working with ESP32, I see that you will necessarily have to perform these additional steps mentioned by @yorkfrog in this link:

http://arduino-er.blogspot.com/2020/06/install-esp32esp8266-to-arduino-ide-on.html?m=1

Thanks @yorkfrog !!!

fire17, 2021-08-21:
# sudo apt install python-serial python3-serial 

solved it! :D

J0hnnyF1v3, 2021-09-02:
I had this error, fixing permissions remedied the issue;
`chown -R $USER /usr/local/bin/*`
I was able to proceed after this, erase complete.

A-R-S-D, 2022-05-23:
What worked for me:
`pip install pyserial`

eagl1, 2023-05-17:
I had to run these two commands in sequence:

```
1. sudo apt install python3-pip
2. pip install esptool
```

inchoxd, 2024-12-30:
I have same issue.
I was installing pyserial, esptool from pip.
But, I had error when running command bellow.
```
arduino-cli compile --fqbn esp32:esp32:m5stack_cores3
Traceback (most recent call last):
  File "/home/inchoxd/.arduino15/packages/esp32/tools/esptool_py/4.6/esptool.py", line 34, in <module>
    import esptool
  File "/home/inchoxd/.arduino15/packages/esp32/tools/esptool_py/4.6/esptool/__init__.py", line 41, in <module>
    from esptool.cmds import (
    ...<24 lines>...
    )
  File "/home/inchoxd/.arduino15/packages/esp32/tools/esptool_py/4.6/esptool/cmds.py", line 14, in <module>
    from .bin_image import ELFFile, ImageSegment, LoadFirmwareImage
  File "/home/inchoxd/.arduino15/packages/esp32/tools/esptool_py/4.6/esptool/bin_image.py", line 14, in <module>
    from .loader import ESPLoader
  File "/home/inchoxd/.arduino15/packages/esp32/tools/esptool_py/4.6/esptool/loader.py", line 30, in <module>
    import serial
ModuleNotFoundError: No module named 'serial'

Used platform Version Path
esp32:esp32   3.0.5   /home/inchoxd/.arduino15/packages/esp32/hardware/esp32/3.0.5
Error during build: exit status 1
```

When I run `python -c "import serial"`, it is no error.
Plese tell me how to solve error.