espressif/esptool #530: Upload error over remote desktop

spigotx, 2020-05-16:
# Operating system

Windows 10

# Python version

Python 3.8.2

# What Chip

ESP8266

# What development board or other hardware is the chip attached to

Wemos D1 mini

# Is anything else attached to the development board, except for the serial flasher connections?

No

# Are you running esptool.py from an IDE such as Arduino or Eclipse?

MS Visual Studio
Arduino IDE 1.8.12

# Full esptool.py command line that was run:

`upload.py" --chip esp8266 --port "COM5" --baud "921600" --before default_reset --after hard_reset write_flash 0x0 Test.ino.bin"`

# Full output from esptool.py

```
esptool.py v2.8
Serial port COM5
Connecting....
Traceback (most recent call last):
  File "C:\Users\dtarduino\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.7.1/tools/upload.py", line 65, in <module>
    esptool.main(cmdline)
The uploader process failed
  File "C:/Users/dtarduino/AppData/Local/arduino15/packages/esp8266/hardware/esp8266/2.7.1/tools/esptool\esptool.py", line 2890, in main
    esp.connect(args.before)
  File "C:/Users/dtarduino/AppData/Local/arduino15/packages/esp8266/hardware/esp8266/2.7.1/tools/esptool\esptool.py", line 475, in connect
    last_error = self._connect_attempt(mode=mode, esp32r0_delay=False)
  File "C:/Users/dtarduino/AppData/Local/arduino15/packages/esp8266/hardware/esp8266/2.7.1/tools/esptool\esptool.py", line 455, in _connect_attempt
    self.sync()
  File "C:/Users/dtarduino/AppData/Local/arduino15/packages/esp8266/hardware/esp8266/2.7.1/tools/esptool\esptool.py", line 394, in sync
    timeout=SYNC_TIMEOUT)
  File "C:/Users/dtarduino/AppData/Local/arduino15/packages/esp8266/hardware/esp8266/2.7.1/tools/esptool\esptool.py", line 347, in command
    p = self.read()
  File "C:/Users/dtarduino/AppData/Local/arduino15/packages/esp8266/hardware/esp8266/2.7.1/tools/esptool\esptool.py", line 292, in read
    return next(self._slip_reader)
  File "C:/Users/dtarduino/AppData/Local/arduino15/packages/esp8266/hardware/esp8266/2.7.1/tools/esptool\esptool.py", line 2041, in slip_reader
    read_bytes = port.read(1 if waiting == 0 else waiting)
  File "C:/Users/dtarduino/AppData/Local/arduino15/packages/esp8266/hardware/esp8266/2.7.1/tools/pyserial\serial\serialwin32.py", line 293, in read
    raise SerialException("GetOverlappedResult failed ({!r})".format(ctypes.WinError()))
serial.serialutil.SerialException: GetOverlappedResult failed (OSError(22, 'Reached the end of the file.', None, 38))
```

# Do you have any other information from investigating this?

# Is there any other information you can think of which will help us reproduce this problem?

Upload over direct connection is working. This problem occurs only over remote desktop connection.

Latest working ESP8266 version is 2.4.2
Esptool command from 2.4.2:
`esptool.exe -cd nodemcu -cb 921600 -cp "COM5" -ca 0x00000 -cf "Test.ino.bin"`

Arduino IDE settings v2.4.2/2.7.1
[image]

projectgus, 2020-05-22:
Hi @spigotx ,

Thanks for the detailed report.

The `esptool.exe` in older versions of Arduino IDE is actually a totally different program to this esptool, that program ([esptool-ck](https://github.com/igrr/esptool-ck)) was written in C not Python. I can't explain why esptool-ck working but `esptool.py` is not - they both access the same serial port but maybe in a different way or with a different sequence of operating system calls.

In this case, pyserial (the serial library used by esptool) is calling the [Windows ReadFile() function to read from the open COM port](https://github.com/pyserial/pyserial/blob/master/serial/serialwin32.py#L293) and Windows is returning this 'Reached the end of the file' error. I can't explain why this would happen.

> Upload over direct connection is working. This problem occurs only over remote desktop connection.

Is the WeMos D1 Mini physically connected to the same computer in both cases? Is any feature like remote desktop serial port redirection is being used?

spigotx, 2020-05-22:
Hi @projectgus 

Thank you for the clarification.

> 
> Is the WeMos D1 Mini physically connected to the same computer in both cases? Is any feature like remote desktop serial port redirection is being used?
>

In both cases I'm talking about the same pc (virtual machine) I'm using for Arduino.

Direct connection - I'm using VMware hypervisor (ESXi) on physical servers. Wemos D1 mini is connected to USB port (server) which is redirected over passthrough from an ESXi host (server) to the virtual machine. 

Remote desktop - Wemos D1 mini is connected to USB port on my physical pc and this USB port is redirected to the virtual machine over remote desktop.

[image]

projectgus, 2020-05-25:
Hi @spigotx ,

Thanks for the explanation and the helpful diagram.

* Are you able to run esptool.py on the local PC directly, and does this work? It's OK if this is not possible for whatever reason, but it rules out some problem with the local PC's connection to the port.
* What operating system is running on the local PC, and what Remote Desktop client program do you use (if not the standard Windows one)?

spigotx, 2020-05-25:
Hi @projectgus

- I'm able to run esptool on my local PC:
> esptool.py v2.8
> Serial port COM5
> Connecting....
> Chip is ESP8266EX
> Features: WiFi
> Crystal is 26MHz
> MAC: xx:xx:xxx:xx:xx
> Uploading stub...
> Running stub...
> Stub running...
> Changing baud rate to 460800
> Changed.
> Configuring flash size...
> Auto-detected Flash size: 4MB
> Compressed 437296 bytes to 308387...
> 
> Writing at 0x00000000... (5 %)
> Writing at 0x00004000... (10 %)
> Writing at 0x00008000... (15 %)
> Writing at 0x0000c000... (21 %)
> Writing at 0x00010000... (26 %)
> Writing at 0x00014000... (31 %)
> Writing at 0x00018000... (36 %)
> Writing at 0x0001c000... (42 %)
> Writing at 0x00020000... (47 %)
> Writing at 0x00024000... (52 %)
> Writing at 0x00028000... (57 %)
> Writing at 0x0002c000... (63 %)
> Writing at 0x00030000... (68 %)
> Writing at 0x00034000... (73 %)
> Writing at 0x00038000... (78 %)
> Writing at 0x0003c000... (84 %)
> Writing at 0x00040000... (89 %)
> Writing at 0x00044000... (94 %)
> Writing at 0x00048000... (100 %)
> Wrote 437296 bytes (308387 compressed) at 0x00000000 in 7.0 seconds (effective 496.2 kbit/s)...
> Hash of data verified.
> 
> Leaving...
> Hard resetting via RTS pin...

- Windows 10 on local and VM. RD client is standard Microsoft RD (integrated one) and also Remote desktop manager (https://remotedesktopmanager.com/).

unmateria, 2020-10-04:
After some hours trying to upload via RDP (console working ok, but couldnt use esptool.exe to upload because of the GetOverlappedResult failed error). I tried to use some versions of Esptool and I ended recompiled version 8. After that, the same error... so I just used it with --trace option to see what happened and ESP32 was TX/RX correctly so I just tried to **press the boot button** on the PCB **all the time**, without waiting to see that ....-----....---- progress bar... and worked! :)

Try it first without replacing esptool.exe, and if it doesnt work, install try my compiled version from here [esptool compiled](https://mega.nz/file/x7BiQDxT#DysBATaY-qHvTjaWe2MC-bmKQBJZloj_y58hEqjAwJQ)