espressif/esptool #532: esptool stuck in getting chip_id

elacour, 2020-05-18:
# Operating system
Debian GNU/Linux 10 (buster)

# Python version
2.7.16

# What Chip
ESP8266-07

# What development board or other hardware is the chip attached to
Arduino uno (chip removed)

# Is anything else attached to the development board, except for the serial flasher connections?
Nothing else.

# Are you running esptool.py from an IDE such as Arduino or Eclipse?
No command prompt, esptool 2.8

# Full esptool.py command line that was run:
esptool --port /dev/ttyACM0  chip_id

# Full output from esptool.py
esptool.py v2.8
Serial port /dev/ttyACM0
Connecting...
Detecting chip type...

A fatal error occurred: Timed out waiting for packet header

# Do you have any other information from investigating this?
esp8266 is using it's default AT firmware and is responding to AT commands when GPIO0 is not grounded.

projectgus, 2020-05-19:
Hi @elacour,

It looks like the connection is failing shortly after esptool.py connects to the ROM bootloader.

There seem to be a lot of problems reported when reusing Arduino boards as USB/serial adapters, the Arduino USB/serial firmware seems like it does some things (to support Arduino auto-reset) that mean it doesn't behave like a normal USB/serial interface.

Please try `esptool --before no_reset --port /dev/ttyACM0 chip_id` to disable any setting of DTR & RTS while negotiating the link. This might help.

If that doesn't help, could you please post the output of `esptool --trace --before no_reset --port /dev/ttyACM0 chip_id`?

elacour, 2020-05-19:
Thanks, --before no_reset doesn't change anything. Here is the full trace:

$ esptool --trace --before no_reset --port /dev/ttyACM0 chip_id
esptool.py v2.8
Serial port /dev/ttyACM0
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.007 Read 1 bytes: c0
TRACE +0.000 Read 29 bytes: 
    0108020007071220 0000c0c001080200 | ....... ........
    070712200000c0c0 0108020007       | ... .........
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.004 Read 1 bytes: 07
TRACE +0.000 Read 46 bytes: 
    12200000c0c00108 0200070712200000 | . ........... ..
    c0c0010802000707 12200000c0c00108 | ......... ......
    0200070712200000 c0c001080200     | ..... ........
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.003 Read 1 bytes: 07
TRACE +0.000 Read 18 bytes: 
    0712200000c0c001 0802000707122000 | .. ........... .
    00c0                              | ..
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.000 Received full packet: 01080200070712200000

Detecting chip type...TRACE +0.000 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=78000060
TRACE +0.000 Write 14 bytes: c0000a04000000000078000060c0
TRACE +3.003 Timed out waiting for packet header

A fatal error occurred: Timed out waiting for packet header

projectgus, 2020-05-21:
Hi @elacour ,

Thanks for providing the trace.

The trace shows that the initial contact with the ESP8266 bootloader was successful, but then it sends the next command and never gets any response.

How is the ESP8266 powered? If it's powered from the Arduino Uno 3.3V pin then this is probably the problem, the internal regulator of an Uno is not powerful enough to power the ESP8266 reliably.

elacour, 2020-05-22:
Thanks, I just ordered a 5v->3.3v converter to use 5v output of the Arduino (up to 500mA) instead of the power limited 3.3v (50mA). Can you please keep this bug report open until I test this.

elacour, 2020-05-25:
I now uss a LM1117 to get decent 3.3v input, but unfortunatly I get the same results. Maybe the esp is damaged?

projectgus, 2020-05-26:
Hi @elacour,

It's possible, at this point I'd probably suspect the Arduino Uno usb/serial interface - there seem to be a lot of problems using these to replace a "standard" interface. If you an swap to a dedicated usb/serial interface then this may fix the problem.

Also, do you have a photo of your setup?

projectgus, 2020-08-28:
Hi @elacour ,

Did you make any progress with this problem?

Angus

elacour, 2020-08-28:
I'm sorry, I though I already updated this issue. It's working fine using the LM1117 directly connected to the esp, without arduino. But I'm really not able to use this esp as a wifi shield for arduino, it's not stable enough. I ordered an arduino wifi rev2 wich is rock stable. I will probably recycle this esp to use it standalone. Thanks for your help.

radimkarnis, 2020-12-08:
I’m closing this issue because it has been inactive for a few months and it seems like relevant anymore. Please reopen if you still encounter this issue.

closed 2020-12-08 by radimkarnis