espressif/esptool #534: Error: Timed out waiting for packet header

Devolick, 2020-06-07:
### Operating system Windows 10
### Python version v 3.8.3
### What Chip ESP-8266
### What development board WeMos D1R2 V2.1.0

Hello. I had buy this board and that is my first board. I had start play with that from Arduino IDE & VsCode and everything worked fine with blink project and with my code too. After I connect DHT22/AM2302 it was work, but I think I had do wrong Pin connection and it work and upload code. In one moment I got: 'Timed out waiting for packet header'. Perhaps burned out?

I tried ERASE and RESET(hold button and connect..). I tried python **esptool** to reset. It's always comes with this error. 

Thank you.

DuncanWilliamGibbons, 2020-06-09:
I'm getting the same, but with a nodemcu 8266

projectgus, 2020-06-10:
Hi @Devolick ,

Is anything connected to the WeMos board now? If you remove the connections one at a time, does flashing start to succeed?

Note in particular that GPIOs 6-11 are connected to the internal flash chip, and usually can't be connected to anything else.

Angus

radimkarnis, 2020-12-09:
Hello,
I’m closing this issue because it has been inactive for a few months and it seems like it's not relevant anymore. Please reopen if further discussion is needed.

closed 2020-12-09 by radimkarnis