espressif/esptool #536: Error uploading

DuncanWilliamGibbons, 2020-06-10:
Windows 10

3.8.3

nodeMCU Amica 8266

LED to GND and D7

Arduino ESP8266 IDE version 2.6.3

`esptool.py v2.6
Serial port COM4
Traceback (most recent call last):
  File "C:\Users\Duncan William\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3/tools/upload.py", line 65, in <module>
    esptool.main(cmdline)
  File "C:/Users/Duncan William/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/esptool\esptool.py", line 2652, in main
    esp = chip_class(each_port, initial_baud, args.trace)
  File "C:/Users/Duncan William/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/esptool\esptool.py", line 222, in __init__
    self._port = serial.serial_for_url(port)
  File "C:/Users/Duncan William/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/pyserial\serial\__init__.py", line 88, in serial_for_url
    instance.open()
  File "C:/Users/Duncan William/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/pyserial\serial\serialwin32.py", line 78, in open
    self._reconfigure_port()
  File "C:/Users/Duncan William/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/pyserial\serial\serialwin32.py", line 222, in _reconfigure_port
    'Original message: {!r}'.format(ctypes.WinError()))
serial.serialutil.SerialException: Cannot configure port, something went wrong. Original message: PermissionError(13, 'A device attached to the system is not functioning.', None, 31)
serial.serialutil.SerialException: Cannot configure port, something went wrong. Original message: PermissionError(13, 'A device attached to the system is not functioning.', None, 31)
`

projectgus, 2020-06-12:
Hi @DuncanWilliamGibbons,

The operating system is telling esptool that it can't open the serial port:

> serial.serialutil.SerialException: Cannot configure port, something went wrong. Original message: PermissionError(13, 'A device attached to the system is not functioning.', None, 31) 

There probably isn't anything we can do from the esptool side to fix this.

* Is the COM port correct and matches the device's COM port in Device Manager? If this doesn't match, this might be an easy fix (change to the correct  COM port).
* Otherwise this may be a driver issue, or a permissions issue for running Arduino IDE in Windows. Checking in Device Manager is a good first step, it may give a clue if the USB device is not functioning correctly.

DuncanWilliamGibbons, 2020-06-12:
Thanks for the tips

Ive tried to install the other CH341 driver but device manager keeps reverting to the CP210 by silicon labs.

projectgus, 2020-06-12:
These are two different USB/serial chips from different vendors. So if Device Manager says it's a CP2110, then this is likely the correct driver to use.

radimkarnis, 2020-12-09:
Hello @DuncanWilliamGibbons,
I’m closing this issue because it has been inactive for a few months and it seems like it's not relevant anymore. Please reopen if further discussion is needed.

closed 2020-12-09 by radimkarnis