espressif/esptool #538: ESP8266 issue with writing firmware

stmark707, 2020-06-12:
Hey there, the operating system I am using is windows 10
Python Version 3.7.6
I brought this chip from HiLetGo amazon account, the chip's full name is:
"HiLetgo 3pcs ESP8266 NodeMCU CP2102 ESP-12E Internet WiFi Development Board Open Source Serial 
Wireless Module Works Great for Arduino IDE/Micropython"

No other hardware is attached to the board at the moment, the hardware that is
going to be attached will be kind of experimental so I don't know how likely
it is that I'll be able to get outside assistance.
I am using VSCODE terminal that executes python (Will attach photo)

I am having a problem with some of the commands but the main one I care about
is writing the firmware, I am able to Connect and recieve information like
Chip_ID: 
Features: Wifi
Mac ID:
Manufacturing:
Device:
Detected flash size:
"Hard Resetting via RTS pin..."

I also added a input function because I could never tell what was successful, 
Now I know that if the program is ran successfully it waits for me to hit enter
if it is not successful then it just blinks through like it never ran.

basically what I need is the correct command format to write_flash
examples of things I have tried are:
esptool.py --port /dev/ttyUSB0 --baud 460800 write_flash --flash_size=detect 0 esp8266-20170108-v1.8.7.bin
and 
esptool.py --port /dev/ttyUSB0 --baud 460800 write_flash --flash_size=detect -fm dio 0 esp8266-20170108-v1.8.7.bin

it is always to no avail. Thank you for any help that you may provide. 

[image]

----------------------------- Delete above -----------------------------

# Operating system

# Python version

Can run `python -V` to check this.

# What Chip

(For example: ESP8266, ESP32-PICO-D4, ESP32-WROOM32 module, etc)

# What development board or other hardware is the chip attached to

(For example: DevKitC, NodeMCU board, plain module on breadboard, etc)

If your hardware is custom or unusual, please attach a photo to the issue.

# Is anything else attached to the development board, except for the serial flasher connections?

Example: GPIO 18 & 19 are connected to I2C devices.

# Are you running esptool.py from an IDE such as Arduino or Eclipse?

Example: No IDE, Windows command prompt

Example 2: Arduino ESP8266 IDE version 2.5.77

# Full esptool.py command line that was run:

esptool.py -p COM999 write_flash 0x0 bootloader.bin

# Full output from esptool.py

(Please copy and paste all lines of output here)

# Do you have any other information from investigating this?

Example: The command succeeds 1 in every 3 tries

# Is there any other information you can think of which will help us reproduce this problem?

Example: I also have a Windows 95 PC, esptool command works correctly on this PC.

projectgus, 2020-06-15:
Hi @stmark707 ,

Sorry, I don't really understand what you're asking.

* The command lines you posted as text both seem to be correct esptool command lines. The only problem is that the `--port /dev/ttyUSB0` argument you posted is a Linux port name. On Windows this would be a COM port.
* The screenshot shows the terminal in VSCode running commands were `esptool.py --chip esp8266 -p COM3 write_flash` and `esptool.py --chip esp8266 -p COM3 flash_id`. The first command would be expected to produce an error because no file names or addresses are specified to write to flash. The second command is working and the output is shown in the screenshot.

Is the problem that you don't know what files to write to flash? esptool doesn't know this by itself. Every development environment or ESP8266 SDK (including Arduino) generates different binary files, esptool just writes the files to the flash chip. You need to look at whatever SDK or environment that you're using to create and compile your firmware in order to find out what/where the binaries are and where to flash them.

stmark707, 2020-06-15:
Thank you for commenting, I tried several different variations of trying to flash the esp8266 to the point where i started doing different things just to see if anything would change. I will try again and let you know. I have the flash files downloaded into a downloads, i have tried things like \Downloads\esp8266-20191220-v1.12.bin even when i provide the location i still cannot get it to flash. I just keep trying different combinations but none of them see to work because when I use putty, I just get a bunch of gibberish. I will update more combinations soon, maybe you can see something i do not.

stmark707, 2020-06-16:
[image]

This is an example of the things I have tried, I cannot get this firmware to flash, I am asking for help for the right combination of what to write to get it flashed

stmark707, 2020-06-16:
Here are some more examples of what I have tried. Nothing seems to write the firmware to this device :( but I know the device is connected and functional because I can do other commands and the execution page stays until I hit enter.

[image]

stmark707, 2020-06-16:
Here is just the command flash_id, Just thought it would help.

[image]

stmark707, 2020-06-16:
Here is the command read_flash_status.

[image]

stmark707, 2020-06-16:
I think i may have figured it out! my file was .elf instead of .bin! 

[image]

stmark707, 2020-06-16:
Nope, i have gotten further but it is still not right! Sorry for the comments I think it is important to document struggle to help other people if they run into similar issues. 

[image]

stmark707, 2020-06-16:
I have finally figured it out. It all had to do with file location, once i moved the file under site packages in vs code it was able to find the firmware and give it a clean install. thanks for the effort at helping me!
[image]

closed 2020-06-16 by stmark707