espressif/esptool #539: esptool.FatalError: Timed out waiting for packet header

Devi416, 2020-06-13:
i would like to uploading a simple example code in my NODEMCU module but i got an error like : 

    0000010000000400 0000c0           | ...........
TRACE +0.078 Read 1 bytes: c0
TRACE +0.000 Read 1 bytes: 01
TRACE +0.000 Read 8 bytes: 0702009401600000
TRACE +0.000 Read 1 bytes: 00
TRACE +0.000 Read 1 bytes: c0
Traceback (most recent call last):
TRACE +0.000 Received full packet: 01070200940160000000
  File "C:\Users\admin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2/tools/upload.py", line 25, in <module>
Running stub...
    esptool.main(fakeargs)
TRACE +0.109 command op=0x06 data len=8 wait_response=1 timeout=0.050 data=0000000004e01040
  File "C:/Users/admin/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool\esptool.py", line 2694, in main
TRACE +0.000 Write 18 bytes: 
    detect_flash_size(esp, args)
    c000060800000000 000000000004e010 | ................
  File "C:/Users/admin/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool\esptool.py", line 2073, in detect_flash_size
    40c0                              | @.
    flash_id = esp.flash_id()
TRACE +0.000 Read 1 bytes: c0
  File "C:/Users/admin/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool\esptool.py", line 567, in flash_id
TRACE +0.000 Read 1 bytes: 01
    return self.run_spiflash_command(SPIFLASH_RDID, b"", 24)
TRACE +0.000 Read 10 bytes: 060200940160000000c0
  File "C:/Users/admin/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool\esptool.py", line 812, in run_spiflash_command
TRACE +0.000 Received full packet: 01060200940160000000
    old_spi_usr = self.read_reg(SPI_USR_REG)
TRACE +0.047 Read 6 bytes: c04f484149c0
  File "C:/Users/admin/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool\esptool.py", line 475, in read_reg
TRACE +0.000 Received full packet: 4f484149
    val, data = self.command(self.ESP_READ_REG, struct.pack('<I', addr))
Stub running...
  File "C:/Users/admin/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool\esptool.py", line 332, in command
Configuring flash size...
    p = self.read()
TRACE +0.000 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=1c020060
  File "C:/Users/admin/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool\esptool.py", line 277, in read
TRACE +0.000 Write 14 bytes: c0000a0400000000001c020060c0
    return next(self._slip_reader)
TRACE +3.011 Timed out waiting for packet header
  File "C:/Users/admin/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool\esptool.py", line 1877, in slip_reader
    raise FatalError("Timed out waiting for packet %s" % waiting_for)
esptool.FatalError: Timed out waiting for packet header
esptool.FatalError: Timed out waiting for packet header

projectgus, 2020-06-17:
Hi @Devi416 ,

This looks like a hardware issue of some kind not a problem with esptool.

Is anything connected to your dev board? Try removing it. Especially if connected to GPIOs 6-11, as these are shared with the internal flash chip and usually can't be used for other purposes.

If removing everything doesn't change the output then [suggest looking at the troubleshooting steps for power here](https://github.com/espressif/esptool#insufficient-power) or try swapping USB cables, USB ports, etc.

deepesh101, 2020-07-01:
I am also getting the same issue with my NodeMCU while uploading the code.
But in my case, the simple example codes like Blink, etc. are uploading easily but as the code becomes bigger and complex, I receive the same error.
The below is the list of things that I tried but didn't work:
1.  Keep pressed the Flash button while uploading the code.
2.  Degrading to a lower version of Esp8266 library.
3.  Removing all the GPIO and Power Pins, etc. while uploading the code.

My code looks like as below:

```C
#include <MAX30100_PulseOximeter.h>
#include<FirebaseESP8266.h>
#include<ESP8266WiFi.h>
#include<Wire.h>
/////////////////////Define Pins Here///////////////
const int MPU6050pin=D0;
const int MAX30100pin=D3;
////////////////////////////////////////////////////
#define FIREBASE_HOST "//my host"                          // database URL 
#define FIREBASE_AUTH "//my auth"                  // secret key
#define WIFI_SSID "//my ssd"                                                  //Wifi Name
#define WIFI_PASSWORD "//my  wifi password"          //Wifi Password
#define REPORTING_PERIOD_MS 1000
/////////////////////////////////////////////////////
/////////////////////Define Sensors Here//////////////
const int MPU_addr=0x68;  // I2C address of the MPU-6050
//////////////////////////////////////////////////////
////////////////////Define Variables Here/////////////
float AcX,AcY,AcZ,Tmp,GyX,GyY,GyZ;
FirebaseData firebaseData;
int MPU6050state,MAX30100state;
boolean MPU6050flag,MAX30100flag;
PulseOximeter pox;
uint32_t tsLastReport = 0;
int MAX30100firstTime=0;
///////////////////////////////////////////////////////
void setup() {
Serial.begin(115200);
WiFi.begin(WIFI_SSID,WIFI_PASSWORD);                                             //Connecting to Wifi Network
Firebase.begin(FIREBASE_HOST,FIREBASE_AUTH);
//////////////////Set Up Pins Here///////////////
pinMode(MPU6050pin,OUTPUT);
pinMode(MAX30100pin,OUTPUT);
/////////////////////////////////////////////////
/////////////////Initialize all Sensors to OFF///////
digitalWrite(MPU6050pin,LOW);
digitalWrite(MAX30100pin,LOW);
////////////////////////////////////////////////////
}
void onBeatDetected()
{
    Serial.println("Beat!");
}
void UploadMPUToFirebase(){
Firebase.setFloat(firebaseData,"/Readings/MPU6050/RoomTemperature",(Tmp/340.00)+36.53);
Firebase.setFloat(firebaseData,"/Readings/MPU6050/Acceleration/0",AcX/16384);
Firebase.setFloat(firebaseData,"/Readings/MPU6050/Acceleration/1",AcY/16384);
Firebase.setFloat(firebaseData,"/Readings/MPU6050/Acceleration/2",AcZ/16384);
Firebase.setFloat(firebaseData,"/Readings/MPU6050/Gyro/0",GyX/131);
Firebase.setFloat(firebaseData,"/Readings/MPU6050/Gyro/1",GyY/131);
Firebase.setFloat(firebaseData,"/Readings/MPU6050/Gyro/2",GyZ/131);
}
void UploadMAXtoFirebase(){
}
void MPU6050Data(){
Wire.begin();
Wire.beginTransmission(MPU_addr);
Wire.write(0x6B);  // PWR_MGMT_1 register
Wire.write(0);     // set to zero (wakes up the MPU-6050)
Wire.endTransmission(true);
[... 132 lines cut ...]
Writing at 0x00018000... (31 %)Traceback (most recent call last):
  File "C:\Users\Prashang\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.1/tools/upload.py", line 65, in <module>
    esptool.main(cmdline)
  File "C:/Users/Prashang/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.7.1/tools/esptool\esptool.py", line 2938, in main
    operation_func(esp, args)
  File "C:/Users/Prashang/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.7.1/tools/esptool\esptool.py", line 2374, in write_flash
    esp.flash_defl_block(block, seq, timeout=DEFAULT_TIMEOUT * ratio * 2)
  File "C:/Users/Prashang/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.7.1/tools/esptool\esptool.py", line 104, in inner
    return func(*args, **kwargs)
  File "C:/Users/Prashang/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.7.1/tools/esptool\esptool.py", line 672, in flash_defl_block
    self.ESP_FLASH_DEFL_DATA, struct.pack('<IIII', len(data), seq, 0, 0) + data, self.checksum(data), timeout=timeout)
  File "C:/Users/Prashang/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.7.1/tools/esptool\esptool.py", line 369, in check_command
    val, data = self.command(op, data, chk, timeout=timeout)
  File "C:/Users/Prashang/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.7.1/tools/esptool\esptool.py", line 347, in command
    p = self.read()
  File "C:/Users/Prashang/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.7.1/tools/esptool\esptool.py", line 292, in read
    return next(self._slip_reader)
  File "C:/Users/Prashang/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.7.1/tools/esptool\esptool.py", line 2045, in slip_reader
    raise FatalError("Timed out waiting for packet %s" % waiting_for)
esptool.FatalError: Timed out waiting for packet header
esptool.FatalError: Timed out waiting for packet header
```

Please help me solve this issue.
Thank you.

projectgus, 2020-07-02:
Hi @deepesh101 ,

I have a couple of questions to try and pin this down:

* Does the upload fail in the same place each time (in terms of how much data is already written), or in a different place?
* If you reduce the flashing baud rate to the minimum that Arduino allows, does it still fail and does it fail in the same place?
* Would it be possible for you to replace any private data in the firmware and then send me the .bin file that is failing to flash?

deepesh101, 2020-07-02:
Thanks for your reply @projectgus 
NodeMCU does work for sometimes like I tried today (the same code as above) and it worked but soon the same errors started appearing.
I tried changing the upload speed as well but that didn't work.
And I am new to all this so I don't know how to get .bin file which you are requesting. So, please tell me how can I get that so that I can send it to you.

projectgus, 2020-08-28:
Hi @deepesh101 ,

Sorry I didn't see this question when you asked.

You can get the .bin file by choosing Sketch -> Export Compiled Binary in the Arduino IDE.

Angus

radimkarnis, 2020-12-11:
Hello @deepesh101,

have you been able to look into this? 

Radim

radimkarnis, 2020-12-18:
I am closing this issue because it seems like it is not relevant anymore. Please reopen if you think further discussion is needed.

closed 2020-12-18 by radimkarnis