espressif/esptool #540: MacOS Big Sur dlsym(RTLD_DEFAULT, kIOMasterPortDefault): symbol not found

jeksys, 2020-06-25:
* Operating system: MacOS Big Sur
* Python version: Python 3.8.1

# Full esptool.py command line as run:
`esptool.py chip_id`

# Full output from esptool.py
```
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.8/bin/esptool.py", line 57, in <module>
    import serial.tools.list_ports as list_ports
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/serial/tools/list_ports.py", line 29, in <module>
    from serial.tools.list_ports_posix import comports
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/serial/tools/list_ports_posix.py", line 31, in <module>
    from serial.tools.list_ports_osx import comports
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/serial/tools/list_ports_osx.py", line 32, in <module>
    kIOMasterPortDefault = ctypes.c_void_p.in_dll(iokit, "kIOMasterPortDefault")
ValueError: dlsym(RTLD_DEFAULT, kIOMasterPortDefault): symbol not found
```

# What is the expected behaviour?
Connected device info

# Do you have any other information from investigating this?
I worked before the MacOS upgrade

# Is there any other information you can think of which will help us reproduce this problem?

SwiftySquirrel, 2020-06-25:
Same issue here!

projectgus, 2020-06-26:
Hi @jeksys @SwiftySquirrel ,

Thanks for letting us know about this.

esptool relies on [pyserial](https://github.com/pyserial/pyserial/) for serial port functionality, and this is where the exception is coming from. Ultimately whatever fix is required for MacOS Big Sur support will need to be applied there and then esptool will hopefully start working again.

Until then, the best we can do in esptool is apply a workaround so it's still possible to use esptool with a specific port specified on the command line, in the case that importing this utility function fails.

jeksys, 2020-06-26:
@projectgus 
Thank you for the suggestion. I think the additional parameter will work for some cases. 
Though I think Arduio studio will still fail.
When do you think it is possible to make a release with a fix? 

I reported the issue to pyserial
Though I think Arduio studio will still fail.

ehagerty, 2020-06-26:
same issue here.  forgive me if I'm being thick, but just putting --port /dev/tty.usbserial-xxxxxxxx flash-id for example gives the same error.  did I misunderstand the suggestion of forcing the port parameter?  Thanks so much, I should know better than to upgrade my primary machine!

projectgus, 2020-06-27:
Yes, sorry I didn't explain clearly. The workaround is a potential one, it still involves a code change in esptool.

As a very quick hack you can try commenting this line in esptool.py:

```
  File "/Library/Frameworks/Python.framework/Versions/3.8/bin/esptool.py", line 57, in <module>
    import serial.tools.list_ports as list_ports
```

... and then always use the `--port` argument

...

I'm trying to get my hands on a Big Sur Preview install to do some more debugging.

xrolfex, 2020-06-27:
@projectgus I can confirm that if you comment out the try: import block per your suggestion above it does work on Big Sur.

SwiftySquirrel, 2020-06-28:
@xrolfex It doesn't for me...

p-vernaeckt, 2020-06-28:
Same issue here when using `esptool` through PlatformIO:
[image]

xrolfex, 2020-06-28:
@p-vernaeckt Are you explicitly using the —port option after commenting out the four lines?

ehagerty, 2020-06-30:
Hi, sorry for the delay, can confirm:

OS11 Beta (20A4299v)
pyenv 1.2.19 
pyenv-virtualenv 1.1.5
python 3.8.2
micropython esp32-idf4-20191220-v1.12.bin
esp32 (huzzah32 esp32-wroom32)

commenting out lines 56-61 and using the explicit --port works.  Thank you!

JuanEL0, 2020-07-04:
Hi, I have the same issue on my mac with BigSur.

I'm using Arduino IDE and when I try to verify my code, it shows the same message. 

I am totally new in this world and don't know how to configure the port from the Arduino IDE so I can not verify and load my code into my ESP8266.

If somebody has any idea about how to do it, I would appreciate it.

jeksys, 2020-07-08:
The suggested fix worked for me.

1. Comment below lines in the file "/Library/Frameworks/Python.framework/Versions/3.8/bin/esptool.py"
```
#try:
#    import serial.tools.list_ports as list_ports
#except ImportError:
#    print("The installed version (%s) of pyserial appears to be too old for esptool.py (Python interpreter %s). "
#          "Check the README for installation instructions." % (sys.VERSION, sys.executable))
#    raise
```
2. List all available ports
`ls /dev/tty.*`

3. Use `esptool.py` with --port command
`esptool.py --port /dev/tty.usbserial-01EED907 --baud 921600 erase_flash`

4. Update `esptool.py` or env variable with a default port so `esptool.py` can be called without a parameter.
TODO

hpwit, 2020-07-26:
A have a question which is linked to that I have python 3.7.7 installed and that is my current python but esptools via Arduino keep on trying to get to 2.7
  File "esptool.py", line 57, in <module>
  File "/Library/Python/2.7/site-packages/PyInstaller/loader/pyimod03_importers.py", line 389, in load_module
how to change that
i have reinstalled Arduino and also es-tools

0x77dev, 2020-07-29:
Helped for me:
```shell
brew install udunits 
```

ehagerty, 2020-08-20:
NB, in case anyone is interested, this works for the v3.0-dev version of esptool.py in case you are using that with your s2 modules...

acmdorio, 2020-08-25:
> Yes, sorry I didn't explain clearly. The workaround is a potential one, it still involves a code change in esptool.
>
> As a very quick hack you can try commenting this line in esptool.py:
>
> ```
>   File [...]

@projectgus  Thanks, it works for me on BS 11.0 Beta (20A5354i)! By the way, a big thanks for your teamwork on this tool!
Maybe you should put this in esptool.py code as a test plus a warning before using the line 57, so other users barely note this issue. And brew update, then upgrade keep things working flawlessly. Just a humble idea. ;)

projectgus, 2020-08-27:
Commit cb2060c is a workaround for this issue, so that esptool.py can run (without being able to detect the port). If you pull the latest master branch version of esptool.py (or follow the instructions in the README to install the "development version" via GitHub) then no more modifications of the code should be required

It looks like the actual fix will come in a Python update once this issue is resolved: https://bugs.python.org/issue41100

Will keep this issue open, at least until the upstream issue is resolved.

R0dri, 2020-08-27:
@projectgus Thankyou for the workaround. I can confirm `esptool.py chip_id` now catches the error instead of an abrupt termination of the process. 

I've been trying to implement this to use arduino-cli or Arduinos own IDE with no luck. Inspecting the JSON board definition files i found arduino tries to compile the code not with python+esptool.py but with a binary stored in $arduino_packages/esp32/esptool_py/2.6.1/esptool which I guess its a compilation out of esptool.py.

I have never used esptool standalone and I don't know where to go next. Of course, I know I could wait for the actual python fix and keep using my raspberry pi for now. But I would love to learn a little more in the area. What would you recommend?

projectgus, 2020-08-27:
@R0dri I'm not actually familiar with arduino-esp32 enoug to give you an exact answer. However if you can get esptool.py master to run locally (by following the steps in the README for a development install) then you can probably update the Arduino platform.txt file here to specify a new path and command to run:
https://github.com/espressif/arduino-esp32/blob/ed96d2a1b70ea02d925435d06b71617bd451ffec/platform.txt#L6

exrector, 2020-09-04:
Hey. I tried to apply a workaround and I got an error. What advice can you give? 😭

`bash-3.2$ git clone https://github.com/espressif/esptool.git
xcrun: error: invalid active developer path (/Library/Developer/CommandLineTools), missing xcrun at: /Library/Developer/CommandLineTools/usr/bin/xcrun
bash-3.2$ `

petykowski, 2020-09-04:
> Hey. I tried to apply a workaround and I got an error. What advice can you give? 😭
>
>
>
> `bash-3.2$ git clone https://github.com/espressif/esptool.git
>
> xcrun: error: invalid active developer path [...]

This message is unrelated to the fix provided by Expressif. This error indicates that you need to update to the latest Xcode developer tools build.

exrector, 2020-09-04:
> Xcode
Thank you. But I have checked the versions and I have the latest 11.7.
Big Sur beta 11.0 (20A5364e)
Any variants? 😇

petykowski, 2020-09-05:
@exrector I am running the same configuration as described in your comment. Funny enough, I was experiencing error when attempting to commit to an unrelated repo. To resolve the issue, you should download the Apple Command Line Tools via the [Developer Downloads](url) portal, or via the direct link to the latest Xcode 12 beta 5 tools [here](https://download.developer.apple.com/Developer_Tools/Command_Line_Tools_for_Xcode_12_beta_5/Command_Line_Tools_for_Xcode_12_beta_5.dmg).

AndreasRavn, 2020-09-08:
> I've been trying to implement this to use arduino-cli or Arduinos own IDE with no luck. Inspecting the JSON board definition files i found arduino tries to compile the code not with [...]

I changed line 6 and 7 in platform.txt from 
tools.esptool_py.path={runtime.platform.path}/tools/esptool
tools.esptool_py.cmd=esptool

to
tools.esptool_py.path={runtime.platform.path}/tools
tools.esptool_py.cmd=esptool.py

that did the trick for me, the arduino IDE is now using esptool.py instead of the compiled binary in the subfolder.

philletourneau, 2020-10-09:
> Commit [cb2060c](https://github.com/espressif/esptool/commit/cb2060c0d768da38c9aa463e315e187a91f761f8) is a workaround for this issue, so that esptool.py can run (without being able to detect the port). If you pull the latest master branch version of esptool.py (or follow the instructions in the README to install in "Development Mode" via GitHub) then no more modifications of the code should be required
> 
> It looks like the actual fix will come in a Python update once this issue is resolved: https://bugs.python.org/issue41100
> 
> Will keep this issue open, at least until the upstream issue is resolved.

I've install the development version, but I'm still getting the same error...?

hpwit, 2020-10-11:
stupid question but how to change the version of python used for the esp32 I have the 3.7 installed and everything seems to work well on a terminal but my Arduino keeps on using 2.6 version for esptools

projectgus, 2020-10-11:
@philletourneau That's odd. Is the error exactly the same? Can you please check the commit of esptool.py you have? And also that the output of `which esptool.py` or (`where esptool.py`) is the development mode version and not another copy?

@hpwit This is determined by Arduino not esptool, and it depends on which OS you're using. One (hacky) option may be to edit the `#!/usr/bin/env python` line at the top of `esptool.py` to be `#!/usr/bin/env python3`.

hpwit, 2020-10-20:
@projectgus Hello I couldn't change it but I have found the workaround
1=> switch to version 2.7
2=>install pySerial
3=>do the modification in esptools.py documented earlier
4=>change in the platform.txt
```
tools.esptool_py.path={runtime.platform.path}/tools
tools.esptool_py.cmd=esptool.py
```
Thank to all of you guys

bfaliszek, 2020-10-20:
My temporary solution for macOS BigSur - ESP32 at ArduinoIDE problem:
1) copy `/Users/XXXXX/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/esptool.py` to `/Users/XXXXX/Library/Arduino15/packages/esp32/tools/esptool_py/2.6.1/esptool.py`
2) change in the platform.txt(`/Users/XXXXX/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/platform.txt`):
> `tools.esptool_py.cmd=esptool` >>> `tools.esptool_py.cmd=esptool.py`
3) save platform.txt, restart ArduinoIDE

lucasmarcelf, 2020-10-29:
> My temporary solution for macOS Big Sur(20A5395g) - ESP32 at ArduinoIDE problem:
> 
> 1. copy `/Users/XXXXX/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/esptool.py` to `/Users/XXXXX/Library/Arduino15/packages/esp32/tools/esptool_py/2.6.1/esptool.py`
> 2. change in the platform.txt(`/Users/XXXXX/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/platform.txt`):
> 
> > `tools.esptool_py.cmd=esptool` >>> `tools.esptool_py.cmd=esptool.py`
> 
> 1. save platform.txt, restart ArduinoIDE

I have this error: 
fork/exec /Library/Arduino15/packages/esp32/tools/esptool_py/2.6.1/esptool.py: permission denied
Error compiling to the board ESP32 Dev Module

bfaliszek, 2020-10-29:
@lucasmarcelf maybe try:
`chmod +x /Users/XXXXX/Library/Arduino15/packages/esp32/tools/esptool_py/2.6.1/esptool.py` at Terminal

R0dri, 2020-10-29:
> @lucasmarcelf maybe try:
> `chmod +x /Users/XXXXX/Library/Arduino15/packages/esp32/tools/esptool_py/2.6.1/esptool.py` at Terminal

Yes, I had to do `chmod 755` for it to work. Did not try `chmod +x`. I did not fully tested it, but I confirm i don't have any compilation errors. I came close to @bfaliszek solution earlier thanks to @projectgus but for some reason it was not working.

aungthawdar, 2020-11-12:
> A have a question which is linked to that I have python 3.7.7 installed and that is my current python but esptools via Arduino keep on trying to get to 2.7
> File "esptool.py", line 57, in
> File "/Library/Python/2.7/site-packages/PyInstaller/loader/pyimod03_importers.py", line 389, in load_module
> how to change that
> i have reinstalled Arduino and also es-tools

> Hi, I have the same issue on my mac with BigSur.
>
> I'm using Arduino IDE and when I try to verify my code, it shows the same message.
>
> I am totally new in this world and don't know how to configure [...]

> Hi, I have the same issue on my mac with BigSur.
>
> I'm using Arduino IDE and when I try to verify my code, it shows the same message.
>
> I am totally new in this world and don't know how to configure [...]

I've tested on BigSur Beta (20A5384c) ,  with python 2.7 version.
Read on.
https://github.com/espressif/arduino-esp32/issues/4408

hawtdawg, 2020-11-13:
> Commit [cb2060c](https://github.com/espressif/esptool/commit/cb2060c0d768da38c9aa463e315e187a91f761f8) is a workaround for this issue, so that esptool.py can run (without being able to detect the [...]

I replaced my esptool.py with the commit you linked but now I get this error in Arduino when compiling:
`
Traceback (most recent call last):
  File "/Users/Mate/Library/Arduino15/packages/esp32/tools/esptool_py/2.6.1/esptool.py", line 38, in <module>
    import serial
ImportError: No module named serial
exit status 1
Error compiling for board ESP32 Dev Module.
`

I did everything else, replacing the line in platform.txt, changing the permissions in terminal, etc... I even tried commenting out the line 57 in esptool.py to no avail. I'd really love some help :(

aungthawdar, 2020-11-13:
> > Commit [cb2060c](https://github.com/espressif/esptool/commit/cb2060c0d768da38c9aa463e315e187a91f761f8) is a workaround for this issue, so that esptool.py can run (without being able to detect the port). If you pull the latest master branch version of esptool.py (or follow the instructions in the README to install in "Development Mode" via GitHub) then no more modifications of the code should be required
> > It looks like the actual fix will come in a Python update once this issue is resolved: https://bugs.python.org/issue41100
> > Will keep this issue open, at least until the upstream issue is resolved.
> 
> I replaced my esptool.py with the commit you linked but now I get this error in Arduino when compiling:
> `Traceback (most recent call last): File "/Users/Mate/Library/Arduino15/packages/esp32/tools/esptool_py/2.6.1/esptool.py", line 38, in <module> import serial ImportError: No module named serial exit status 1 Error compiling for board ESP32 Dev Module.`
> 
> I did everything else, replacing the line in platform.txt, changing the permissions in terminal, etc... I even tried commenting out the line 37 and 38 in esptool.py to no avail. I'd really love some help :(

I've tested on BigSur Beta (20A5384c) , with python 2.7 version.
Read on.
https://github.com/espressif/arduino-esp32/issues/4408

Elijahg, 2020-11-13:
Building a new version of Python from source with brew fixes it without having to modify the esptool.py: 

https://github.com/pyserial/pyserial/issues/509#issuecomment-726966076

brunohorta82, 2020-11-17:
> The suggested fix worked for me.
> 
> 1. Comment below lines in the file "/Library/Frameworks/Python.framework/Versions/3.8/bin/esptool.py"
> 
> ```
> #try:
> #    import serial.tools.list_ports as list_ports
> #except ImportError:
> #    print("The installed version (%s) of pyserial appears to be too old for esptool.py (Python interpreter %s). "
> #          "Check the README for installation instructions." % (sys.VERSION, sys.executable))
> #    raise
> ```
> 
> 1. List all available ports
>    `ls /dev/tty.*`
> 2. Use `esptool.py` with --port command
>    `esptool.py --port /dev/tty.usbserial-01EED907 --baud 921600 erase_flash`
> 3. Update `esptool.py` or env variable with a default port so `esptool.py` can be called without a parameter.
>    TODO

WORK FOR MIM ON PLATFORMIO

me-no-dev, 2020-11-17:
Could you please try the esptool binary from the zip? No need to edit platform.txt
[esptool.zip](https://github.com/espressif/esptool/files/5556525/esptool.zip)

me-no-dev, 2020-11-17:
problem is in the version of pyserial. Big Sur needs version 3.4. I have posted a recompiled binary for Mac [here](https://github.com/espressif/arduino-esp32/issues/4408#issuecomment-729246012). Please give it a go

stephech, 2020-11-17:
> problem is in the version of pyserial. Big Sur needs version 3.4. I have posted a recompiled binary for Mac [here](https://github.com/espressif/arduino-esp32/issues/4408#issuecomment-729246012). Please give it a go

This fixed the issue for me.  I put the this binary in my Library/Arduino15/packages/esp32/tools/esptool_py/3.0.0 directory

gretel, 2020-11-18:
```pip install -U --pre pyserial``` does make me happy on the big sur.

dobairoland, 2020-11-18:
Dear all, there are at least two PRs on the pyserial site regarding BigSur issues (https://github.com/pyserial/pyserial/pull/546, https://github.com/pyserial/pyserial/pull/542). Maybe you could check those out and report the results there. They are preparing the release of 3.5. If a proper fix is possible and would it make into that release then it would be great.

> problem is in the version of pyserial. Big Sur needs version 3.4. I have posted a recompiled binary for Mac here. Please give it a go

@me-no-dev Is in it by any chance anything worth submitting as PR to pyserial in top of the above two PRs?

me-no-dev, 2020-11-18:
@dobairoland I did not have to change anything, only updated pyserial to 3.4 and recompiled the binary (BTW this happens on an old OS X 10.9 for compatibility) then tried it on Big Sur and all went well. TBH did not try flashing, just compiled fine.

gretel, 2020-11-18:
@dobairoland mind my post

me-no-dev, 2020-11-18:
@gretel just `pip install pyserial` did it for me as well. But that is to get the esptool.py to work on Big Sur. The binary that is the other discussion is another story :) 

@dobairoland I tested upload as well. All good. So I guess 3.4 is fine if you do not list the ports.

gretel, 2020-11-18:
ah sure binaries.. nevermind. regards

me-no-dev, 2020-11-18:
@projectgus @dobairoland I am hitting reliably an issue that Arduino users reported. When trying to flash my Saola board (using external ESP-Prog), download is not able to start. Even though I see data back and fourth and entered Download manually, esptool still times out. Same ESP-Prog flashes another ESP32 based board no problem, and flashing S2 over CDC also works fine. Not sure how to debug this further.

machinegunjoeviterbo, 2020-11-18:
This steps worked out for me running macOS Big Sur (11.0.1) and Arduino 1.8.13
1. Install pyserial
	pip install -U --pre pyserial
2. Open:
	/Users/<user>/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/platform.txt
3. Change: 
	tools.esptool_py.cmd=esptool
	to:
	tools.esptool_py.cmd=esptool.py
4. Copy:
	/Users/<user>/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/esptool.py
	to:
	/Users/<user>/Library/Arduino15/packages/esp32/tools/esptool_py/2.6.1
5. Open:
	/Users/<user>/Library/Arduino15/packages/esp32/tools/esptool_py/2.6.1/esptool.py
	Replace first row with:
	#!/usr/bin/env python3

me-no-dev, 2020-11-19:
@machinegunjoeviterbo point was to just replace the binary from the zip above and not install pyserial or edit platform.txt
I am glad it worked for you, but it does not serve as proof if that binary that is supposed to go into Arduino actually works.

billvoss, 2020-11-21:
> `pip install -U --pre pyserial` does make me happy on the big sur.

I'm using platformIO, I had to use `pip3.8 install -U --pre pyserial` to use ~/.platformio/penv/bin/pip3.8 instead of /usr/local/bin/pip. Then it worked!

SeanZhao1, 2020-11-22:
> These steps worked out for me running macOS Big Sur (11.0.1) and Arduino 1.8.13
> 
> 1. Install pyserial
>    pip install -U --pre pyserial
> 2. Open:
>    /Users//Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/platform.txt
> 3. Change:
>    tools.esptool_py.cmd=esptool
>    to:
>    tools.esptool_py.cmd=esptool.py
> 4. Copy:
>    /Users//Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/esptool.py
>    to:
>    /Users//Library/Arduino15/packages/esp32/tools/esptool_py/2.6.1
> 5. Open:
>    /Users//Library/Arduino15/packages/esp32/tools/esptool_py/2.6.1/esptool.py
>    Replace first row with:
>    #!/usr/bin/env python3

after I did this I got another error:

fork/exec /Users/sean/Library/Arduino15/packages/esp32/tools/esptool_py/2.6.1/esptool.py: permission denied
Error compiling for board Adafruit ESP32 Feather.

what should I do then?

billvoss, 2020-11-22:
@SeanZhao1 
Try adding executable permission?
`chmod +x /Users/sean/Library/Arduino15/packages/esp32/tools/esptool_py/2.6.1/esptool.py`

BaiqiangZhao, 2020-11-22:
> @SeanZhao1
> Try adding executable permission?
> `chmod +x /Users/sean/Library/Arduino15/packages/esp32/tools/esptool_py/2.6.1/esptool.py`

that's is error message I received from arduino. it is not a command.

oLuciaZo, 2020-11-22:
> These steps worked out for me running macOS Big Sur (11.0.1) and Arduino 1.8.13
>
> 1. Install pyserial
>    pip install -U --pre pyserial
> 2. Open: [...]

Solved my problem

SeanZhao1, 2020-11-23:
> @SeanZhao1
> Try adding executable permission?
> `chmod +x /Users/sean/Library/Arduino15/packages/esp32/tools/esptool_py/2.6.1/esptool.py`

sorry, it is actually working, but I received another error.

Traceback (most recent call last):
  File "/Users/sean/Library/Arduino15/packages/esp32/tools/esptool_py/2.6.1/esptool.py", line 37, in <module>
    import serial
ModuleNotFoundError: No module named 'serial'
exit status 1
Error compiling for board Adafruit ESP32 Feather.

any idea?

SeanZhao1, 2020-11-23:
> > @SeanZhao1
> > Try adding executable permission?
> > `chmod +x /Users/sean/Library/Arduino15/packages/esp32/tools/esptool_py/2.6.1/esptool.py`
> 
> sorry, it is actually working, but I received another error.
> 
> Traceback (most recent call last):
> File "/Users/sean/Library/Arduino15/packages/esp32/tools/esptool_py/2.6.1/esptool.py", line 37, in 
> import serial
> ModuleNotFoundError: No module named 'serial'
> exit status 1
> Error compiling for board Adafruit ESP32 Feather.
> 
> any idea?

I finally figured out, I did have PySerial installed, but my default python version is 2.7.
Then I changed the default to python 3, and installed PySerial again.
Then I did those steps listed above. The problem solved!!
Thank you guys. love you

FredMeib, 2020-11-23:
I have the exactly same problem and I do not have any programming experiences. How can I change the "default" to python 3? I checked via Terminal the installed version of python: Version 3.8.2 is installed.

SeanZhao1, 2020-11-23:
> I have the exactly same problem and I do not have any programming experiences. How can I change the "default" to python 3? I checked via Terminal the installed version of python: Version 3.8.2 is installed.

if you enter ' python --version' and the output shows the version is 3, your default is python 3

risingtiger, 2020-11-24:
This may help others like me that fought this bloody thing for too long:

I followed the steps outlined by machinegunjoeviterbo but was getting an error saying the serial module could not be found. 

I realized that I had installed Command Line Tools, which installs a version of python3 (python 3.8.2), and that Arduino was using that library and therefore wasn't finding the pyserial module that I had installed (via pip) in my homebrew python install. 

To fix this I ran Command Line Tool's pip3:

`sudo /Library/Developer/CommandLineTools/usr/bin/pip3 install -U --pre pyserial`

Worked just fine after that

FredMeib, 2020-11-24:
Trying to follow your steps the call of sudo... requested a password. What kind of password is requested here?

[36 more replies]

closed 2020-12-09 by dobairoland