espressif/esptool #541: No Memory Map Defined for ESP32S2, Inherited ESP32 Memory Map Incorrect

rmspeers, 2020-06-30:
* Operating system: macOS
* Python version: (`python -V` to check this) 3.7

# Full esptool.py command line as run:
N/A, see code at https://github.com/espressif/esptool/blob/master/esptool.py#L1465

# Full output from esptool.py
N/A

# What is the expected behaviour?
Correct memory map exists to allow for image parsing to happen when `LoadFirmwareImage` is called

# Do you have any other information from investigating this?
I have tried patching the ESP32's memory map with the `IROM_MAP_START` and `DROM_MAP_START` constants from ESP32S2 however these have overlap with other regions (IRAM and IROM) and appear to not be the only differences.

# Is there any other information you can think of which will help us reproduce this problem?
Parse an ESP32S2 image, then try to use memory map to label regions. It says some code is IRAM and IROM due to the overlapping regions, for instance.

rmspeers, 2020-06-30:
Attempted to rewrite it from the data sheet but it yields very bad results - do **not** use this version, but it shows what I think is needed - a new memory map for that class:
```
        image.ROM_LOADER.MEMORY_MAP = [
            [0x3F000000, 0x3F400000, "DROM"],             #External Memory > Extenral Flash > R (ICache)
            [0x3F500000, 0x3FF80000, "EXTRAM_DATA"],      #External Memory > External SRAM > RW (DCache)
            [0x3FF9E000, 0x3FFA0000, "RTC_DRAM"],         #RTC FAST Memory
            [0x3FFA0000, 0x3FFB0000, "BYTE_ACCESSIBLE"],  #Internal ROM 1
            [0x3FFB0000, 0x3FFB8000, "IRAM"],             #Internal SRAM 0 first
            [0x3FFB8000, 0x40000000, "DRAM"],             #Internal SRAM 1 first
            [0x40000000, 0x40010000, "IROM"],             #Internal ROM 0
            [0x40010000, 0x40020000, "BYTE_ACCESSIBLE"],  #Internal ROM 1  ??? naming ???
            [0x40020000, 0x40028000, "IRAM"],             #Internal SRAM 0
            [0x40028000, 0x40070000, "DIRAM_DRAM"],       #Internal SRAM 1
            [0x40070000, 0x40072000, "RTC_IRAM"],         #RTC FAST Memory
            [0x40080000, 0x40800000, "IROM"],             #External Memory > Instruction ??
            [0x50000000, 0x50002000, "RTC_DATA"]]         #RTC SLOW Memory
```

KonstantinKondrashov, 2020-07-08:
Hi @rmspeers! Thanks for the report. I prepared a fix for this it will be merged ASAP.

closed 2020-07-13 by antmak