espressif/esptool #542: windows10下WSL安装Ubuntu搭建esp32编译环境串口无法连接和下载

zanzzd, 2020-07-04:
# Operating system：windows10 WSL ubuntu18.04LTS

# Python version: 2.7

# What Chip
esp32-s2-saola-1_v1.2核心板

# Are you running esptool.py from an IDE such as Arduino or Eclipse?

vs code 

# Full output from esptool.py

(Please copy and paste all lines of output here)

esp@DESKTOP-OO1I4GU:~/esp/esp-idf/examples/get-started/hello_world$ idf.py -p /dev/ttyS6 flash
Executing action: flash
Running ninja in directory /home/esp/esp/esp-idf/examples/get-started/hello_world/build
Executing "ninja flash"...
[1/4] Performing build step for 'bootloader'
ninja: no work to do.
[1/2] cd /home/esp/esp/esp-idf/components/esptool_py && /usr/bin/cmake -D IDF_PAT...llo_world/build" -P /home/esp/esp/esp-idf/components/esptool_py/run_esptool.cmake
esptool.py --chip esp32s2 -p /dev/ttyS6 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 2MB 0x8000 partition_table/partition-table.bin 0x1000 bootloader/bootloader.bin 0x10000 hello-world.bin
esptool.py v3.0-dev
Serial port /dev/ttyS6
Connecting........_____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to ESP32-S2: Invalid head of packet (0x66)
CMake Error at run_cmd.cmake:14 (message):
  esptool.py failed
Call Stack (most recent call first):
  run_esptool.cmake:21 (include)

FAILED: CMakeFiles/flash 
cd /home/esp/esp/esp-idf/components/esptool_py && /usr/bin/cmake -D IDF_PATH="/home/esp/esp/esp-idf" -D ESPTOOLPY="/home/esp/.espressif/python_env/idf4.2_py2.7_env/bin/python /home/esp/esp/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32s2" -D ESPTOOL_ARGS="--before=default_reset --after=hard_reset write_flash @flash_args" -D WORKING_DIRECTORY="/home/esp/esp/esp-idf/examples/get-started/hello_world/build" -P /home/esp/esp/esp-idf/components/esptool_py/run_esptool.cmake
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

# Do you have any other information from investigating this?

projectgus, 2020-08-28:
Hi @zanzzd ,

Thanks for being patient while someone got back to you.

Did you manage to resolve this issue? This may be a WSL issue, can you try running esptool from Windows directly?

Angus

radimkarnis, 2020-12-09:
Hello,
I’m closing this issue because it has been inactive for a few months and it seems like it's not relevant anymore. Please reopen if further discussion is needed.

closed 2020-12-09 by radimkarnis