espressif/esptool #543: A fatal error occurred: Timed out waiting for packet header

dejanmeznarc, 2020-07-13:
Hello!
I've been trying to upload to ESP for a whole day now. Sadly, without success.

**OS:** Fedora 31, tried on Windows too.
**Python:** 3.7
**Chip:** ESP8266
**Dev board:** ESP-12-F on costom made PCB (image below)

##### Is anything else attached to the development board, except for the serial flasher connections?
Voltage regulator and logic shifter. (see curcit below)
[image]

##### Are you running esptool.py from an IDE such as Arduino or Eclipse?
Same error on both Arduino IDE and CLI.

#### Command:
What I do:
1. Run command
2. Connect GPIO 0 to ground
2. Connect USB-TTL converver to ESP (tx, rx, 5V, gnd)
4. Break connection between GPIO 0 and ground
After that dots and _ stop showing and ESP tool starts to detect chip type.

```shell
$  esptool.py --port /dev/ttyUSB0 write_flash 0x0 ./esphello.ino.generic.bin
esptool.py v2.8
Serial port /dev/ttyUSB0
Connecting........_____.....
Detecting chip type...

A fatal error occurred: Timed out waiting for packet header
```

##### Do you have any other information from investigating this?
* AT commands work (when GPIO 0 isn't pulled to ground), so serial connection/logic shifter are ok.
* Arduino > Get board info shows VID and PID
* `esptool.py chip_id`returns same error
* Selected serial port is correct

#### What I already tried
* Pull-up reset pin
* capacitor between GND and 3v3 lines

[image]
[image]
[image]
Resistors, capacitors, U3 and U2 are in place. Everytig else is disconnected.

projectgus, 2020-07-13:
Hi @dejanmeznarc ,

Some things to try:

* Suggest connecting to the ESP8266 in the order 2,3,1,4 of the steps listed, although I don't think that's the problem.
* Add an argument like `esptool.py --baud 9600 ...` to sync and communicate at a much lower baud rate, and see if this works.
* Try running `esptool.py --trace ...` and see if anything is received back at all when detecting chip type, before it fails.
* Try running `esptool.py --chip esp8266 ...` to skip the detecting chip step. I think the next step fail instead, but it's one more data point.
* If you have an oscilloscope, look at the TX & RX signals on both sides of the level shifter and check they look clean.
* If you have an oscilloscope, look at the LDO output voltage and check it's stable while connecting.

dejanmeznarc, 2020-07-14:
Thaks for reply and suggestions @projectgus !

Specifying chip and slower baud produces exact same behaviour. I also tried other baud rates (slower and faster), no luck.

Messing with the order of steps, that I listed above, gives same error but diffrent trace:
* when I try steps in order 2, 3, 1, 4 - ESP doesnt send anything back
```bash
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.100 Timed out waiting for packet header
_TRACE +0.050 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.100 Timed out waiting for packet header
_TRACE +0.050 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.100 Timed out waiting for packet header
_

A fatal error occurred: Failed to connect to ESP8266: Timed out waiting for packet header
```
* when I try steps in order 1,2,3,4 - ESP sends something back, but still same error.
```bash
0000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.100 Timed out waiting for packet header
_TRACE +0.050 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.100 Timed out waiting for packet header
_TRACE +0.050 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes:                                    # Here I disconnected GPIO 0 to ground
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.082 Read 1 bytes: c0
TRACE +0.000 Read 31 bytes: 
    0108020007071220 0000c0c001080200 | ....... ........
    070712200000c0c0 01080200070712   | ... ...........
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.033 Read 1 bytes: 20
TRACE +0.000 Read 31 bytes: 
    0000c0c001080200 070712200000c0c0 | ........... ....
    0108020007071220 0000c0c0010802   | ....... .......
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.033 Read 1 bytes: 00
TRACE +0.000 Read 31 bytes: 
    070712200000c0c0 0108020007071220 | ... ........... 
    0000c0c001080200 070712200000c0   | ........... ...
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.000 Received full packet: 01080200070712200000

TRACE +0.000 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=5c00f03f
TRACE +0.000 Write 14 bytes: c0000a0400000000005c00f03fc0
TRACE +3.003 Timed out waiting for packet header

A fatal error occurred: Timed out waiting for packet header
```

I'll try to probe pins on scope now.

Again, thanks for your help.

projectgus, 2020-07-14:
That is all quite strange!

> when I try steps in order 2, 3, 1, 4 - ESP doesnt send anything back
> when I try steps in order 1,2,3,4 - ESP sends something back, but still same error.

I can't really explain why this makes a difference. When the ESP8266 powers on, it reads the GPIO0 pin to [determine the boot mode](https://github.com/espressif/esptool/wiki/ESP8266-Boot-Mode-Selection). The boot mode should stay the same until the next time the chip resets or is power cycled.

For this reason I suggested not starting esptool until after the chip is powered up in the UART download boot mode - at this point it should sit there indefinitely waiting for something to talk to it over serial.

> # Here I disconnected GPIO 0 to ground

For the same reason, releasing or not releasing GPIO0 shouldn't matter - you should be able to jumper this pin to ground permanently and still flash successfully (you just can't run the firmware successfully until you remove the jumper and let GPIO0 go high again.)

One possible explanation is the EN or RST pin is not pulled high consistently enough, so the chip is putting itself back into reset randomly. Suggest permanently connecting these to 3.3V externally if they're not already, at least until the problem is resolved.

Otherwise, the oscilloscope traces will hopefully shed some light.

dejanmeznarc, 2020-07-15:
There is 10K pullup to EN pin. I forgot to mention that I've tried to pull up RST with 10K too. But it didn't work so I removed it. Should I directly connect both RST and EN to 3.3V?

dejanmeznarc, 2020-07-15:
Finnaly got my scope back. 

I probed TX (blue) and RX (yellow) on ESP.
#### AT command
When ESP is not in flash mode.
[image]

#### ESP in flash mode
[image]

---
I checked with multimeter and there is no connection between GPIO 0 and Tx/Rx pins.

projectgus, 2020-08-28:
Hi @dejanmeznarc,

Sorry I didn't get back to you earlier.

One thing to check if you're still investigating this is that there is some serial output from the ESP8266 when you reset the chip into download mode (GPIO0 low). If you connect a serial monitor at 74880bps then you should see an indication of the boot mode as shown here:
https://github.com/espressif/esptool/wiki/ESP8266-Boot-ROM-Log

And can verify the correct boot mode based on this output.

You should also see this on the oscilloscope. If you don't see anything at all, the ESP8266 isn't coming out of reset correctly.

Kashyapkoshti, 2020-08-28:
rst:0x10 (RTCWDT_RTC_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 

I am getting this error in my esp32 devkit v1
How to solve it?
I am using arduino IDE for programming the esp32

s0170071, 2020-09-02:
I had a similar issue flashing a ESP32CAM with Arduino IDE 1.8.10. Turned out the problem was that the serial programming speed was set to 460800 which is way too fast for most FTDI Adapters. **Unfortunately, there is no option to set the baud rate manually any more**. I am using the ESP32 1.0.2 libraries.

Quick fix: grab the esptool.py command line from the output and paste it into a cmd/shell. There you can change the  --baud parameter. I set it to 115200 and that solved various programming errors.

Kashyapkoshti, 2020-09-03:
@s0170071 thanks for your comments, i never worked with cmd. So can you please give me step by step instructions? Your help will be appreciated.

s0170071, 2020-09-03:
Work with your IDE, I use Arduino IDE. After the failed upload you will find the command line that is used for uploading. Mine was: 

1. Work with your IDE, I use Arduino IDE. After the failed upload you will find the command line that is used for uploading in the output window. Mine was: 
`python /home/john/.arduino15/packages/esp32/tools/esptool_py/2.6.1/esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 460800 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0xe000 /home/john/.arduino15/packages/esp32/hardware/esp32/1.0.2/tools/partitions/boot_app0.bin 0x1000 /home/john/.arduino15/packages/esp32/hardware/esp32/1.0.2/tools/sdk/bin/bootloader_qio_80m.bin 0x10000 /tmp/arduino_build_713656/CameraWebServer.ino.bin 0x8000 /tmp/arduino_build_713656/CameraWebServer.ino.partitions.bin 
`

2. Grab that line and paste it into any shell/cmd window. On windows press the start button and type "cmd" to open a command window. On linux you most likely know where to find the console ;-)

3. Modify the baud parameter. I use 115200. 

4. Hit enter.

Two notes. 

- I found that the ESPCAM module consumes considerably more power than the ESP8266. Be aware of that. Especially if you use a opto-isolated programmer, they may not be able to supply enough current (>250mA). 
- The compiled binary sits in a temp folder. This folder name can change if you restart the IDE. So be sure to grab that line of code right after you compile.

projectgus, 2020-09-14:
@Kashyapkoshti The error you have is different to the error discussed elsewhere in this issue, and it is probably not a problem with esptool - it seems like the contents of the flash are invalid, or the ESP32 can't read from the flash chip. It's only an esptool problem if flashing failed and you got an error message when you tried to flash, as well. Suggest checking if you've accidentally connected some external hardware to dev board pins which are shared with the SPI flash chip (GPIOs 6-11), or maybe there is a hardware issue with your board. If you're still having problems, try asking on the https://esp32.com forums - please provide as many details as possible about the steps you are following, and any output printed by the tool(s).

@s0170071 

> Unfortunately, there is no option to set the baud rate manually any more. I am using the ESP32 1.0.2 libraries.

I'm glad you found a solution for this problem, but this should be possible to set this under Tools -> Upload Speed in Arduino. If this option is not visible or not working then suggest trying an update to the 1.0.4 ESP32 package release, and if it still doesn't work then please open an issue on the [Arduino repo](https://github.com/espressif/arduino-esp32/issues/).

radimkarnis, 2020-12-10:
Hello everyone,
I’m closing this issue because it has been inactive for a few months and it seems like it's not relevant anymore. Please reopen if further discussion is needed.

closed 2020-12-10 by radimkarnis