espressif/esptool #544: MX25L512 fails to work with some commands

dale6998, 2020-07-17:
# Operating system
Arch Linux

# Python version 
3.8.3

Can run `python -V` to check this.

# What Chip
ESP32DW0 
(For example: ESP8266, ESP32-PICO-D4, ESP32-WROOM32 module, etc)

# What development board or other hardware is the chip attached to
Custom PCB
The SPI FLASH Chip is a MX25L51245GZ2I-08G by Macronix and it's wired up the precise same way that it is in the ESP32-WROOM/WROVER (The pinouts  are the same).

# Is anything else attached to the development board, except for the serial flasher connections?

There are several high impedance output connections, a UART on pins 25, 27, 5, and 35, I2C on 21 and 22, a pulled up input on 33 and some ADC stuff on 36 and 39.

# Are you running esptool.py from an IDE such as Arduino or Eclipse?

No...using a shell.

Example 2: Arduino ESP8266 IDE version 2.5.77

# Full esptool.py command line that was run:

esptool -t erase_flash

# Full output from esptool.py

esptool.py v2.8
Serial port /dev/esp32_wroom
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.101 Timed out waiting for packet header
.TRACE +0.051 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.101 Timed out waiting for packet header
.TRACE +0.051 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.010 Read 1 bytes: c0
TRACE +0.000 Read 63 bytes: 
    0108040007122055 00000000c0c00108 | ...... U........
    0400071220550000 0000c0c001080400 | .... U..........
    0712205500000000 c0c0010804000712 | .. U............
    205500000000c0c0 01080400071220   |  U............ 
TRACE +0.000 Received full packet: 010804000712205500000000
TRACE +0.000 Received full packet: 010804000712205500000000
TRACE +0.000 Received full packet: 010804000712205500000000
TRACE +0.000 Received full packet: 010804000712205500000000
[... 524 lines cut ...]
TRACE +0.000 command op=0xd0 data len=0 wait_response=1 timeout=120.000 data=
TRACE +0.000 Write 10 bytes: c000d0000000000000c0
TRACE +0.004 Read 1 bytes: c0
TRACE +0.000 Read 8 bytes: 01d0020000000000
TRACE +120.098 Timed out waiting for packet content

A fatal error occurred: Timed out waiting for packet content

# Do you have any other information from investigating this?

I can write flash and see app work every time if and only if I select QOUT SPI Mode and if I bump up the offset of the partition table to 0x10000 from 0x08000.
The issues addressed by these changes are as follows:

If I don't bump up the partition table offset, esptool simply refuses to flash it, complaining about how 
there is an overwrite detected up against 0x08000 and that's that goodbye.  This seems odd because this wasn't an issue with the DevKitC's we were using, but then again, the flash chip is totally similar but different.  If I do flash the chip with 0x800000, the chip simply bootloops with a Guru Mediation error regarding trying to execute 0xffffffff as it's next instruction fetched from SPI-Flash.

Otherwise, I can run something like:
`esptool.py -p /dev/esp32_wroom -b 2000000 --before default_reset --after hard_reset write_flash --flash_mode qout --flash_freq 40m --flash_size 4MB 0x8000 build/partition_table/partition-table.bin 0xd000 build/ota_data_initial.bin 0x1000 build/bootloader/bootloader.bin 0x10000 build/application.bin
`
and it will flash and run and that's great!
But, when I try to do other jobs (like parttool.py which uses esptool.py for it's heavy lifiting, or the erase_flash subcommand of esptool itself), i get into the situation shown in the rlogs above.

# Is there any other information you can think of which will help us reproduce this problem?

The chip that is standing in for the flash found on the DevkitC's is MX25L51245GZ2I-08G and it's datasheet can be found here:  http://www.macronix.com/Lists/Datasheet/Attachments/7437/MX25L51245G,%203V,%20512Mb,%20v1.6.pdf

projectgus, 2020-08-27:
Hi @dale6998,

Thanks for being patient while someone got back to you.

The first thing to stress is that there is no universal standard for SPI flash chips. They mostly support the same commands, but there is a lot of difference in the details especially regarding quad I/O modes. If you want to know about guaranteed support for a particular flash chip in production then please contact Espressif Sales. But I know we've not tested or verified the chip you're using now.

> If I don't bump up the partition table offset, esptool simply refuses to flash it, complaining about how
there is an overwrite detected up against 0x08000 and that's that goodbye.

This part is a software issue, so it's not dependent on the flash chip communications at all. The bootloader.bin size depends on the bootloader configuration (log level, features enabled/disabled, flash mode, etc.) The bootloader.bin is flashed at offset 0x1000, and if the partition table is flashed at 0x8000 then the bootloader.bin cannot be larger than 0x7000 bytes. The solution is either to compile the bootloader to be smaller (disable some settings, lower the log level, etc) or to move the partition table to make more room.

I can't explain why this worked on the previous module, but my guess is the bootloader.bin file was not 100% the same in both cases.

> Erasing flash (this may take a while)...
> TRACE +0.000 command op=0xd0 data len=0 wait_response=1 timeout=120.000 data=
> TRACE +0.000 Write 10 bytes: c000d0000000000000c0
> TRACE +0.004 Read 1 bytes: c0
> TRACE +0.000 Read 8 bytes: 01d0020000000000

The trace shows that the ESP32 has started the chip erase command, but it never seems to finish. This command sequence is pretty standard, and a quick look at the MX25L512 datasheet suggests it should complete the same as for other flash chips - there's nothing unusual there.

My best guess is that the chip erase is triggering a power consumption spike which is causing the flash chip to brown out or glitch (or possibly the ESP32). To check this isn't happening, try increasing the decoupling capacitance on the flash chip (and maybe the ESP32), and if necessary monitor the flash chip VCC with an oscilloscope to look for a power glitch. Make sure to place the probe as close to the flash chip pins as possible.

radimkarnis, 2020-12-09:
Hello @dale6998,

Have you done any further investigation or tried any of the checks @projectgus proposed?

dale6998, 2020-12-09:
hey, sorry, I forgot I posted this.

changing the flash mode to DIO from QOUT fixed it.

Feel free to close this issue.  Thank you!

radimkarnis, 2020-12-09:
Perfect! I am glad you were able to resolve your problem. Thank you for reporting this and your cooperation. I am therefore closing the issue.

closed 2020-12-09 by radimkarnis