espressif/esptool #545: Cannot flash ESP-01 with diymore ESP-01 adapter (while ESP-01s work well)

gitterman, 2020-07-20:
I have a diymore.cc ESP-01 adapter which allows to easily flash/run an ESP-01 using an onboard switch.  
I am able to program my 4 ESP-01s boards without problems but the 2 ESP-01 boards, which I also own, always give me a timeout.  
Do I need to set different parameters for the ESP-01 compared to the ESP-01s?  

This is the error I get, when I try to upload the same (blink example) sketch to the ESP-01:  
```
esptool.py v2.7
Serial port COM3
Connecting........_____....._____....._____....._____....._____....._____.....____Traceback (most recent call last):
  File "C:\Users\PeterK\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.0/tools/upload.py", line 25, in <module> esptool.main(fakeargs)
  File "C:/Users/PeterK/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.0/tools/esptool\esptool.py", line 2841, in main esp.connect(args.before)
  File "C:/Users/PeterK/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.0/tools/esptool\esptool.py", line 483, in connect raise FatalError('Failed to connect to %s: %s' % (self.CHIP_NAME, last_error))
esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet headerd

```

raspiduino, 2020-07-31:
Please press reset and try again.

projectgus, 2020-08-28:
Hi @gitterman ,

Thanks for being patient while someone got back to you.

Espressif doesn't manufacture the ESP-01 or ESP-01S modules or the adapter you're using, we just make the ESP8266 chip on the module. So I'm afraid I can't answer this question with any certainty.

I can tell you that the output from esptool suggests that the ESP8266 is either not reset into the correct boot mode, not powered on, or that the COM port setting is wrong.

Suggest going through the troubleshooting steps here: https://github.com/espressif/esptool/#Troubleshooting - if this doesn't work then try contacting the manufacturer of the adapter/modules.

closed 2020-08-28 by projectgus

fdacio, 2025-05-28:
I was having the same problem with this recorder for esp 01 from the manufacturer Diymore, so I made a small adaptation. I soldered a 10k resistor between CH_EN (chip enable) and VCC and now it is working both in PROGRAMMING mode and in UART mode.