espressif/esptool #548: Failed to detect ESP32 chip type

lukegluke, 2020-08-06:
Windows 10
Python version 3.8.5
Chip ESP32-WROOM-32U
Custom board. Nothing else attached except UART0, BOOT and EN pins.
No IDE, Windows command prompt

```
esptool.py --trace -p COM5 flash_id
esptool.py v2.8
Serial port COM5
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.001 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.017 Read 1 bytes: c0
TRACE +0.001 Read 111 bytes:
    0108040007071220 00000000c0c00108 | ....... ........
    0400070712200000 0000c0c001080400 | ..... ..........
    0707122000000000 c0c0010804000707 | ... ............
    122000000000c0c0 0108040007071220 | . .............
    00000000c0c00108 0400070712200000 | ............. ..
    0000c0c001080400 0707122000000000 | ........... ....
    c0c0010804000707 122000000000c0   | ......... .....
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.003 Received full packet: 010804000707122000000000
TRACE +0.002 Received full packet: 010804000707122000000000
TRACE +0.003 Received full packet: 010804000707122000000000
TRACE +0.009 Received full packet: 010804000707122000000000
TRACE +0.002 Received full packet: 010804000707122000000000
TRACE +0.006 Received full packet: 010804000707122000000000
TRACE +0.005 Received full packet: 010804000707122000000000

Detecting chip type...TRACE +0.010 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=78000060
TRACE +0.001 Write 14 bytes: c0000a04000000000078000060c0
TRACE +0.006 Read 1 bytes: c0
TRACE +0.003 Read 13 bytes: 010d04000707122000000000c0
TRACE +0.001 Received full packet: 010d04000707122000000000
TRACE +3.006 Timed out waiting for packet header

A fatal error occurred: Timed out waiting for packet header

```

I tried with --before no_reset, same result.

I also have ESP32 DevKit V1 with ESP-WROOM32. flash_id reading is worked. Part of it trace result:

```
Detecting chip type...TRACE +0.014 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=78000060
TRACE +0.001 Write 14 bytes: c0000a04000000000078000060c0
TRACE +0.011 Read 1 bytes: c0
TRACE +0.000 Read 13 bytes: 010a04000025121500000000c0
TRACE +0.000 Received full packet: 010a04000025121500000000
 ESP32
```

As I see esptool detects ESP32 by DATE_REG_VALUE = 0x15122500 and see this in ESP32 DevKit V1
But ESP32-WROOM-32U return different packet "010d04000707122000000000". What that means?

Thanks in advance for help!

projectgus, 2020-08-11:
Hi @lukegluke ,

The packet which is received before the failure is another response to the "connecting" request which is the first thing that esptool sends to the chip (you can see another 10 or so packets with the same contents have already been received.) This packet was probably already received in a buffer somewhere when the request to read the `DATE_REG_VALUE` was being sent by esptool.

That received packet isn't the problem, esptool will have happily discarded that and will continue waiting for an actual response to the "read_reg" command to read the data code register. That packet has not been received at all, and then esptool times out.

Usually these kind of failures where the initial connection succeeds but then it immediately fails are due to hardware issues with power stability or the USB to serial connection. The chip is browning out or going into some other invalid state.

Suggest going through the [Troubleshooting list here](https://github.com/espressif/esptool#troubleshooting) and also take a look at the Schematic section of the ESP32-WROOM datasheet and the  [ESP32 Hardware Design Guidelines](https://www.espressif.com/en/support/documents/technical-documents) document and check your design meets all the relevant points.

Angus

lukegluke, 2020-09-02:
Now I got it worked. Successfully flashed esp with flash download tool.
Don't have particular answer what was wrong.
VCC is still the same, maybe enable/boot pin was not pulled in full (but I don't change resistors) or something else.
Anyway, thanks for participation!

closed 2020-09-02 by lukegluke