espressif/esptool #551: Yet another "Timed out waiting for packet header"...

choness, 2020-08-24:
I'm having trouble burning firmware onto a newly-built custom board (hosting ESP-WROOM-32 sourced from Digikey) from PlatformIO on Ubuntu (same problem with Arduino IDE). This is the second board I've made; the first worked perfectly using the exact same platformio configuration, UART and cabling until I fried the ESP (user stupidity). 

The board is going into download mode. I see:
```
rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download
```
But then after compiling and linking the upload fails:
```
Checking size .pio/build/az-delivery-devkit-v4/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  13.2% (used 70208 bytes from 532480 bytes)
Flash: [========  ]  81.9% (used 1074093 bytes from 1310720 bytes)
Configuring upload protocol...
AVAILABLE: esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Auto-detected: /dev/ttyUSB0
Uploading .pio/build/az-delivery-devkit-v4/firmware.bin
esptool.py v2.6
Serial port /dev/ttyUSB0
Connecting........_____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
```
So the serial connection works, and the board is in programming mode, but the upload connection fails.

Things I've checked include:

1. All my voltages look good, and in particular:
GPIO0 is LOW
GPIO2 is unconnected
GPIO12 is driven LOW  through a 100K
GPIOs 6-11 are not connected
PSU is capable of 2A and is well smoothed and filtered

2. Reduced upload_speed to 9600 in platformio.ini. I tried 115200 and 460800 also

3. Initiated an upload immediately after loading platformio, to be sure no serial connection was interfering

4. Remade all solder connections. As yet I haven't disconnected all other GPIO pins as that would require pulling lots of tiny surface mount components off the board

5. Worked through https://github.com/espressif/esptool#troubleshooting with no luck

Calling esptool directly (outside of platformio) so as to generate a trace of its attempting to interact with the ESP:
```
$ esptool.py --port /dev/ttyUSB0 --baud 115200 --chip esp32 --trace read_flash 0x00000 0x400000 delme.bin
esptool.py v2.6
Serial port /dev/ttyUSB0
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.004 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.103 Timed out waiting for packet header
.TRACE +0.050 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.101 Timed out waiting for packet header
.TRACE +0.050 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.101 Timed out waiting for packet header
.TRACE +0.051 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
```
A whole bunch of "Timed out waiting for packet header"s.

Likely I'm making some silly mistake, and I'll kick myself when someone points it out. Any thoughts on what might be wrong?

Ubuntu 20.04.1 LTS, Python 2.7.18rc1, ESP-WROOM-32

choness, 2020-08-26:
Problem went away when I replaced the UART. Strange that it worked with the previous board, and worked just fine for monitoring, but not for uploading - even at low speeds. Much as I'd like to dig deeper, I have to move on now that it's working. :-)

closed 2020-08-26 by choness

projectgus, 2020-08-27:
Hi @choness ,

Glad you got it working, and I wanted to say thanks for such a well structured issue report.

The one thing I was going to suggest before I saw it solved itself is that possibly the UART TX path wasn't getting through - ie it's confirmed you can receive from the ESP32 but perhaps when you send it's not making it all the way. I was going to suggest putting a logic analyzer or a scope probe as close to the U0_RXD pin of the module as possible. Cold solder joint, missing connection, pad jumper wire in the UART adapter, are all possible root causes.

The other one is that in Linux it's possible to have the same serial port open by more than one process. All processes will receive the RX side, but only the first process that opens the serial port can send to it. Although I think this should trigger a different error from the OS layer when you try to send, so it's probably not that.