espressif/esptool #553: esp32 in reset loop

Kashyapkoshti, 2020-08-28:
rst:0x10 (RTCWDT_RTC_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 

I am getting this error in my esp32 devkit v1
How to solve it?
I am using arduino IDE for programming the esp32.

Louis-Riel, 2020-09-08:
I'm facing the same issue. I have things connected to pins. I'm using UART2 on GPIO 25 and 26 and a digital output on GPIO 27 to drive a GPS module. Pin 27 is to the modules EN pin.

When the thing gets into a boot loop with that error, the error stops when I remove power from the GPS module. I think I'm touching pins that should not be touched or something along those lines. My board is the lolin d32 pro.

Do you have things connected to the pins, any code?

Kashyapkoshti, 2020-09-08:
@Louis-Riel i removed all the connection from pcb and tried all en/boot button combinations but this error is still there.

Kashyapkoshti, 2020-11-19:
@Louis-Riel did you  find any solution?

radimkarnis, 2021-04-28:
The "flash read err" implies the ROM bootloader can't read (or didn't find) the bootloader header at 0x1000. Some of the flash chip-related settings in Arduino IDE might be wrong.
Anyway, I am closing this issue because it most likely isn't an esptool issue (this happens after the application is loaded) and probably isn't relevant anymore. Feel free to reopen if you think otherwise

closed 2021-04-28 by radimkarnis