espressif/esptool #554: can no longer read or flash ESP8285 module

XDM525, 2020-09-02:
I have an esp8285 module that I had purchased from Adafruit. I had successfully flashed Tasmota firmware with esptool and had it operating correctly but when I had tried to go back and erase the flash using esptool, it will no longer connect. I have tried reading, writing, read_mac, chip_id and flash_id, nothing works, it just says connecting then a fatal error has occurred. It used to work just fine and I have not changed anything on my setup. I am using a FTDI programmer and have VCC to 3.3V, GND to GND, RX to TX, TX to RX, GPIO0 to GND and EN to 3.3V. If I remove GPIO0 from GND then power it up, it will operate correctly with the Tasmota firmware, I can just no longer connect to it through esptool. I used to have to unplug the FTDI and plug it back in to send each command but that no longer works either.  

# Operating system
Windows 10
# Python version

2.7.13

# What Chip

ESP8285 module 

# What development board or other hardware is the chip attached to

Module on breadboard with nothing else attached

# Is anything else attached to the development board, except for the serial flasher connections?

nothing else is attached

# Are you running esptool.py from an IDE such as Arduino or Eclipse?

 No IDE, Windows command prompt

# Full esptool.py command line that was run:

esptool.py -p COM4 erase_flash
esptool.py -p COM4 chip_id
esptool.py -p COM4 flash_id
esptool.py -p COM4 read_mac

# Full output from esptool.py
C:\Python27\Scripts>esptool.py -p COM4 flash_id 
esptool.py v2.8
Serial port COM4
Connecting........_____........_____........_____

A Fatal error occurred: Failed to connect to Espressif device: timed out waiting for packet header

# Do you have any other information from investigating this?

It used to work just fine. Had flashed tasmota firmware and got it working properly, went to erase flash but now it no longer connects. 

# Is there any other information you can think of which will help us reproduce this problem?

projectgus, 2020-10-22:
Hi @XDM525,

Sorry noone got back to you earlier.

There is nothing esptool can do which prevents the module from going into bootloader mode, this can only be related to hardware.

Given your description of the problem it sounds like maybe a hardware issue (GPIO0 is not connected to the chip any more, somehow). If pulling GPIO0 to GND successfully then the only other root cause would be that the module isn't actually resetting to enter the new boot mode - you could try to work around this by removing and replacing power, while keeping GPIO0 connected to GND the nentire time.

I also notice you mention a breadboard - these are often responsible for unusual hardware problems in my experience. Suggest trying to remove it if possible, and make direct connections.

radimkarnis, 2020-12-09:
Hello @XDM525,
I’m closing this issue because it has been inactive for a few months and it seems like it's not relevant anymore. Please reopen if further discussion is needed.

closed 2020-12-09 by radimkarnis