espressif/esptool #559: A fatal error occurred: Timed out waiting for packet header

megs-rs, 2020-09-11:
Hi there,

I had a problem with 'esptool.py'.

I was trying download a binary to ESP-01 with 1MB Flash.

The binary that i have tried can be found by the  [link](https://blogmasterwalkershop.com.br/arquivos/firmware/Ai-Thinker_ESP8266_DOUT_8Mbit_v1.5.4.1-a_20171130.bin).

The problem with the tool show is: "A fatal error occurred: Timed out waiting for packet header" after 100% sent.

Looking the code i see that the problem was in 'flash_md5sum' in value of 'timeout'.

I change the value to '60.0' and this fixed the problem.

I think that the function 'timeout_per_mb' is no so good.

Note that this binary have 1MB and this is not so common to ESP devices. I guess.

I think that this can help someone that have the same problem, and the developers can fix the procedure with the problem.

Best regards,

hansvandenelzen, 2020-09-12:
Hello megs-rs and others, I have the same problem when trying to upload a data sketch in my ESP8266 01S. The .ino loaded correctly.
I tried to downgrade the sw version of the latest esp8266 package from 2.7.4 to 2.5.2 and 2.6.3 They generate other errors but  don't work either. 
In your post you mention: the problem was in 'flash_md5sum' in value of 'timeout'. Where can I find this and how to change it?

See below the error messages:

[SPIFFS] data    : D:\Arduino\Arduino Projecten\ESP8266 Wifi module\ESP8266_Webserver_SPIFFS\data
[SPIFFS] size    : 512
[SPIFFS] page    : 256
[SPIFFS] block   : 8192
/index.html
/style.css
[SPIFFS] upload  : C:\Users\xxxx\AppData\Local\Temp\arduino_build_566349/ESP8266_Webserver_SPIFFS.spiffs.bin
[SPIFFS] address  : 0x7B000
[SPIFFS] reset    : --before default_reset --after hard_reset
[SPIFFS] port     : COM7
[SPIFFS] speed    : 115200
[SPIFFS] python   : C:\Users\xxxx\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1\python3.exe
[SPIFFS] uploader : C:\Users\xxxx\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\tools\upload.py

esptool.py v2.8
Serial port COM7
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 84:f3:eb:0e:xx:xx
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 1MB
Compressed 524288 bytes to 1524...
Wrote 524288 bytes (1524 compressed) at 0x0007b000 in 0.1 seconds (effective 29974.6 kbit/s)...
Traceback (most recent call last):
  File "C:\Users\xxxx\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\tools\upload.py", line 65, in <module>
    esptool.main(cmdline)
  File "C:/Users/xxxx/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/tools/esptool\esptool.py", line 2938, in main
    operation_func(esp, args)
  File "C:/Users/xxxx/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/tools/esptool\esptool.py", line 2398, in write_flash
    res = esp.flash_md5sum(address, uncsize)
  File "C:/Users/xxxx/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/tools/esptool\esptool.py", line 104, in inner
    return func(*args, **kwargs)
  File "C:/Users/xxxx/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/tools/esptool\esptool.py", line 691, in flash_md5sum
    timeout=timeout)
  File "C:/Users/xxxx/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/tools/esptool\esptool.py", line 369, in check_command
    val, data = self.command(op, data, chk, timeout=timeout)
  File "C:/Users/xxxx/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/tools/esptool\esptool.py", line 347, in command
    p = self.read()
  File "C:/Users/xxxx/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/tools/esptool\esptool.py", line 292, in read
    return next(self._slip_reader)
  File "C:/Users/xxxx/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/tools/esptool\esptool.py", line 2045, in slip_reader
    raise FatalError("Timed out waiting for packet %s" % waiting_for)
esptool.FatalError: Timed out waiting for packet header
SPIFFS Upload failed!

megs-rs, 2020-09-13:
Hi hansvandenelzen,

I changed the file '~user/.local/bin/esptool.py'.

I've attached the file for your convenience.

But, the better is to fix the procedure 'timeout_per_mb'.

I put the value of timeout fixed properly to my case.

Best regards,

Miguel Silva

hansvandenelzen, 2020-09-14:
Hi Miguel,

 

Could you attach the file mentioned. It wasn’t there

 

Met vriendelijke groet,

Hans van den Elzen

hansvandenelzen, 2020-09-14:
Hi Miguel

 

I changed te value from 8 to 60:

MD5_TIMEOUT_PER_MB = 60                # timeout (per megabyte) for calculating md5sum 

 

Now it loads perfectly.

Thanks a lot

 

Met vriendelijke groet,

Hans van den Elzen

megs-rs, 2020-09-14:
> Hi Miguel I changed te value from 8 to 60: MD5_TIMEOUT_PER_MB = 60 # timeout (per megabyte) for calculating md5sum Now it loads perfectly. Thanks a lot Met vriendelijke groet, Hans van den Elzen Van

That's great!

I think that the problem about the attach file is the system that remove the attached files.

If you could find the position to change the code, this is better.

Best regards,

darlanalves, 2020-11-28:
@megs-rs a big THANKS for your comments. The 60s timeout did the trick indeed.

This issue was driving me crazy here... I have multiple CH340 connectors and a bunch of ESP's.

I tried a lot of combinations, from changing wires to external power and so on, thinking it was a hardware problem...
I've used the same hardware setup for over 2 years and have never seen those errors before. I wonder why all of a sudden it's happening 🤔

darlanalves, 2020-11-28:
Um abraço, tchê!

megs-rs, 2020-11-28:
Que bom que te ajudou! 
Outro abraço tchê!

darlanalves, 2020-11-29:
I noticed something here... when I build my project, sometimes I get compilation errors, sometimes it builds and the final binary is incomplete, and sometimes it works. If I try 2 ~ 3 times it goes fine.

For each of these I keep an eye on what's the binary size and what's the compressed size... if the ratio is too high, e.g. binary is 200k but compressed is 500 bytes, something is definitely wrong.
That's when I see a flash failure on my set up. So the root cause is a failed compilation, not an esptool problem.

ElToberino, 2021-03-15:
I faced the same issue uploading data to a 192kB SPIFFS partition on an ESP with 1MB flash. File upload was successful - but upload error message appeared.

Changing the value MD5_TIMEOUT_PER_MB to 30 helped.

dongustavo, 2021-04-02:
> Hi there,
>
> I had a problem with 'esptool.py'.
>
> I was trying download a binary to ESP-01 with 1MB Flash.
>
> The binary that i have tried can be found by the [...]

It works! Thanks

radimkarnis, 2021-10-18:
@naamah75 Your report doesn't seem to be related to `MD5_TIMEOUT_PER_MB`. I suggest you open a new issue and provide detailed information about your problem.

DeepeshPatidar, 2022-08-30:
Hello All, I was getting below error when programming nodeMCU from arduino IDE. After diagnosis I found that it was cuased by the DHT11 sensor whose data Pin i connected to GPIO9 as named on nodeMCU S2. After removing sensor pin from GPIO9 node MCU programming properly. FYIP

esptool.py v3.0
Serial port COM4
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 48:55:19:c8:de:be
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 265040 bytes to 195064...
Writing at 0x00000000... (8 %)
Writing at 0x00004000... (16 %)
Writing at 0x00008000... (25 %)
Traceback (most recent call last):
  File "C:\Users\deepe\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2/tools/upload.py", line 66, in <module>
    esptool.main(cmdline)
  File "C:/Users/deepe/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/3.0.2/tools/esptool\esptool.py", line 3604, in main
    operation_func(esp, args)
  File "C:/Users/deepe/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/3.0.2/tools/esptool\esptool.py", line 2987, in write_flash
    esp.flash_defl_block(block, seq, timeout=DEFAULT_TIMEOUT * ratio * 2)
  File "C:/Users/deepe/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/3.0.2/tools/esptool\esptool.py", line 113, in inner
    return func(*args, **kwargs)
  File "C:/Users/deepe/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/3.0.2/tools/esptool\esptool.py", line 762, in flash_defl_block
    self.ESP_FLASH_DEFL_DATA, struct.pack('<IIII', len(data), seq, 0, 0) + data, self.checksum(data), timeout=timeout)
  File "C:/Users/deepe/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/3.0.2/tools/esptool\esptool.py", line 413, in check_command
    val, data = self.command(op, data, chk, timeout=timeout)
  File "C:/Users/deepe/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/3.0.2/tools/esptool\esptool.py", line 386, in command
    p = self.read()
  File "C:/Users/deepe/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/3.0.2/tools/esptool\esptool.py", line 331, in read
    return next(self._slip_reader)
  File "C:/Users/deepe/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/3.0.2/tools/esptool\esptool.py", line 2636, in slip_reader
    raise FatalError("Timed out waiting for packet %s" % waiting_for)
esptool.FatalError: Timed out waiting for packet header
esptool.FatalError: Timed out waiting for packet header

nico0201me, 2023-04-26:
> @megs-rs a big THANKS for your comments. The 60s timeout did the trick indeed.
>
> This issue was driving me crazy here... I have multiple CH340 connectors and a bunch of ESP's.
>
> I tried a lot of [...]

Hi, I spent more than two hours to try to solve my error in the log windows:
`A fatal error occurred: Timed out waiting for packet header`
I tried to reinstall Arduino IDE, I installed portable version, in installed various versions of esp8266 library, I installed forks of SPIFFS library, I changed port number....none of these actions succeed.
Then I gone to this forum and you saved my life.
I just put `MD5_TIMEOUT_PER_MB = 60                # timeout (per megabyte) for calculating md5sum` and it worked perfectcly also for me.

Many thanks to @megs-rs who found how to solve and also to @hansvandenelzen @darlanalves who followed this nice trick before me.

I create my GitHub account especially to thank you and confirm that it worked for all the users who may be interrested.

darlanalves, 2023-04-26:
Nice! That's the real power of a community 😃 

Happy hacking everyone!

radimkarnis, 2023-04-26:
@nico0201me @darlanalves nice job! there is now an option to use a configuration file exactly for the purpose of tinkering with the timeouts. A custom configuration for slower environments can be created. Please read about the feature [here](https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/configuration-file.html).

You need the latest esptool version, though. Judging from the `Timed out waiting for packet header` line you see, you are using a quite outdated version.

nico0201me, 2023-04-26:
> @nico0201me @darlanalves nice job! there is now an option to use a configuration file exactly for the purpose of tinkering with the timeouts. A custom configuration for slower environments can be [...]

thanks @radimkarnis but there is something I don't understand.
I regularly upgrade Arduino IDE to the latest 1.x version. Currently I use 1.8.19 which is the latest one. You mean that when I upgrade not all the components are updated, right?

You are certainly right, the path to the esptool  contains Arduino15. It means Arduino 1.5.x branch ???
C:\Users\myusername\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\tools\esptool

radimkarnis, 2023-04-26:
It's not about updating Arduino IDE, but the arduino core for esp32 (see [here](https://github.com/espressif/arduino-esp32)). The latest core contains updated tools, including esptool 😊

radimkarnis, 2023-12-11:
I am closing this issue, as it's not relevant anymore.
Esptool now provides more information in its error messages and handles more edge cases like this. The `Timed out waiting for packet header` doesn't even exist in recent releases.

For anyone experiencing similar issues - the internal variables like `MD5_TIMEOUT_PER_MB` do not have to be hacked. Esptool now supports [configuration files](https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/configuration-file.html), these allow anyone to configure the timeouts and other variables according to their environment needs without having to dig in the code.

closed 2023-12-11 by radimkarnis