espressif/esptool #564: D1 mini nodeMcu ESP8266MOD 12-F: esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header

jordivandijk, 2020-10-03:
I'm kinda new to this. I have some arduinos and decided to try this D1 mini.
on 1 of my computers the chip did not even show up no matter what i did (installing drivers, re-installing arduino ide / pyhton and the esp8266 comunitie lib). On my onther pc it shows up but i get this message:

esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header

i've tried multiple things such as comnecting pin d4 to ground while uploading. but it just wont work. 

# Windows 10

# Python version 3.8

Can run `python -V` to check this.

# What: Chip ESP 8266MOD 12-F

# What development board or other hardware is the chip attached to: D1 mini nodeMcu

Arduino ESP8266 IDE version 2.7.4

# Full esptool.py command line that was run:

Arduino:1.8.13 (Windows Store 1.8.42.0) (Windows 10), Board:"LOLIN(WEMOS) D1 R2 & mini, 80 MHz, Flash, Disabled (new can abort), All SSL ciphers (most compatible), 4MB (FS:2MB OTA:~1019KB), v2 Lower Memory, Disabled, None, Only Sketch, 921600"

Executable segment sizes:

IROM   : 228640          - code in flash         (default or ICACHE_FLASH_ATTR) 

IRAM   : 26756   / 32768 - code in IRAM          (ICACHE_RAM_ATTR, ISRs...) 

DATA   : 1248  )         - initialized variables (global, static) in RAM/HEAP 

RODATA : 688   ) / 81920 - constants             (global, static) in RAM/HEAP 

BSS    : 24880 )         - zeroed variables      (global, static) in RAM/HEAP 

De schets gebruikt 257332 bytes (24%)  programma-opslagruimte. Maximum is 1044464 bytes.

Globale variabelen gebruiken 26816 bytes (32%) van het dynamisch geheugen. Resteren 55104 bytes voor lokale variabelen. Maximum is 81920 bytes.

esptool.py v2.8

Serial port COM1

Connecting........_____....._____....._____....._____....._____....._____.....____Traceback (most recent call last):

  File "C:\Users\jordi\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.7.4/tools/upload.py", line 65, in <module>

    esptool.main(cmdline)

  File "C:/Users/jordi/Documents/ArduinoData/packages/esp8266/hardware/esp8266/2.7.4/tools/esptool\esptool.py", line 2890, in main

    esp.connect(args.before)

  File "C:/Users/jordi/Documents/ArduinoData/packages/esp8266/hardware/esp8266/2.7.4/tools/esptool\esptool.py", line 483, in connect

    raise FatalError('Failed to connect to %s: %s' % (self.CHIP_NAME, last_error))

esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header

esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header

projectgus, 2020-10-08:
Hi @jordivandijk,

Are you sure the serial port COM1 is correct?

The ESP8266 doesn't seem to be responding. This may be because the hardware is not working propertly at all, or it may be because there is some problem with resetting into the correct mode.

> i've tried multiple things such as comnecting pin d4 to ground while uploading. but it just wont work.

I'm think for WeMOS D1 Mini the pin to ground is D3. (Normally it's referred to as GPIO0, but WeMos use their own pin numbering scheme - may need to check whatever docs came with your board)

If you have GPIO3 (aka D3) grounded, and nothing else connected to the D1 Mini at all (including removing it from any breadboard), try pressing Upload in Arduino IDE and then quickly pressing and releasing the Reset button on the module, and see if it can upload this way.

If that doesn't work, suggest going through the other items here:
https://github.com/espressif/esptool/#bootloader-wont-respond

... and if nothing works, I'm afraid you may have faulty hardware.

radimkarnis, 2020-12-09:
Hello @jordivandijk, 
is there any update? have you been able to resolve your issue by going through the tips mentioned above?

Radim

radimkarnis, 2020-12-18:
I am closing this issue because it seems like it is not relevant anymore. Please reopen if you think further discussion is needed.

closed 2020-12-18 by radimkarnis