espressif/esptool #567: Failed to connect to Espressif device: Timed out waiting for packet header

l1g4v, 2020-10-15:
# Operating system
Windows 10 10.0.19041 Build 19041
# Python version
3.8.2
# What Chip
ESP32-WROOM32

# What development board or other hardware is the chip attached to

ESP32 dev kit (more specific, a board I bought from here https://articulo.mercadolibre.com.mx/MLM-727796906-esp32-38-pines-esp-wroom-32-_JM) 

# Is anything else attached to the development board, except for the serial flasher connections?

Nothing

# Are you running esptool.py from an IDE such as Arduino or Eclipse?

Windows command prompt and Arduino IDE

# Full esptool.py command line that was run:

esptool.py -p COM4 --trace erase_flash

# Full output from esptool.py

```
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.001 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.113 Timed out waiting for packet header
.TRACE +0.120 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.113 Timed out waiting for packet header
.TRACE +0.116 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.114 Timed out waiting for packet header
.TRACE +0.118 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.113 Timed out waiting for packet header
.TRACE +0.118 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.001 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
[... 575 lines cut ...]
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.115 Timed out waiting for packet header
_TRACE +0.117 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.001 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.114 Timed out waiting for packet header
_
```

# Do you have any other information from investigating this?

The first day I tested the board, just a simple blink example through Arduino IDE, it uploaded then the next sketch I tried to upload failed with the message "A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header" so I tried a flash erase with esptool and neither this worked
I've tried fixes like connecting EN and GND with a capacitor
Shorting GPIO0 with GND
Lowering the baud rate as low as 9600
And putting the board in ROM serial mode pressing BOOT and RST (actually the board puts itself in this mode, I verified it by connecting a generic USB-UART to the TX-RX on the board and watching the output during the execution of this command and during upload with the Arduino IDE ```rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))``` as expected

# Is there any other information you can think of which will help us reproduce this problem?

No, it just failed

l1g4v, 2020-10-17:
After even connecting a usb-uart directly to the module it didn't work so I can only conclude this was a hardware failure

closed 2020-10-17 by l1g4v

projectgus, 2020-10-18:
Hi @l1g4v ,

Agree with your conclusion. Maybe a cold solder joint on the U0RXD pin meant that you see the ROM boot log but the ESP32 was only sometimes receiving data from serial?

Angus