espressif/esptool #569: Arduino IDE: esptool.FatalError: Invalid head of packet (0x46)

FrFrey98, 2020-10-23:
Hey,
I have a problem while trying to use the Feather Huzzah (https://www.adafruit.com/product/3213) with the Ardurino IDE. I configurated the IDE like it is recommended in here : https://learn.adafruit.com/adafruit-fea ... rduino-ide, but when I try to upload my code I get the message "esptool.FatalError: Invalid head of packet (0x46)". I even tried it not with my own code but with the Blink test code from the same article, but the error message stays the same.
Maybe someone has an idea where this problem comes from.
I am from Germany so excuse me if my english is not perfect.
Here ist the completet log of the Arduino IDE:

Arduino: 1.8.13 (Windows 10), Board: "Adafruit Feather HUZZAH ESP8266, 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), 4MB (FS:2MB OTA:~1019KB), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

Executable segment sizes:

IROM : 237664 - code in flash (default or ICACHE_FLASH_ATTR)

IRAM : 28080 / 32768 - code in IRAM (ICACHE_RAM_ATTR, ISRs...)

DATA : 1260 ) - initialized variables (global, static) in RAM/HEAP

RODATA : 804 ) / 81920 - constants (global, static) in RAM/HEAP

BSS : 25464 ) - zeroed variables (global, static) in RAM/HEAP

Der Sketch verwendet 267808 Bytes (25%) des Programmspeicherplatzes. Das Maximum sind 1044464 Bytes.

Globale Variablen verwenden 27528 Bytes (33%) des dynamischen Speichers, 54392 Bytes für lokale Variablen verbleiben. Das Maximum sind 81920 Bytes.

esptool.py v2.8

Serial port COM3

Connecting....

Chip is ESP8266EX

Features: WiFi

WARNING: Detected crystal freq 20.22MHz is quite different to normalized freq 26MHz. Unsupported crystal in use?

Crystal is 26MHz

MAC: 00:08:ff:00:00:00

Uploading stub...

Running stub...

Traceback (most recent call last):

File "C:\IoTW\arduino-1.8.13\portable\packages\esp8266\hardware\esp8266\2.7.2/tools/upload.py", line 65, in

esptool.main(cmdline)

File "C:/IoTW/arduino-1.8.13/portable/packages/esp8266/hardware/esp8266/2.7.2/tools/esptool\esptool.py", line 2909, in main

esp = esp.run_stub()

File "C:/IoTW/arduino-1.8.13/portable/packages/esp8266/hardware/esp8266/2.7.2/tools/esptool\esptool.py", line 637, in run_stub

p = self.read()

File "C:/IoTW/arduino-1.8.13/portable/packages/esp8266/hardware/esp8266/2.7.2/tools/esptool\esptool.py", line 292, in read

return next(self._slip_reader)

File "C:/IoTW/arduino-1.8.13/portable/packages/esp8266/hardware/esp8266/2.7.2/tools/esptool\esptool.py", line 2057, in slip_reader

raise FatalError('Invalid head of packet (0x%s)' % hexify(b))

esptool.FatalError: Invalid head of packet (0x46)

esptool.FatalError: Invalid head of packet (0x46)

Dieser Bericht wäre detaillierter, wenn die Option
"Ausführliche Ausgabe während der Kompilierung"
in Datei -> Voreinstellungen aktiviert wäre.

projectgus, 2020-11-04:
> WARNING: Detected crystal freq 20.22MHz is quite different to normalized freq 26MHz. Unsupported crystal in use?

This warning seems strange. Do you see this each time?

It's possible there is some fundamental problem with the hardware you have. Also, the serial baud rate is used to estimate the crystal speed so please check if you have the latest Silicon Labs CP2102 drivers (this is the usb/serial chip used on Adafruit's Huzzah board).

You may also wish to try changing the flashing baud rate in Arduino to a lower or higher value, see if you get a different outcome.

colin-odowd, 2020-11-17:
I am experiencing the same issue on a generic ESP8266 board.

aeronauta1, 2020-12-07:
Seems to be a common problem , I am having exactly the same , any progress?? please
thank youJorge

colin-odowd, 2020-12-07:
@aeronauta1 So I realized I was actually using an ESP32 board. I changed my board to ESP32 Dev Module and used this for my board manager URL and it worked: https://dl.espressif.com/dl/package_esp32_index.json

aeronauta1, 2020-12-07:
@colin-odowd thank you , I managed to get the sketch to compile and load? allthough the led did not flash..., when attempting to do the WiFi test fails as it does not have the right libs (expects Huzzah) and so on ... tired of trying , I will find something more reliable to play with...this is for me a hobby and I want to enjoy it , not be fighting with it all along...

Jorge

radimkarnis, 2020-12-11:
Hello everyone, 
I am closing this issue because it seems like it is not relevant anymore. Please reopen if you think further discussion is needed.

closed 2020-12-11 by radimkarnis

13protons, 2021-04-04:
> @aeronauta1 So I realized I was actually using an ESP32 board. I changed my board to ESP32 Dev Module and used this for my board manager URL and it worked: https://dl.espressif.com/dl/package_esp32_index.json

This helped me get to working! Although the board is clearly labeled as the Adafruit Feather Huzzah, I needed the generic ESP32 board profile to make things work. 

Also a note - the example blink code form the Adafruit site tries to blink pin0 - I changed it all to `13` (based on the printed label) and the red LED on the feather blinks like it should.

[image]

colin-odowd, 2021-04-05:
@13protons glad I could help, sometimes getting these boards up and running is the most frustrating part!

sercasti, 2023-10-26:
slowing down the upload speed fixed it for me