espressif/esptool #570: Regression: when flashing ESP32 - 2.1 works great, 2.6, 2.8 do not

dalbert2, 2020-10-30:
I spent gobs of time trying to get my ESP32-WROOM to flash reliably using esptool; after trying all of the hardware "fixes" (adding bulk capacitance, better power supply, varying RC timing on EN and GPIO0, and trying every baud rate from 960Kbps to 9600bps, I eventually found that the problem is a regression in esptool.  When I downgrade to 2.1, it works like a champ every time (up to 430kbps).  With the newer versions, most attempts fail.

# Operating system Windows 10

# Python version  Python 3.8.5

# What Chip ESP-WROOM-32 module

# What development board or other hardware is the chip attached to  Plain module on custom PCB

If your hardware is custom or unusual, please attach a photo to the issue.

# Is anything else attached to the development board, except for the serial flasher connections?
32kHz xtal connected, GPIO2 pulled low through 4k7, GPIO12,15 are floating, I2C I/O expander, SPI radio transceiver, cellular modem on UART2 (inactive).

# Are you running esptool.py from an IDE such as Arduino or Eclipse? VSCode 1.5.0 with PlatformIO Core 5.0.1

# Full esptool.py command line that was run:
Sorry, this gets hidden by the tool, I'm not sure how to find this

# Full output from esptool.py

This is the version that works:

```
esptool.py v2.1
Connecting....
Chip is ESP32D0WDQ6 (revision (unknown 0xa))
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 15872 bytes to 10319...

Writing at 0x00001000... (100 %)
Wrote 15872 bytes (10319 compressed) at 0x00001000 in 0.2 seconds (effective 509.7 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 128...

Writing at 0x00008000... (100 %)
Wrote 3072 bytes (128 compressed) at 0x00008000 in 0.0 seconds (effective 2578.7 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...

Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 8338.7 kbit/s)...
Hash of data verified.
Compressed 861904 bytes to 486621...

Writing at 0x00010000... (3 %)
Writing at 0x00014000... (6 %)
Writing at 0x00018000... (10 %)
Writing at 0x0001c000... (13 %)
Writing at 0x00020000... (16 %)
Writing at 0x00024000... (20 %)
Writing at 0x00028000... (23 %)
Writing at 0x0002c000... (26 %)
Writing at 0x00030000... (30 %)
Writing at 0x00034000... (33 %)
Writing at 0x00038000... (36 %)
Writing at 0x0003c000... (40 %)
Writing at 0x00040000... (43 %)
Writing at 0x00044000... (46 %)
Writing at 0x00048000... (50 %)
Writing at 0x0004c000... (53 %)
Writing at 0x00050000... (56 %)
Writing at 0x00054000... (60 %)
Writing at 0x00058000... (63 %)
Writing at 0x0005c000... (66 %)
Writing at 0x00060000... (70 %)
Writing at 0x00064000... (73 %)
Writing at 0x00068000... (76 %)
Writing at 0x0006c000... (80 %)
Writing at 0x00070000... (83 %)
Writing at 0x00074000... (86 %)
Writing at 0x00078000... (90 %)
Writing at 0x0007c000... (93 %)
Writing at 0x00080000... (96 %)
Writing at 0x00084000... (100 %)
C:\Users\david\.platformio\packages\tool-esptoolpy@1.20100.0\esptool.py:2236: DeprecationWarning: inspect.getargspec() is deprecated since Python 3.0, use inspect.signature() or inspect.getfullargspec()
Wrote 861904 bytes (486621 compressed) at 0x00010000  operation_args,_,_,_ = inspect.getargspec(operation_func)
 in 11.8 seconds (effective 582.4 kbit/s)...
Hash of data verified.
```

Showing a successful flashing with 2.6 (but normally it fails after a lengthy period of:
Connecting........_____....._____....._____....._____....._____....._____.....____

```
esptool.py v2.6
Configuring upload protocol...
AVAILABLE: esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, 
tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Use manually specified: COM4
Uploading .pio\build\esp32dev\firmware.bin
esptool.py v2.6
Serial port COM4
Connecting........_____....._____....._____....._____....._____....._____.....____
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: 24:6f:28:7a:d8:84
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 15872 bytes to 10319...

Writing at 0x00001000... (100 %)
Wrote 15872 bytes (10319 compressed) at 0x00001000 in 0.3 seconds (effective 498.8 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 128...

Writing at 0x00008000... (100 %)
Wrote 3072 bytes (128 compressed) at 0x00008000 in 0.0 seconds (effective 1635.9 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...

Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 4687.4 kbit/s)...
Hash of data verified.
Compressed 861856 bytes to 486613...

Writing at 0x00010000... (3 %)
Writing at 0x00014000... (6 %)
Writing at 0x00018000... (10 %)
Writing at 0x0001c000... (13 %)
Writing at 0x00020000... (16 %)
Writing at 0x00024000... (20 %)
Writing at 0x00028000... (23 %)
Writing at 0x0002c000... (26 %)
Writing at 0x00030000... (30 %)
Writing at 0x00034000... (33 %)
Writing at 0x00038000... (36 %)
Writing at 0x0003c000... (40 %)
Writing at 0x00040000... (43 %)
Writing at 0x00044000... (46 %)
Writing at 0x00048000... (50 %)
Writing at 0x0004c000... (53 %)
Writing at 0x00050000... (56 %)
Writing at 0x00054000... (60 %)
Writing at 0x00058000... (63 %)
Writing at 0x0005c000... (66 %)
Writing at 0x00060000... (70 %)
Writing at 0x00064000... (73 %)
Writing at 0x00068000... (76 %)
Writing at 0x0006c000... (80 %)
Writing at 0x00070000... (83 %)
Writing at 0x00074000... (86 %)
Writing at 0x00078000... (90 %)
Writing at 0x0007c000... (93 %)
Writing at 0x00080000... (96 %)
Writing at 0x00084000... (100 %)
Wrote 861856 bytes (486613 compressed) at 0x00010000 in 12.2 seconds (effective 566.5 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

# Do you have any other information from investigating this?

2.6 and 2.8 succeed occasionally, but mostly they fail to connect.  Running esptool from the command line with verbose output shows many failed attempts to connect; I can provide output with verbose trace output turned on if that will help.

# Is there any other information you can think of which will help us reproduce this problem?

This has been highly reproducible; 2.6 and 2.8 fail most of the time at any baudrate.  I spent quite a bit of time looking for solutions on the web and it looks like *many* people have this problem.  Anything you can do to help would be much appreciated; I'm happy to run tests and provide output, unfortunately, I'm a C/C++/Java guy, not python.

dalbert2, 2020-10-30:
Output from attempt to get chip_id with trace turned on:
````
C:\Users\david\.platformio\packages\tool-esptoolpy@1.20600.0>python esptool.py --port COM4 --chip esp32 --trace chip_id
esptool.py v2.6
Serial port COM4
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.001 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.115 Timed out waiting for packet header
.TRACE +0.068 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.001 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.005 Read 1 bytes: c0
TRACE +0.000 Read 3 bytes: 010804
TRACE +0.001 Read 8 bytes: 0012205555000000
TRACE +0.000 Read 4 bytes: 00c0c001
TRACE +0.000 Received full packet: 010804001220555500000000
TRACE +0.012 Read 94 bytes:
    0804001220555500 000000c0c0010804 | .... UU.........
    0012205555000000 00c0c00108040012 | .. UU...........
    20555500000000c0 c001080400125555 |  UU...........UU
    00000000c0c00108 0400122055550000 | ........... UU..
    0001c00108040012 20555500000000c0 | ........ UU.....
    c001080400122055 5500000000c0     | ...... UU.....
TRACE +0.000 Received full packet: 010804001220555500000000
TRACE +0.013 Received full packet: 010804001220555500000000
TRACE +0.012 Received full packet: 010804001220555500000000
TRACE +0.011 Received full packet: 0108040012555500000000
TRACE +0.012 Received full packet: 010804001220555500000001
TRACE +0.011 Read invalid data:
    0804001220555500 000000c0c0010804 | .... UU.........
    0012205555000000 00c0c00108040012 | .. UU...........
    20555500000000c0 c001080400125555 |  UU...........UU
    00000000c0c00108 0400122055550000 | ........... UU..
    0001c00108040012 20555500000000c0 | ........ UU.....
    c001080400122055 5500000000c0     | ...... UU.....
TRACE +0.001 Remaining data in serial buffer:
.TRACE +0.074 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.001 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.007 Read 1 bytes: c0
TRACE +0.000 Read 5 bytes: 0108040007
TRACE +0.001 Read 5 bytes: 0712200000
TRACE +0.000 Read 6 bytes: 0000c0c00108
TRACE +0.001 Received full packet: 010804000707122000000000
TRACE +0.011 Read 78 bytes:
    0400070712200000 0000c0c001080400 | ..... ..........
    0707122000000000 c0c0010804000707 | ... ............
    122000000000c0c0 0108040001c00108 | . ..............
    040007071220c007 07122000000000c0 | ..... .... .....
    c001080400070712 2000000000c0     | ........ .....
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.013 Received full packet: 010804000707122000000000
TRACE +0.011 Received full packet: 010804000707122000000000
TRACE +0.011 Received full packet: 0108040001
[... 1747 lines cut ...]
TRACE +0.000 Read 5 bytes: 7a28000000
TRACE +0.001 Read 2 bytes: 00c0
TRACE +0.000 Received full packet: 0106040084d87a2800000000
TRACE +0.006 Read 6 bytes: c04f484149c0
TRACE +0.000 Received full packet: 4f484149
Stub running...
Warning: ESP32 has no Chip ID. Reading MAC instead.
TRACE +0.000 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=08a00160
TRACE +0.000 Write 14 bytes: c0000a04000000000008a00160c0
TRACE +0.002 Read 1 bytes: c0
TRACE +0.000 Read 2 bytes: 010a
TRACE +0.000 Read 4 bytes: 02006f24
TRACE +0.000 Read 4 bytes: 3e000000
TRACE +0.001 Read 1 bytes: c0
TRACE +0.000 Received full packet: 010a02006f243e000000
TRACE +0.013 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=04a00160
TRACE +0.000 Write 14 bytes: c0000a04000000000004a00160c0
TRACE +0.002 Read 1 bytes: c0
TRACE +0.000 Read 3 bytes: 010a02
TRACE +0.000 Read 7 bytes: 0084d87a280000
TRACE +0.001 Read 1 bytes: c0
TRACE +0.000 Received full packet: 010a020084d87a280000
MAC: 24:6f:28:7a:d8:84
Hard resetting via RTS pin...
````

dalbert2, 2020-10-30:
Same command issued from esptoolpy 2.1  (note --trace not supported but response is immediate); this works perfectly every time in 2.1:
````
C:\Users\david\.platformio\packages\tool-esptoolpy@1.20100.0>python esptool.py --port COM4 --chip esp32 chip_id
esptool.py v2.1
esptool.py:2236: DeprecationWarning: inspect.getargspec() is deprecated since Python 3.0, use inspect.signature() or inspect.getfullargspec()
  operation_args,_,_,_ = inspect.getargspec(operation_func)
Connecting........_____....._____....._____....._____....._____.....__^C
C:\Users\david\.platformio\packages\tool-esptoolpy@1.20100.0>python esptool.py --port COM4 --chip esp32 chip_id
esptool.py v2.1
esptool.py:2236: DeprecationWarning: inspect.getargspec() is deprecated since Python 3.0, use inspect.signature() or inspect.getfullargspec()
  operation_args,_,_,_ = inspect.getargspec(operation_func)
Connecting....
Chip is ESP32D0WDQ6 (revision (unknown 0xa))
Uploading stub...
Running stub...
Stub running...
Chip ID: 0x3e246f287ad8
Hard resetting...
````

projectgus, 2020-11-04:
Hi @dalbert2,

Thanks for the very detailed report and diligence looking at older versions, this is quite interesting.

The reason sync is failing is that is expects to see 8 identical responses to the sync command (esptool.py sends one sync request, ESP32 ROM code sends the same response packet 8 times.)

You can see in the trace that most of the time the response packet (`010804000707122000000000`) is correct but not every time, there's a permutation in each set of 8 replies where an extra byte is inserted or dropped or an existing byte corrupted. Eventually 8 responses are received without any of them being corrupt.

I think most likely the root cause here is a noisy RX serial path from the ESP32 to the computer (ground mismatch, electrical noise, etc). This is causing framing errors and invalid bytes.

This "wait for 8 replies" logic has been in esptool since long before v1.0, but it does seem that something changed since v2.1 that makes it less tolerant about what it will accept. Well spotted! Will try to work out exactly what.

It does makes sense to try and restore the v2.1 behaviour and move forward if it looks like the device is in the right boot mode, even if the RX side of the link is noisy. Although some operations may fail later on, some may succeed - and the important operations all have some MD5 integrity verification. Probably the best option is to log a warning if it looks like the link is noisy in this way, and then try and proceed.

To confirm this theory, could you please try running `esptool.py -b 115200 -p PORT read_flash 0 0x10000 test.bin` and see if it completes without error on v2.1? This will try to read 64KB of data back from the chip (followed by an MD5 value to verify the data). If it fails then it probably confirms that there's noise in the RX serial path but maybe not the TX path (i.e. why writing to flash can work as it mostly writes data). If this succeeds then my theory is wrong and something else is going on during the sync that esptool isn't correctly accounting for..

> it looks like many people have this problem

One of the unfortunate limitations of esptool right now is that it reports very similar errors for very different causes. For example, being in the wrong boot mode will produce a very similar error (it will also say `A fatal error occurred: Failed to connect to ESP32: Invalid head of packet (0xx)` but the byte is something else).

On the roadmap for a future version is being smarter about interpreting the known device state and trying to come up with some more context for connection errors, so similar root causes don't look the same.

However I agree it's quite likely other people are seeing some variation of this problem. Good work on narrowing it down.

radimkarnis, 2020-12-30:
Hello @dalbert2,

are you still interested in helping us to narrow down the possible causes of your problem, or is it not relevant anymore?

radimkarnis, 2021-04-07:
I tested this thoroughly but was not able to reproduce the described behavior. This leads me to believe that Angus's theory of a noisy RX line was correct. I am therefore closing this issue, we can revisit this if anyone experiences similar problem.

closed 2021-04-07 by radimkarnis