espressif/esptool #573: argument --version/-e: invalid choice: '3' - macOS Big Sur

GoRo33, 2020-11-07:
* Operating system: macOS Big Sur 11.0.1 Beta
* Python version: Python 3.8.5

# Full esptool.py command line as run: make -j14 app - this is from esp8266 rtos sdk example app. 

Example: esptool.py version: esptool.py v3.0-dev - ESP8266 ROM Bootloader Utility

# Full output from esptool.py

LD /Volumes/Programowanie/Projekty/esp_homekit_env_dev/build/env_sensor.elf
usage: esptool elf2image [-h] [--output OUTPUT] [--version {1,2}]
                         [--min-rev {0,1,2,3}] [--secure-pad] [--secure-pad-v2]
                         [--elf-sha256-offset ELF_SHA256_OFFSET]
                         [--flash_freq {40m,26m,20m,80m}]
                         [--flash_mode {qio,qout,dio,dout}] [--flash_size FLASH_SIZE]
                         [--spi-connection SPI_CONNECTION]
                         input
esptool elf2image: error: argument --version/-e: invalid choice: '3' (choose from '1', '2')
make: *** [/Volumes/Programowanie/Projekty/esp_homekit_env_dev/build/env_sensor.bin] Error 2

# What is the expected behaviour?

Generate app image and flash the device. 

# Do you have any other information from investigating this?

Example: On earlier versions of macOS it works without problems. 

# Is there any other information you can think of which will help us reproduce this problem?

projectgus, 2020-11-09:
Hi @GoRo33 ,

Which version of the ESP8266 SDK are you using? The ESP8266 SDK currently has a fork of esptool included in it, and the fork adds a new elf2image version 3. You'll need to use that version of esptool, however the SDK should be pulling this in directly not calling out to the separate esptool.

Angus

projectgus, 2020-11-09:
Ah, I just realised you're using MacOS Big Sur. I'm guessing you upgraded the esptool version to work around [this bug](https://github.com/espressif/esptool/issues/540#issuecomment-655324530) until the ESP8266 changes are merged into the main esptool, you can [try applying this workaround](https://github.com/espressif/esptool/issues/540#issuecomment-655324530) to the bundled version of esptool in the SDK.

Sorry for the inconvenience.

GoRo33, 2020-11-09:
@projectgus  Hi! Thank you for reply. I have make walk around that you mentioned and now it works. I'm aware that this is beta os, but i think it will be good to mention about my issue.