espressif/esptool #574: Failed Windows installation

bertiebeez, 2020-11-08:
* Operating system: 
Windows 8.1
* Python version: (`python -V` to check this) 
3.8

# Full esptool.py command line as run:
 c:\program tools\python38>pip install esptool

Example: esptool.py version

# Full output from esptool.py

(please copy and paste all lines of output)

c:\program files\pyton38>python.py
'esptools' not recognized as an internal or external command,
operable program or batch

# What is the expected behaviour?

Example: Print the version number

# Do you have any other information from investigating this?

Example: I tried on my friend's Windows 10 PC and the command works there

# Is there any other information you can think of which will help us reproduce this problem?

Example: Only crashes on first day of the month

projectgus, 2020-11-09:
Hi @bertiebeez ,

I'm sorry, there isn't enough information in this report to give you any help. Can you please give some more information, for example *exactly* what command you executed?

> c:\program files\pyton38>python.py
> 'esptools' not recognized as an internal or external command,

The command is `esptool.py`, not `esptools.py` - so maybe this is the root cause of the problem?

bertiebeez, 2020-11-11:
I was finally able to get it all sorted out. I was following a video from Dr zzs. He did not mention that you needed to run the cmd administrator. Once I did that, it installed just fine and worked as it should. Thank you for your responce!DavidSent from my Samsung Galaxy Tab®|PRO

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/espressif/esptool/issues/574#issuecomment-723803678",
"url": "https://github.com/espressif/esptool/issues/574#issuecomment-723803678",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

projectgus, 2020-11-12:
Thanks @bertiebeez for the update. In that case I'll close this issue.

closed 2020-11-12 by projectgus