espressif/esptool #575: Uploading stub..... A fatal error occurred: Invalid head of packet (0x13)

srhart, 2020-11-09:
# Windows 10

# 2.7.16

# ESP8266

# Espruino WiFi

# No

# Are you running esptool.py from an IDE such as Arduino or Eclipse?

 No IDE, Windows command prompt

# Full esptool.py command line that was run:

All commands give same error
esptool.py --port com13 --baud 74880 --chip esp8266 --before no_reset_no_sync erase_flash

# Full output from esptool.py

esptool.py v3.1-dev
Serial port com13
Connecting...
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 5c:cf:xx:c0:f4:xx
Uploading stub...

A fatal error occurred: Invalid head of packet (0x13)

# Do you have any other information from investigating this?

I have successfully flashed another instance of this hardware

# Is there any other information you can think of which will help us reproduce this problem?

Last part of trace..
    eefafee202039a9a 9aff9a9ee202049a | ................
    ff9a9ee202059aff 9a9ee202069aff9a | ................
    9ee202079aff9aee eaff8b223a924739 | ...........":.G9
    dbdc402341b022b0 b090604602000032 | ..@#A."...`F...2
    02001b223aeeeaff 2a39bd024733ef31 | ...":...*9..G3.1
    67fa2d0e426131c0                  | g.-.Ba1.
TRACE +0.768 Read 1 bytes: 13
TRACE +0.001 Read invalid data: 13
TRACE +0.002 Remaining data in serial buffer:

A fatal error occurred: Invalid head of packet (0x13)

projectgus, 2020-11-11:
Hi @srhart,

Usually this kind of failure with invalid serial data indicates a problem with the serial link or overall electrical noise.

* Is using the `--before default_reset` option an alternative instead of no_reset_no_sync? This may help if the problem is a baud rate mismatch.
* Can you describe anything about the exact hardware you have and the connections to the ESP8266?

srhart, 2020-11-11:
I have retried this morning and changed the ESP_RAM_BLOCK and FLASH_WRITE_SIZE to 0x100, I also moved it away from my mobile phone, just in case and all seems ok.

Thank you.
Please close.

closed 2020-11-11 by srhart