espressif/esptool #576: ESP32 Timed out waiting for packet header error only on some boards on macOS but not on Windows on same machine

alexqzd, 2020-11-13:
# Operating system 
macOS 11.0

# Python version
3.9

# What Chip
ESP32-WROOM32

# What development board or other hardware is the chip attached to
Plan module on bread board

# Is anything else attached to the development board, except for the serial flasher connections?
No

# Are you running esptool.py from an IDE such as Arduino or Eclipse?
Tried both on command line and PlatformIO

# Full esptool.py command line that was run:
 ./esptool.py --port /dev/cu.usbserial-AL05NZOI flash_id

# Full output from esptool.py
esptool.py v3.1-dev
Serial port /dev/cu.usbserial-AL05NZOI
Connecting........_____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header

# Do you have any other information from investigating this?
This is not a the Big Sur bug since I'm using the work around. I can successfully connect to some boards using the same FTDI and auto reset circuit. Using esptool under Windows never fails on any board. Mac can only connect to some boards.

# Is there any other information you can think of which will help us reproduce this problem?
I can successfully connect to it if I manually connect IO0 to GND during connection attempts

alexqzd, 2020-11-17:
Did some further testing and I found that the ESP32-WROOM-32 is the only one that I cannot use with esptool. ESP32-WROOM-32**D** works just fine.

W-M-D, 2021-04-21:
TBH windows comm ports and drivers are a pain in the neck. I wouldn't even bother.

alexqzd, 2021-04-21:
@W-M-D Windows works fine, the issue occurs only on macOS

alexqzd, 2021-05-14:
The latest FTDI driver extension update for macOS solves this issue.

closed 2021-05-14 by alexqzd