espressif/esptool #578: ESP32 automatic boot mode not working

Lzolcsi, 2020-11-28:
# Operating system
macOS Big Sur 11.0.1

# Python version
3.9.0

# What Chip
ESP32-WROOM-32E

# What development board or other hardware is the chip attached to
Custom board
[image]

# Is anything else attached to the development board, except for the serial flasher connections?
GPIO 34,36,39 are input pins with pull-up resistors
GPIO32 is connected to a DS18B20
GPIO16-19 are output pins connected to relays through an ULN2803

Standard programming circuit is implemented with two transistors for automatic resetting and programming mode selection. (DTR and RTS lines)

# Are you running esptool.py from an IDE such as Arduino or Eclipse?
Tried from both Arduino and mac OS terminal.

# Full esptool.py command line that was run:
esptool.py -p /dev/cu.usbserial-A50285BI -b 115200 read_mac   

# Full output from esptool.py
Python 3.9.0
zolcsi@MacBook-Pro ~ % esptool.py -p /dev/cu.usbserial-A50285BI -b 115200 read_mac                       
esptool.py v3.0
Serial port /dev/cu.usbserial-A50285BI
Connecting........_____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header

# Do you have any other information from investigating this?
The board is connected to a USB programmer with an FTDI chip. If I short GPIO0 to GND, the programming succeeds. That would indicate that this is an issue with the board, however, I used this board before. I had a Macbook air, with Mac OS Catalina, and everything was working fine using the same board, FTDI programmer and USB cable. As a matter of fact, if I unplug it from my new computer and plug it into my old one, it all works fine. Problems started happening when I switched to a new Macbook pro 16 inch with macOSX Big Sur. 
I can verify that serial communication is working fine, because if I upload a sketch to the chip that outputs some data over the UART port, I can see it in the serial monitor. Also I can program it from my new computer, but I need to manually put it into programming mode (short GPIO0 to GND when esptool tries to connect). So it seems like this is an issue that is related to managing the DTR and RTS lines. 
I have a scope and I successfully verified that somethings happening on the DTR and RTS lines, also the GPIO0 gets pulled down during the programming sequence, but my scope only has one channel, so I'm not able to verify the timings of these.
[image]
[image]

So the only differences that I can think of are:
The new system has a different operating system (Bug Sur vs. Catalina)
The new system has a USB type C connector and I need to use a dongle to connect the regular USB cable to it
The new system may use a different driver for the FTDI programmer
I haven't installed some piece of software that I did on my old machine

Any help and ideas would be much appreciated!

dobairoland, 2020-11-28:
Hi @Lzolcsi. Could you please post the output of your `esptool` command with the `--trace` argument?

Could you post the part of your schematics where the reset logic is shown? Or could you please compare it to an Espressif's ESP32 board. Sometimes using different values for capacitors and resistors result in that putting the chip into boot mode automatically is not working on all platforms/systems.

Lzolcsi, 2020-11-28:
Hi @dobairoland, I ran the same command with --trace and also set connection_attempts to 2 to reduce the amount. I have attached the output due to its size, hope that's OK.
[esptoolOutput.txt](https://github.com/espressif/esptool/files/5610609/esptoolOutput.txt)
When the Serial message starts reading this: "VizsLAK booting" it means that the software that is already on the ESP is booting and it's outputting debug message on the serial port. I think this means that esptool was able to reset the device but GPIO0 wasn't pulled low.
I have attached the schematics of my project (it's a pretty simple one). Reset logic is at the bottom with the two transistors and two resistorts.
[Schematic_ESP32 DIN Rail I_O module_2020-11-28_13-55-57.pdf](https://github.com/espressif/esptool/files/5610612/Schematic_ESP32.DIN.Rail.I_O.module_2020-11-28_13-55-57.pdf)

dobairoland, 2020-11-28:
Hi @Lzolcsi. Could you please try to add a capacitor between EN and GND? I don't  see one on your schematics.

Please see on the page "ESP32 Module" an example:
https://dl.espressif.com/dl/schematics/esp32_devkitc_v4-sch-20180607a.pdf

Lzolcsi, 2020-11-29:
Hi @dobairoland, adding the capacitor did solve the issue, thanks a lot!
Just out of curiosity, do you happen to know why does it work on one computer without it, and doesn't work on the other one?

closed 2020-11-29 by Lzolcsi

dobairoland, 2020-11-29:
@Lzolcsi Great! I'm happy that you solved the issue.

> Just out of curiosity, do you happen to know why does it work on one computer without it, and doesn't work on the other one?

I'm sorry I can't explain this. A little difference in timing could cause this issue. It can happen even on the same computer when switching between Linux and Windows.

It is recommended to use the reset circuitry and the additional resistors and capacitors as shown on the official Espressif schematics because those boards are already tested with various hardware and software setups.