espressif/esptool #581: "Timed out waiting for packet header" when flashing highly compressible images

neinseg, 2020-12-06:
I'm opening an issue here to collect some other issues that seem to have the same root cause:
https://github.com/espressif/esptool/issues/559
https://github.com/espressif/esptool/issues/577
https://github.com/espressif/esptool/issues/576

* Operating system:
This seems to happen on any OS.

* Python version: (`python -V` to check this)
3.9.0, but that is likely not related.

# Full esptool.py command line as run:
```
~/esptool <3 esptool.py --chip esp8266 --port /dev/ttyS6 --baud 115200 write_flash --flash_mode keep 0xdb000 highly_compressible_image.bin
esptool.py v3.0
Serial port /dev/ttyS6
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: ##:##:##:##:##:##
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Compressed 131072 bytes to 305...                                                                                                                                       Wrote 131072 bytes (305 compressed) at 0x000db000 in 0.0 seconds (effective 32779.2 kbit/s)...

A fatal error occurred: Timed out waiting for packet header 
```

# What is the expected behaviour?

```
~/esptool <3 esptool.py --chip esp8266 --port /dev/ttyS6 --baud 115200 write_flash --no-compress 0xdb000 filesystem.spiffs.bin
esptool.py v3.0
Serial port /dev/ttyS6                                                                                                                                                  Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 2c:3a:e8:42:a3:dc
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Wrote 131072 bytes at 0x000db000 in 11.4 seconds (91.7 kbit/s)...
Hash of data verified.
Leaving...
Hard resetting via RTS pin... 
```

# Do you have any other information from investigating this?

There is a multitude of issues open right now that all point to the same problem.

# Is there any other information you can think of which will help us reproduce this problem?

The issue here seems to be that in write_flash a timeout expires too early when the image is transferred too quickly. For poorly compressible images or with compression turned off, the serial port is slow enough that this timeout does not expire, and flashing works. If however one is flashing a highly compressible image such as a mostly empty filesystem image, the compressed transfer is almost instantaneous and it seems the timeout in write_flash expires before the chip is done actually writing the flash.

# Workaround

As a workaround, simply pass ``--no-compress`` to esptool.py

dobairoland, 2020-12-06:
Hi @neinseg. Are you experiencing this with the latest master branch? If not then please try it because this was addressed recently (https://github.com/espressif/esptool/commit/018b9928bc3acc07c980756b8a548077f672cbaf).

#577 is related but the other linked open issues doesn't seem to be.

d-c-d, 2020-12-16:
I just pulled the latest master code for esptool.py and it did not help, but I did have success by using the --no-compress flag
( I had tried some of the MD5 timeouts as suggested elsewhere, that did not help. )

dobairoland, 2020-12-16:
Could you please try what value of `COMP_BLOCK_WRITE_TIMEOUT_PER_MB` shown in https://github.com/espressif/esptool/commit/018b9928bc3acc07c980756b8a548077f672cbaf starts to work for you?

projectgus, 2021-02-04:
Hi @d-c-d and @neinseg,

We've updated the mechanism for setting timeouts again in 2ff4eb9a71dd4baebb23aa6e485cdb6aaac9e375 - we're not able to reproduce any timeouts with compressed flashing now, and the algorithm now takes account of files with highly variable compression rates.

Could you please try again and let us know if the flashing still fails?

Thanks,

Angus

radimkarnis, 2021-03-29:
I am closing this issue. A possible fix has been already released and no more information has been provided for a long time. Please reopen if you feel like further discussion is needed.

closed 2021-03-29 by radimkarnis