espressif/esptool #582: explain how to launch (via python -m) in readme

PPoliakoff, 2020-12-06:
**Problem**
When I try to run esptool on the command line (git command line in Windows10) with the command described in the readme.md : "esptool.py" or "esptool" I get the error " 'esptool.py' is not recognized as an internal or external command,
operable program or batch file." 

**Proposed Solution**
Please add to the readme.md that the tool can also be executed via _python -m esptool_

**Describe alternatives you've considered**
This solution is probably obvious for any python user but the esptool users are C programmers so they are not familiar with the Python command line.

dobairoland, 2020-12-07:
Hi @PPoliakoff. Thank you for pointing this out. Probably we will update the readme.

I would like to point out that `esptool.exe` is installed and `esptool` works in Windows command line. Git bash command line is a little closer to Unix since it is bash and things work there differently.

PPoliakoff, 2020-12-08:
Thank you for your explanation. I have tried in the "normal" Windows 10 cmd and the result is the same: "python -m esptool" is the only way to run esptool. This is obviously a path problem. But I have no idea where pip install has put esptool.exe
(note Python was installed using the Microsoft store which could explain the strange path settings)

Below is the trace in the "normal" windows 10 cmd:

Microsoft Windows [Version 10.0.19041.630]
(c) 2020 Microsoft Corporation. All rights reserved.

C:\Users\p>esptool
'esptool' is not recognized as an internal or external command,
operable program or batch file.

C:\Users\p>esptool.exe
'esptool.exe' is not recognized as an internal or external command,
operable program or batch file.

C:\Users\p>esptool.py
'esptool.py' is not recognized as an internal or external command,
operable program or batch file.

C:\Users\p>esptool
'esptool' is not recognized as an internal or external command,
operable program or batch file.

C:\Users\p>python esptool
python: can't open file 'esptool': [Errno 2] No such file or directory

C:\Users\p>python -m  esptool
esptool.py v3.0
usage: esptool [-h] [--chip {auto,esp8266,esp32,esp32s2,esp32s3beta2,esp32c3}] [--port PORT] [--baud BAUD]
               [--before {default_reset,no_reset,no_reset_no_sync}] [--after {hard_reset,soft_reset,no_reset}]
               [--no-stub] [--trace] [-

dobairoland, 2020-12-09:
Thank you for the valuable feedback.

> (note Python was installed using the Microsoft store which could explain the strange path settings)

In my experience, that is the worst option.

How did you install esptool? By running `pip install esptool`? In Windows CMD or git bash?

You could also do some sanity checks by running the following commands:
```
where pip
where python
where esptool
```

PPoliakoff, 2020-12-09:
Yes installing programs via the Microsoft store can sometimes lead to problems (e.g. for the Arduino IDE )
this is what "where" reports:

 Microsoft Windows [Version 10.0.19041.630]
(c) 2020 Microsoft Corporation. All rights reserved.

C:\Users\p>where python
C:\Users\p\AppData\Local\Microsoft\WindowsApps\python.exe

C:\Users\p>where pip
C:\Users\p\AppData\Local\Microsoft\WindowsApps\pip.exe

C:\Users\p>where esptool
INFO: Could not find files for the given pattern(s).

But  the exe are not in the reported directory: the directory contains some kind of links (of 0 bytes long!) towards the exe that are themself in an unknown (and probably encrypted) place :

C:\Users\p\AppData\Local\Microsoft\WindowsApps>dir P*.exe

12/06/2020  10:40 PM                 0 pip.exe
12/06/2020  10:40 PM                 0 pip3.8.exe
12/06/2020  10:40 PM                 0 pip3.exe
12/06/2020  10:40 PM                 0 python.exe
12/06/2020  10:40 PM                 0 python3.8.exe
12/06/2020  10:40 PM                 0 python3.exe
12/06/2020  10:40 PM                 0 pythonw.exe
12/06/2020  10:40 PM                 0 pythonw3.8.exe
12/06/2020  10:40 PM                 0 pythonw3.exe
               9 File(s)              0 bytes
               0 Dir(s)  103,354,638,336 bytes free

dobairoland, 2020-12-18:
@PPoliakoff I've found this issue: https://stackoverflow.com/questions/57421669/question-about-pip-using-python-from-windows-store. It seems that the packages installed for a Python installed from Windows Store cannot be run directly. Could you please use a Python installed from python.org?

I will update the main README.

closed 2020-12-22 by espressif-bot