espressif/esptool #583: firmware integrated esp hardware flasher

123swk123, 2020-12-09:
**Is your feature request related to a problem? Please describe.**
- for firmware distribution it would be good to integrate firmware inside the esptool flasher utilty
- allow in memory firmware uploader there by possibility to protect the firmware with encryption
- easy usability for the manufacturers by exposing only bare minimum options like serial port number

**Describe the solution you'd like**
- Tool allows firmware of esp8266, esp32 based hardware to be encapsulated in the flasher executable. Which makes easy secured firmwware distribution rather than distributing the build binary. 
- very simple command line interface, thus hidding the complex esptool options from user
- Easy to integrate with esp-idf build process

It uses [esptool.py](https://github.com/espressif/esptool) flash programming feature with a modification in `<address> <filename>` pair command line option of **write_flash** command. When `*@$` is prefixed in **filename** parameter its recognized as base64 encoded zlib compressed BLOB string, this way it allows to use in memory file IO
to upload the firmware to esp hardware in a secure way.

**Describe alternatives you've considered**
using shell script or powershell script this can be achieved but in less efficient way.

**Additional context**
There is working version of this idea here.
https://github.com/123swk123/esptool/tree/feature/firmware-integrated

xfdr0805, 2021-01-04:
> https://github.com/123swk123/esptool/tree/feature/firmware-integrated
thanks  for your good job,but  why my build' program can't work fine?
upx=True :

[image]

upx=Fasle:
[image]

123swk123, 2021-01-04:
You are right upx should be false its currently broken, in my setup I removed upx executable from my pc so during building pyinstaller automatically turned off the upx. So set the upx to false for now and try it. ~SwK

xfdr0805, 2021-01-04:
> You are right upx should be false its currently broken, in my setup I removed upx executable from my pc so during building pyinstaller automatically turned off the upx. So set the upx to false for now and try it.

thanks but not work    [Errno 22] Invalid argument

output log:

C:\Users\101367\Desktop\esptool-feature-firmware-integrated\example\dist>flasher.exe
[14148] PyInstaller Bootloader 3.x
[14148] LOADER: executable is C:\Users\101367\Desktop\esptool-feature-firmware-integrated\example\dist\flasher.exe
[14148] LOADER: homepath is C:\Users\101367\Desktop\esptool-feature-firmware-integrated\example\dist
[14148] LOADER: _MEIPASS2 is NULL
[14148] LOADER: archivename is C:\Users\101367\Desktop\esptool-feature-firmware-integrated\example\dist\flasher.exe
[14148] LOADER: Extracting binaries
[14148] LOADER: Executing self as child
[14148] LOADER: set _MEIPASS2 to C:\Users\101367\AppData\Local\Temp\_MEI141482
[14148] LOADER: Setting up to run child
[14148] LOADER: Creating child process
[14148] LOADER: Waiting for child process to finish...
[14260] PyInstaller Bootloader 3.x
[14260] LOADER: executable is C:\Users\101367\Desktop\esptool-feature-firmware-integrated\example\dist\flasher.exe
[14260] LOADER: homepath is C:\Users\101367\Desktop\esptool-feature-firmware-integrated\example\dist
[14260] LOADER: _MEIPASS2 is C:\Users\101367\AppData\Local\Temp\_MEI141482
[14260] LOADER: archivename is C:\Users\101367\Desktop\esptool-feature-firmware-integrated\example\dist\flasher.exe
[14260] LOADER: SetDllDirectory(C:\Users\101367\AppData\Local\Temp\_MEI141482)
[14260] LOADER: Already in the child - running user's code.
[14260] LOADER: manifestpath: C:\Users\101367\AppData\Local\Temp\_MEI141482\flasher.exe.manifest
[14260] LOADER: Activation context created
[14260] LOADER: Activation context activated
[14260] LOADER: ucrtbase.dll found: C:\Users\101367\AppData\Local\Temp\_MEI141482\ucrtbase.dll
[14260] LOADER: Python library: C:\Users\101367\AppData\Local\Temp\_MEI141482\python38.dll
[14260] LOADER: Loaded functions from Python library.
[14260] LOADER: Manipulating environment (sys.path, sys.prefix)
[14260] LOADER: sys.prefix is C:\Users\101367\AppData\Local\Temp\_MEI141482
[14260] LOADER: Pre-init sys.path is C:\Users\101367\AppData\Local\Temp\_MEI141482\base_library.zip;C:\Users\101367\AppData\Local\Temp\_MEI141482
[14260] LOADER: Setting runtime options
[14260] LOADER: Bootloader option: pyi-windows-manifest-filename flasher.exe.manifest
[14260] LOADER: Initializing python
[14260] LOADER: Overriding Python's sys.path
[14260] LOADER: Post-init sys.path is C:\Users\101367\AppData\Local\Temp\_MEI141482\base_library.zip;C:\Users\101367\AppData\Local\Temp\_MEI141482
[14260] LOADER: Setting sys.argv
[14260] LOADER: setting sys._MEIPASS
[14260] LOADER: importing modules from CArchive
[14260] LOADER: extracted struct
[14260] LOADER: callfunction returned...
[14260] LOADER: extracted pyimod01_os_path
[14260] LOADER: callfunction returned...
[14260] LOADER: extracted pyimod02_archive
[14260] LOADER: callfunction returned...
[14260] LOADER: extracted pyimod03_importers
[14260] LOADER: callfunction returned...
[14260] LOADER: Installing PYZ archive with Python modules.
[14260] LOADER: PYZ archive: PYZ-00.pyz
[14260] LOADER: Running pyiboot01_bootstrap.py
[14260] LOADER: Running pyi_rth_multiprocessing.py
[14260] LOADER: Running flasher.py
usage: esptool write_flash [-h] [--erase-all] [--flash_freq {keep,40m,26m,20m,80m}]
                           [--flash_mode {keep,qio,qout,dio,dout}] [--flash_size FLASH_SIZE]
                           [--spi-connection SPI_CONNECTION] [--no-progress] [--verify] [--encrypt]
                           [--ignore-flash-encryption-efuse-setting] [--compress | --no-compress]
                           <address> <filename> [<address> <filename> ...]
esptool write_flash: error: argument <address> <filename>: [Errno 22] Invalid argument: '*@$eNpbFcDIxDCBgYEhgYEhr6yYAR2sCmBkZPgAZAgwMBRkVMZn5mWWoMqDSEYg5mdIS0wuyS+qRNH/fxSMglEwCkbBKBgFo2AUjIJRMGgAANqYoaw='
[14148] LOADER: Back to parent (RC: 2)
[14148] LOADER: Doing cleanup
[14148] LOADER: Freeing archive status for C:\Users\101367\Desktop\esptool-feature-firmware-integrated\example\dist\flasher.exe

C:\Users\101367\Desktop\esptool-feature-firmware-integrated\example\dist>

123swk123, 2021-01-04:
Are you running the pyinstaller from the example folder or root folder?

It looks like pyinstaller is not using the modified esptool.py instead its using your original esptool.py.

This line takes care of the issue which you are facing.
https://github.com/123swk123/esptool/blob/b9cc26bdad96c4fd1d53e1e654cbaca5ad27c5be/esptool.py#L3793

You should copy the example folder content to root and run the pyinstaller from root folder.

123swk123, 2021-01-04:
In the example folder there is powershell script gen_flasher.ps1

In that set the $env:ESP_TOOL_PATH to the root folder path you cloned from github.
$env:ESP_PRJ_PATH to example folder path

```
python ..\rom_blob_gen\rom_blob_gen.py .\build\gpio.bin app_rom.py
python ..\rom_blob_gen\rom_blob_gen.py .\build\bootloader\bootloader.bin boot_rom.py
python ..\rom_blob_gen\rom_blob_gen.py .\build\partition_table\partition-table.bin part_rom.py 

$env:ESP_TOOL_PATH='C:\Users\self\esptool_flasher'
$env:ESP_PRJ_PATH='C:\Users\self\esptool_flasher\example'
pyinstaller --onefile flasher.spec
```

xfdr0805, 2021-01-04:
> You should copy the example folder content to root and run the pyinstaller from root folder.
Yeah，now work fine,thank you very much.

123swk123, 2021-01-04:
Congrats! Good to know you got it working.

radimkarnis, 2021-01-17:
Hello @123swk123,
thank you very much for this feature request.

There are already plenty of tools that can make an encrypted self-extracting archive that runs a script file, for example 7zip can do this with their -sfx feature. Adding it to esptool itself would be a lot of extra complexity for us to maintain.

What’s more, the "encryption" you have in mind is not really a security feature; the firmware has to be plaintext when esptool actually processes it so this is trivial to reverse.

To be honest, we ale already working on a feature which will allow users to have an easy standalone executable that they can embed into whatever self-extracting scripting thing they would like to use.

Radim

closed 2021-01-19 by radimkarnis