espressif/esptool #584: Having trouble read protecting fuses

HalexRomagno, 2020-12-11:
Hi,

I'm trying to use the new batch mode to write fuses as well as read/write protect them.
Here is what I'm doing and the result:

_Command:_
`py -3 -m espefuse --port COM3 --before default_reset --do-not-confirm burn_efuse ABS_DONE_0 1 FLASH_CRYPT_CONFIG 0xF FLASH_CRYPT_CNT 1 CONSOLE_DEBUG_DISABLE 1 DISABLE_DL_ENCRYPT 1 DISABLE_DL_DECRYPT 1 DISABLE_DL_CACHE 1  JTAG_DISABLE 1 WR_DIS 1412 RD_DIS 11`

_Response:_
```
Connecting.....
Detecting chip type... ESP32
espefuse.py v3.0
The efuses to burn:
  from BLOCK0
     - ABS_DONE_0
     - FLASH_CRYPT_CONFIG
     - FLASH_CRYPT_CNT
     - CONSOLE_DEBUG_DISABLE
     - DISABLE_DL_ENCRYPT
     - DISABLE_DL_DECRYPT
     - DISABLE_DL_CACHE
     - JTAG_DISABLE
     - WR_DIS
     - RD_DIS

Burning efuses:

    - 'ABS_DONE_0' (Secure boot V1 is enabled for bootloader image) 0b0 -> 0b1

    - 'FLASH_CRYPT_CONFIG' (Flash encryption config (key tweak bits)) 0x0 -> 0xf

    - 'FLASH_CRYPT_CNT' (Flash encryption mode counter) 0b0000000 -> 0b0000001

    - 'CONSOLE_DEBUG_DISABLE' (Disable ROM BASIC interpreter fallback) 0b1 -> 0b1
	The same value for CONSOLE_DEBUG_DISABLE is already burned. Do not change the efuse.

    - 'DISABLE_DL_ENCRYPT' (Disable flash encryption in UART bootloader) 0b0 -> 0b1

    - 'DISABLE_DL_DECRYPT' (Disable flash decryption in UART bootloader) 0b0 -> 0b1

    - 'DISABLE_DL_CACHE' (Disable flash cache in UART bootloader) 0b0 -> 0b1

    - 'JTAG_DISABLE' (Disable JTAG) 0b0 -> 0b1

    - 'WR_DIS' (Efuse write disable mask) 0x0180 -> 0x0584

    - 'RD_DIS' (Efuse read disable mask) 0x3 -> 0xb

Check all blocks for burn...
idx, BLOCK_NAME,          Conclusion
[00] BLOCK0               is not empty
	(written ): 0x0000000400000000000011340000a0000069c44f3360f85d00030180
	(to write): 0x000003d0f000000000000000000000000000000000000000001b0584
	(coding scheme = NONE)
. 
This is an irreversible operation!

A fatal error occurred: Burn BLOCK0 ([]) was not successful
```

What seems to be the problem is read protecting fuses. If I remove "RD_DIS 11" from the command above, it works successfully.
But then, if I run "burn_efuse RD_DIS 11" or even "read_protect_efuse FLASH_CRYPT_CONFIG", I get the same error as previously when everything is bundled together:

```
Connecting....
Detecting chip type... ESP32
espefuse.py v3.0
Permanently read-disabling efuses CODING_SCHEME, KEY_STATUS, FLASH_CRYPT_CONFIG, BLK3_PART_RESERVE

Check all blocks for burn...
idx, BLOCK_NAME,          Conclusion
[00] BLOCK0               is not empty
	(written ): 0x000003d4f0000000000002350000a0000036c44f3360f84500130584
	(to write): 0x00000000000000000000000000000000000000000000000000080000
	(coding scheme = NONE)
. 
This is an irreversible operation!

A fatal error occurred: Burn BLOCK0 ([]) was not successful
```

Weirdly, if I then run "espefuse summary", RD_DIS is indicating the value 11 which means BLOCK0 was indeed modified and read protecting the fuse actually worked.
If I run "burn_efuse RD_DIS 11" or "read_protect_efuse FLASH_CRYPT_CONFIG" again, the operation will succeed because it says no change is needed.
Am I doing something wrong, perhaps in the order I do things, or is there some kind of bug in esptool 3.0?

# Operating system
Windows 7 SP1 64-bit
# Python version
3.8.6
# What Chip
ESP32-WROOM32

# What development board or other hardware is the chip attached to
FT231XS-R

# Is anything else attached to the development board, except for the serial flasher connections?
No

# Are you running esptool.py from an IDE such as Arduino or Eclipse?
Windows CLI

radimkarnis, 2020-12-14:
Hello @HalexRomagno,
thanks for reporting this behavior. I don't see anything wrong in your procedure, the FLASH_CRYPT_CONFIG should indeed be read protected by setting 'RD_DIS'  0x3 -> 0xb.
We will look into this issue.

Radim

radimkarnis, 2021-03-11:
Hello @HalexRomagno,

there was nothing wrong with your procedure.
The error message happens because you try to burn `FLASH_CRYPT_CONFIG` and read-disable it at the same time. Both of these actions happen correctly, but when espefuse.py reads the efuse block back to check if the burn was successful, `FLASH_CRYPT_CONFIG` is already read-protected and reported back as 0x0 (while espefuse expects to see 0xf).

Thank you for reporting this, we will fix this. In the meantime, I advise you not to write and read-disable the same efuse in one command.

closed 2021-03-15 by espressif-bot