espressif/esptool #586: esptool.py: A fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header

M1S2, 2020-12-16:
# Operating system
Windows10

# Python version
Python 3.9.1
Can run `python -V` to check this.

# What Chip

ESP12E (ESP8266)

# What development board or other hardware is the chip attached to
plain module on breadboard
Connected like described in:
https://github.com/espressif/esptool#entering-the-bootloader
and
https://www.hackster.io/brian-lough/3-simple-ways-of-programming-an-esp8266-12x-module-c514ee (Chapter: Programming with a USB to serial adapter)

# Is anything else attached to the development board, except for the serial flasher connections?
No other pins connected

# Are you running esptool.py from an IDE such as Arduino or Eclipse?
No IDE, Windows command prompt

# Full esptool.py command line that was run:
esptool.py --port COM7 --baud 9600 read_mac

# Full output from esptool.py

C:\Users\Dominik>esptool.py --port COM7 --baud 9600 read_mac
esptool.py v3.0
Serial port COM7
Connecting........_____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header

# Do you have any other information from investigating this?
- Also tried to communicate with the device via Arduino IDE, resulting in the same output. Therefore moved to the esptool to reduce external components.
- No output is show on a serial monitor when entering the bootloader mode or doing everything else
- Already tried everything in https://github.com/espressif/esptool/#troubleshooting
- Tried different power supplies with different ratings (up to an ATX supply)

# Is there any other information you can think of which will help us reproduce this problem?
Tried on two different PCs with the same result.

JayZar21, 2020-12-17:
Hi all,
I've the same issue and it seems to be very frequent:

`=> CALL => python3 -m esptool --chip esp32 --port /dev/ttyDUT --baud 2000000 --before default_reset --after hard_reset erase_flash
esptool.py v3.0
Serial port /dev/ttyDUT
Connecting........_____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header`

This is the command for erasing an ESP32 module, but the same can happen while flashing (less frequent).
The PC is a Win10 LTSC with a standard USB cable.
We put another baudrate (460800) and it happens with a lowered frequency... But it is not the real solution!

M1S2, 2020-12-17:
@JayZar21 You wrote "it seems to be very frequent". Are you able to connect to the device sometimes?

dobairoland, 2020-12-18:
Hi @M1S2.

> No output is show on a serial monitor when entering the bootloader mode or doing everything else

I don't think anything can be done with ESPTOOL at this level. This has to be fixed on the hardware connection level. I can suggest to find the schematics for a development board using the same module and check your connection.

dobairoland, 2020-12-18:
@JayZar21 Please open a separate issue because your issue is probably not related to this one. Unfortunately, the same error message is produced in different cases. Please provide also the output with the `--trace` argument.

JayZar21, 2020-12-18:
Ok. Yes, the issue is not always present (we can say at a 1-3% of the times). I must think how to use the trace because the script is used in a full build integration flow and it is not the only operation.

Thank you.

M1S2, 2020-12-18:
I wanted to try the schematic from the NodeMCU development board (https://github.com/nodemcu/nodemcu-devkit-v1.0/blob/master/NODEMCU_DEVKIT_V1.0.PDF) as suggested. But this is using the RTS and DTR lines of the USB-to-Serial converter. My converter (https://www.christians-shop.de/Seriell-Konverter-CP2102-USB-20-zu-UART-TTL-5V-33V-Adapter) doesn't have this lines. (only TX, RX). So there is no opportunity to try this.

M1S2, 2020-12-18:
I tried to measure the TX and RX lines with an scope while executing the above command.
On the TX line I see some pulses then pause, some pulses then pause, ... Is this correct?
On the RX line I only see a constant 3,3 V level. The line remains high even if I reset the device. Shouldn't there be some pulses belonging to the bootloader output on startup?

dobairoland, 2020-12-18:
@M1S2 The DTR and RTS signals are used to reset the chip or put into download mode by ESPTOOL. Are you putting the chip into download mode manually?

M1S2, 2020-12-18:
Yes, I used two buttons to bring the device to the programming mode like described here:
https://www.hackster.io/brian-lough/3-simple-ways-of-programming-an-esp8266-12x-module-c514ee (Chapter: Programming with a USB to serial adapter)

M1S2, 2020-12-18:
I've also tried the command `esptool.py --trace --baud 9600 --port COM4 read_mac` with the `--trace` option.
<details><summary>Here's the full output</summary>
<p>

```
C:\Users\Dominik>esptool.py --trace --baud 9600 --port COM4 read_mac
esptool.py v3.0
Serial port COM4
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +2.033 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.113 Timed out waiting for packet header
.TRACE +0.116 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.001 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.145 Timed out waiting for packet header
.TRACE +0.118 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.001 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.112 Timed out waiting for packet header
.TRACE +0.117 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.002 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.111 Timed out waiting for packet header
.TRACE +0.120 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.001 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.112 Timed out waiting for packet header
.TRACE +1.902 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.002 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.111 Timed out waiting for packet header
_TRACE +0.117 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.001 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
[... 548 lines cut ...]
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.113 Timed out waiting for packet header
_TRACE +0.117 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.002 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.111 Timed out waiting for packet header
_TRACE +0.116 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.001 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.117 Timed out waiting for packet header
_

A fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header
```
</p></details>

dobairoland, 2020-12-18:
@M1S2 If you connect a terminal program to the port instead of ESPTOOL and reset the chip into bootloader mode then are you able to receive something? If the RX line is constant then probably not. And in that case, as I pointed out earlier, it is not an ESPTOOL issue but a hardware one.

M1S2, 2020-12-18:
I've used Realterm as terminal program with various baud rate settings. No matter what configuration I choosed, there was no output on the RX line (only constant 3,3V) on chip reset. So it really seems to be a hardware issue. 
Is there something I can check to find the cause of this issue? I'm currently running out of ideas...

dobairoland, 2020-12-18:
Do you have a good enough power source?

M1S2, 2020-12-18:
I tried it with different power supplies including an ATX supply that is able to supply 14A at the 3,3V line. I added an 100µF and 100 nF capacitor to the power rails to filter out any high frequent parts. On the scope the 3,3V line is constant without any changes. So power should be ok.

dobairoland, 2020-12-21:
Hi @M1S2. I'm sorry I cannot give you a straight answer. I always used esptool with a devboard and not a module directly.

So I'm just guessing with this one. Please try to find other sources on the Internet how to connect the module in a reliable way. For example, I found https://electronut.in/an-iot-project-with-esp8266/ where there are resistors connected to TxD. But that is RxD on the ESP side so probably will not help.

I would also like to recommend to buy a simple and cheep development board which might help you to debug these first issues.

I'm closing this because it doesn't seem to be an ESPTOOL issue.

closed 2020-12-21 by dobairoland