espressif/esptool #587: [ESP32s2-Wrover] [Arduino IDE] A fatal error occurred: Invalid head of packet (0x50)

LightningTerry, 2020-12-17:
### Operating system
Windows 10

### Python version
Python3.9.0

### What Chip
ESP32-S2-WROVER

### Is anything else attached to the board, except for the serial flasher connections?
No.

### Are you running esptool.py from an IDE such as Arduino or Eclipse?
Yes, Arduino IDE

### Output from Error Message
Arduino: 1.8.14 Hourly Build 2020/10/09 12:33 (Windows 10), Board: "ESP32 Wrover Module, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), QIO, 40MHz, 115200, None"

Sketch uses 259225 bytes (19%) of program storage space. Maximum is 1310720 bytes.

Global variables use 15436 bytes (4%) of dynamic memory, leaving 312244 bytes for local variables. Maximum is 327680 bytes.

esptool.py v2.6

Serial port COM5

Connecting....

Chip is unknown ESP32 (revision 0)

Features: WiFi, BT, Single Core, VRef calibration in efuse, BLK3 partially reserved, Coding Scheme 3/4

MAC: 80:53:05:a1:8f:9d

Uploading stub...

Running stub...

A fatal error occurred: Invalid head of packet (0x50)

A fatal error occurred: Invalid head of packet (0x50)

<details open>
<summary>Full Report with "Show verbose output during compilation"</summary>
<br>
Arduino: 1.8.14 Hourly Build 2020/10/09 12:33 (Windows 10), Board: "ESP32 Wrover Module, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), QIO, 40MHz, 115200, None"

C:\Program Files\arduino-nightly\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files\arduino-nightly\hardware -hardware C:\Users\user\AppData\Local\Arduino15\packages -tools C:\Program Files\arduino-nightly\tools-builder -tools C:\Program Files\arduino-nightly\hardware\tools\avr -tools C:\Users\user\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files\arduino-nightly\libraries -libraries C:\Users\user\Documents\Arduino\libraries -fqbn=esp32:esp32:esp32wrover:PartitionScheme=default,FlashMode=qio,FlashFreq=40,UploadSpeed=115200,DebugLevel=none -vid-pid=0403_6001 -ide-version=10814 -build-path C:\Users\user\AppData\Local\Temp\arduino_build_435839 -warnings=none -build-cache C:\Users\user\AppData\Local\Temp\arduino_cache_845918 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\user\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=C:\Users\user\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.mkspiffs.path=C:\Users\user\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\user\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.esptool_py.path=C:\Users\user\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.esptool_py-2.6.1.path=C:\Users\user\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -verbose C:\Users\user\Documents\Arduino\sketch_dec17a\sketch_dec17a.ino

C:\Program Files\arduino-nightly\arduino-builder -compile -logger=machine -hardware C:\Program Files\arduino-nightly\hardware -hardware C:\Users\user\AppData\Local\Arduino15\packages -tools C:\Program Files\arduino-nightly\tools-builder -tools C:\Program Files\arduino-nightly\hardware\tools\avr -tools C:\Users\user\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files\arduino-nightly\libraries -libraries C:\Users\user\Documents\Arduino\libraries -fqbn=esp32:esp32:esp32wrover:PartitionScheme=default,FlashMode=qio,FlashFreq=40,UploadSpeed=115200,DebugLevel=none -vid-pid=0403_6001 -ide-version=10814 -build-path C:\Users\user\AppData\Local\Temp\arduino_build_435839 -warnings=none -build-cache C:\Users\user\AppData\Local\Temp\arduino_cache_845918 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\user\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=C:\Users\user\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.mkspiffs.path=C:\Users\user\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\user\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.esptool_py.path=C:\Users\user\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.esptool_py-2.6.1.path=C:\Users\user\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -verbose C:\Users\user\Documents\Arduino\sketch_dec17a\sketch_dec17a.ino

Using board 'esp32wrover' from platform in folder: C:\Users\user\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4

Using core 'esp32' from platform in folder: C:\Users\user\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4

Detecting libraries used...

"C:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX "-IC:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/config" "-IC:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/app_trace" "-IC:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/app_update" "-IC:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/asio" "-IC:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/bootloader_support" "-IC:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/bt" "-IC:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/coap" "-IC:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/console" "-IC:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/driver" "-IC:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp-tls" "-IC:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp32" "-IC:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_adc_cal" "-IC:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_event" "-IC:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_http_client" "-IC:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_http_server" "-IC:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_https_ota" "-IC:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_ringbuf" "-IC:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/ethernet" "-IC:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/expat" "-IC:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/fatfs" "-IC:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/freemodbus" "-IC:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/freertos" "-IC:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/heap" "-IC:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/idf_test" "-IC:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/jsmn" "-IC:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/json" "-IC:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/libsodium" "-IC:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/log" "-IC:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/lwip" "-IC:\\Us [... cut ...]
A fatal error occurred: Invalid head of packet (0x50)

A fatal error occurred: Invalid head of packet (0x50)

</details>

### Is there any other information you can think of which will help us?
I installed esptool.py 3.0 with the cmd terminal, but it still uses esptool.py 2.6. This might be, because the Additional Boards Manager URLs, where I used [https://dl.espressif.com/dl/package_esp32_index.json](https://dl.espressif.com/dl/package_esp32_index.json), but I am not sure if this is, what causes the problem.

radimkarnis, 2020-12-18:
Hello @LightningTerry,

It seems like you are trying to flash a ESP32-S2 chip with a ESP32 board selected in your Arduino IDE. 
To enable ESP32-S2 support, please follow the `Using Arduino IDE with the development repository` instructions on [this page](https://github.com/espressif/arduino-esp32#installation-instructions) or follow [this tutorial](https://learn.adafruit.com/adafruit-metro-esp32-s2/arduino-ide-setup). 

Let me know how it went!

Radim

pieterbro05, 2020-12-21:
i have the exact same problem, and i am using the ESP32S2 board in arduino

radimkarnis, 2020-12-21:
Hello @pieterbro05, would you mind sharing more information? You can use @LightningTerry 's report as a template. Thank you

LightningTerry, 2020-12-22:
@radimkarnis thank you very much, your answer solved my problem. I just needed to use the ESP32S2 Dev Module in the Board Settings, instead of the Adafruit Metro ESP32-S2 suggested in the  tutorial. After that it worked perfectly. Maybe you wanna try this too @pieterbro05?

pieterbro05, 2020-12-22:
Operating system
Windows 10

Python version
Python3.9.0

What Chip
ESP32-S2-SOALA-1_v1.2

Is anything else attached to the board, except for the serial flasher connections?
No.

Are you running esptool.py from an IDE such as Arduino or Eclipse?
Yes, Arduino IDE

Output from Error Message
Arduino: 1.8.10 (Windows 10), Board:"ESP32S2 Dev Module, UART0, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi), QIO, 80MHz, 4MB (32Mb), 115200, None"

Arduino: 1.8.10 (Windows 10), Board:"ESP32S2 Dev Module, UART0, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi), QIO, 80MHz, 4MB (32Mb), 115200, None"

[SPIFFS] data   : C:\Users\11903162\Documents\Arduino\Juiste_ESP_File\data
[SPIFFS] start  : 2686976
[SPIFFS] size   : 1472
[SPIFFS] page   : 256
[SPIFFS] block  : 4096
/index.html

[SPIFFS] upload : C:\Users\11903162\AppData\Local\Temp\arduino_build_685070/Juiste_ESP_File.spiffs.bin
[SPIFFS] address: 2686976
[SPIFFS] port   : COM9
[SPIFFS] speed  : 115200
[SPIFFS] mode   : dio
[SPIFFS] freq   : 80m

esptool.py v2.6

Serial port COM9

Connecting......

Chip is unknown ESP32 (revision 1)

Features: WiFi, Dual Core, VRef calibration in efuse, BLK3 partially reserved, Coding Scheme Repeat (UNSUPPORTED)

MAC: 17:cf:46:ba:ea:19

Uploading stub...

Running stub...

A fatal error occurred: Invalid head of packet (0x50)

SPIFFS Upload failed!
[SPIFFS] data   : C:\Users\11903162\Documents\Arduino\Juiste_ESP_File\data
[SPIFFS] start  : 2686976
[SPIFFS] size   : 1472
[SPIFFS] page   : 256
[SPIFFS] block  : 4096
/index.html

[SPIFFS] upload : C:\Users\11903162\AppData\Local\Temp\arduino_build_685070/Juiste_ESP_File.spiffs.bin
[SPIFFS] address: 2686976
[SPIFFS] port   : COM9
[SPIFFS] speed  : 115200
[SPIFFS] mode   : dio
[SPIFFS] freq   : 80m

esptool.py v2.6

Serial port COM9

Connecting....

Chip is unknown ESP32 (revision 1)

Features: WiFi, BT, Single Core, VRef calibration in efuse, BLK3 partially reserved, Coding Scheme Repeat (UNSUPPORTED)

MAC: 17:cf:c6:fa:ca:19

Uploading stub...

Running stub...

A fatal error occurred: Invalid head of packet (0x50)

SPIFFS Upload failed!

the selected board in the arduino IDE is the esp32s2, wich is the one i am using. In the error message it says it uses esptool.py v2.6 even though i have version 3.0 installed. I feel like i have already tried everything on the internet since i have been trying to solve the problem for almost 3 days now, any help would be really aprreciated!

radimkarnis, 2020-12-22:
@LightningTerry I'm glad I could help.

@pieterbro05 judging from esptool detecting the chip as unknown ESP32 (revision 1) and using esptool.py v2.6 (you should be seeing 3.0-dev), I guess there is something wrong with your procedure. Are you sure you've checked out the IDF v4.2 branch (`git checkout --track origin/"idf-release/v4.2"` and that you are choosing the ESP32S2 Dev Module from the ESP32 Arduino (in Sketchbook) folder? If not, please follow the tutorials mentioned above.

pieterbro05, 2020-12-22:
no i havent checked out the v4.2 branch,thats probably it. is there a tutorial anywhere about this?

pieterbro05, 2020-12-22:
hey, i just followed your tutorial, but it doesnt seem to change anything. I can communicate with my board and upload code to it. The problem is that i cant do a data sketch upload (this uses another external software) to upload a html file to my esp.

radimkarnis, 2020-12-22:
sorry @pieterbro05, but if you are able to communicate with the board and upload to it, then this is definitely not an esptool issue. You could try nuking your Arduino IDE and following the mentioned tutorials with a new installation. Btw, which data sketch are you talking about?

pieterbro05, 2020-12-22:
ok ill try that, i was talking about the sketch that was mentioned in this tutorial https://shawnhymel.com/1882/how-to-create-a-web-server-with-websockets-using-an-esp32-in-arduino/

pieterbro05, 2020-12-22:
nothing is working is there no solution or someone you could reffer me too? im kinda getting hopeless...

radimkarnis, 2020-12-22:
@pieterbro05 the example you’ve sent me is for ESP32 boards. You have to try a different one. Sorry, I can’t help you with your issue.

I’m now closing this issue because it is not related to esptool.

closed 2020-12-22 by radimkarnis

dobairoland, 2020-12-22:
@pieterbro05 I never used Arduino but the plugin in the tutorial has hard coded call with ``--chip esp32``: https://github.com/me-no-dev/arduino-esp32fs-plugin/blob/master/src/ESP32FS.java#L371 That obviously cannot work with an ESP32-S2.

So in general, if you want to migrate a tutorial to a different chip then you have to check all linked projects if they are supporting the given chip. To be specific for ESPTOOL, they have to use a version of esptool which supports the given chip and support switching between chips.