espressif/esptool #588: esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header

Gamgee16, 2020-12-18:
Hi, i've bought 3 esp8266 NodeMCU Amica from AZ-Delivery.
I've installed the CP210X driver from Silicon Labs on my Mac and prepared my arduino Ide (latest version).
I'm using a serial port called "/dev/cu.Bluetooth-Incoming-Port" and i've already installed the board NodeMCU in the Arduino Ide. After connecting the board to the USB and loading a simple blink sketch this is what shows up:

esptool.py v2.8
Serial port /dev/cu.Bluetooth-Incoming-Port
Connecting........_____....._____....._____....._____....._____....._____.....____Traceback (most recent call last):
  File "/Users/alviseparenti/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/tools/upload.py", line 65, in <module>
    esptool.main(cmdline)
  File "/Users/alviseparenti/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/tools/esptool/esptool.py", line 2890, in main
    esp.connect(args.before)
  File "/Users/alviseparenti/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/tools/esptool/esptool.py", line 483, in connect
    raise FatalError('Failed to connect to %s: %s' % (self.CHIP_NAME, last_error))
esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header
esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header

The board seems talking with my Mac but it goes no further. I have no circuitry around my NodeMCU and even pressing the flash button does not change anything
can anyone help?

radimkarnis, 2020-12-18:
Hello @Gamgee16,

would you mind trying a different port? `/dev/cu.Bluetooth-Incoming-Port` is not a USB serial port. You should be using `/dev/tty.usbserial` or something similar.

If you don't see any suitable port, try reinstalling your drivers or checking your USB cable. (some are just power cables with no data lines)

radimkarnis, 2020-12-29:
I am closing this issue because it seems like it is not relevant anymore. Please reopen if you think further discussion is needed.

closed 2020-12-29 by radimkarnis