espressif/esptool #591: [ESP32-C3] A fatal error occurred: Invalid head of packet (0x47)

zhuolingmin, 2020-12-24:
**Operating system**
Windows

**What Chip**
ESP32-C3

**development tools**
visual code

After I use Visual Code to compile and download the program, the following prompt appears:
[image]

But after I use ESP32 DOWNLOAD TOOL to test the download, the following prompt appears:

[image]

I hope everyone can help to check what is the problem and can there be a solution? Thank you all in advance.

dobairoland, 2020-12-24:
Hi @zhuolingmin. Are you sure you have an ESP32-C3? ESP-IDF v3.3 supports only ESP32.

Do you see the complete invoked esptool command? Can you copy it and run from console using the latest master branch from this repo?

zhuolingmin, 2020-12-24:
> Hi @zhuolingmin. Are you sure you have an ESP32-C3? ESP-IDF v3.3 supports only ESP32.
> 
> Do you see the complete invoked esptool command? Can you copy it and run from console using the latest master branch from this repo?

what is the esptool?

zhuolingmin, 2020-12-24:
> > Hi @zhuolingmin. Are you sure you have an ESP32-C3? ESP-IDF v3.3 supports only ESP32.
> > Do you see the complete invoked esptool command? Can you copy it and run from console using the latest master branch from this repo?
> 
> what is the esptool? Do you mean that the ESP-IDF V3.3 isn't support ESP32-C3? only ESP-IDF Release v4.2 is support on ESP32-C3.

dobairoland, 2020-12-24:
> what is the esptool?

`esptool.py` is provided by this repo where you opened this issue.

> Do you mean that the ESP-IDF V3.3 isn't support ESP32-C3? only ESP-IDF Release v4.2 is support on ESP32-C3.

I highly doubt that you have an ESP32-C3 because that is in ALPHA stage and I'm not aware of any customers receiving them. ESP32-S2 is supported by v4.2. ESP32-C3 will be supported probably by v4.3.

zhuolingmin, 2020-12-24:
> `esptool.py` is provided by this repo where you opened this issue.
>
>
> I highly doubt that you have an ESP32-C3 because that is in ALPHA stage and I'm not aware of any customers receiving them. [...]

yes , now i have a esp32-c3 board , you mean that ESP32-C3 will be supported probably by v4.3. but now the latest SDK is ESP-IDF Release v4.2, so what can i do?

dobairoland, 2020-12-24:
@zhuolingmin ESP32-C3 is currently not supported by ESP-IDF. Please contact the source of your board for further assistance.

closed 2020-12-24 by dobairoland

wujiangang, 2020-12-24:
@zhuolingmin If you got the board from ESP business team, please contact them for more information. Thanks.

zhuolingmin, 2020-12-24:
I have just contacted and they have not released the SDK for this board. Thank you for your help.

annabelletian, 2020-12-24:
@zhuolingmin Hi, our business support team have received your request, and we will reach you directly for further support. If you have any other questions, please feel free to contact us at sales@espressif.com.

dobairoland, 2020-12-24:
Thank you @wujiangang and @annabelletian for your help with this.