espressif/esptool #592: Failed to connect to Espressif device: Timed out waiting for packet header. Error on different PC.

adateht, 2020-12-27:
Hello,

I have been succesfully using custom esp8266 programmer pcb based on CH340 for flashing/reading ESP8266 (also custom pcb). Recently I have switched to new pc and Im not able to connect to device anymore.

CH340 driver is ok -> arduino boards with CH340 I have, works correctly. Also I can see device connected in dev manager.
SW configuration (python, esptool, drivers) are same as on old pc

Error:
Serial port COM3
Connecting........_____....._____....._____....._____....._____....._____....._____
COM3 failed to connect: Failed to connect to Espressif device: Timed out waiting for packet header

A fatal error occurred: Could not connect to an Espressif device on any of the 1 available serial ports.

Trying different ports didnt help. 
Seems that esp8266 is not entering bootloader mode?

Thank you for help.

# Operating system

Windows 10

# Python version
3.6

# What Chip

ESP8266

# What development board or other hardware is the chip attached to

custom pcb

# Are you running esptool.py from an IDE such as Arduino or Eclipse?

No IDE, Windows command prompt

# Full esptool.py command line that was run:

esptool.py --port COM3 flash_id

# Full output from esptool.py

esptool.py v2.8
Found 1 serial ports
Serial port COM3
Connecting........_____....._____....._____....._____....._____....._____....._____
COM3 failed to connect: Failed to connect to Espressif device: Timed out waiting for packet header

A fatal error occurred: Could not connect to an Espressif device on any of the 1 available serial ports.

# Is there any other information you can think of which will help us reproduce this problem?

Old pc: Windows 10, HP, Intel i5
New pc: Windows 10, Lenovo, AMD Ryzen 5

radimkarnis, 2020-12-28:
Hello @adateht,

Thanks for reporting your problem.
There is a possibility that some AMD chipsets have unusual USB host timing, which is already being discussed in https://github.com/espressif/esptool/issues/516. If this is the case, there is nothing we can fix from the esptool or even the pyserial (esptool's serial interface) layer.
Could you please run `esptool.py --trace -p PORT flash_id`  and post the full output? 

Thank you

adateht, 2020-12-28:
Hello,

if this is the case, why flashing esp from arduino ide (I tried nodemcu and esp-01) works? It also uses esptool.py.

Here is cmd output:

esptool.py v2.8
Serial port COM3
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.025 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.121 Timed out waiting for packet header
.TRACE +0.066 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.112 Timed out waiting for packet header
.TRACE +0.071 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.120 Timed out waiting for packet header
.TRACE +0.065 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.008 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.104 Timed out waiting for packet header
.TRACE +0.070 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.122 Timed out waiting for packet header
.TRACE +1.861 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.008 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.110 Timed out waiting for packet header
_TRACE +0.063 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
[... 546 lines cut ...]
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.107 Timed out waiting for packet header
_TRACE +0.066 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.113 Timed out waiting for packet header
_TRACE +0.063 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.008 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.104 Timed out waiting for packet header
_

A fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header

dobairoland, 2020-12-28:
Hi @adateht, could you please try the following.

1. If you have a button connected to GPIO0 then please try to hold it while running esptool. Does it flashes in this case?
2. Do you have another board, e.g. ESP8266-DevKitC? Please try to flash it if you have one with the same computer. 
3. Can you share the schematics of your board? (the part with connections to EN and GPIO0 is enough)

adateht, 2020-12-28:
Hello @dobairoland 

thank you for reply.

1) My board does not have button for GPIO0, programmer pcb features autoreset circuit, here is snippet of circuit

[image]

On ESP board here is circuit snippet of ESP (RST_BTN signal is RST on programmer board):

[image]

As i said, on all other computers, this works perfectly.

2) I dont have such a board, but I do have this simple esp8266 programmer for esp-01:

[image]

With this (but only when manually connected GPIO0 to GND) it works, see output:

esptool.py v2.8
Found 1 serial ports
Serial port COM3
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 84:f3:eb:30:96:6d
Uploading stub...
Running stub...
Stub running...
Manufacturer: 85
Device: 6014
Detected flash size: 1MB
Hard resetting via RTS pin...

Also, communicating with boards (arduino, nodemcu, esp-01) from Arduino IDE, works good.

dobairoland, 2020-12-28:
@adateht Do you have a capacitor between RST and GND? I don't see it on your schematics. Please try to add one 0.1 uF temporarily.

adateht, 2020-12-28:
@dobairoland there is no cap in my schematic, however no luck with adding 0.1uF (tried also 1 uF)

dobairoland, 2020-12-29:
Thank you @adateht for patiently providing all these information. I'm sorry but currently I'm out of ideas. However, I have a Lenovo with Ryzen processor. I will try to reproduce this issue when I get hand on other boards. For now I've tried two ESP32 boards but have no ESP8266.

In the meantime, please continue to use your workarounds for flashing the chip. We are sorry for the inconvenience.

adateht, 2020-12-29:
@dobairoland thank you for your contribution

dobairoland, 2021-02-26:
@adateht I finally got an esp8266_devkitc_v1 and tested it with my Lenovo Ryzen laptop & Win 10 Pro. I'm sorry but I cannot reproduce this issue. I think it is entirely HW and Windows USB driver related. Having the ESP-01 programmer in the middle might help just because it adds some delay (or filters out some undesired transition).

This is unrelated, but just yesterday I was capturing the signals with a logic analyzer during establishing single serial connections to various ESP boards. I used Putty on the same laptop. The DTR and RTS signal transitions should occur at the same time on connection but there were a delay between them sometimes. The chip was restarting if there was a larger delay. If you have access to a logic analyzer then please try to analyze the communication in a similar way.

Custom boards tend to have problems working with one particular computer while not with another one. I can recommend to analyze publicly available schematics of Espressif boards. Those are tested with various OSs and computers and don't have these issues. Unfortunately, I'm not a hardware engineer so I cannot help more. Please try to ask on esp32.com or esp8266.com in the hardware section. I'm sorry for the inconvenience.

closed 2021-02-26 by dobairoland

dobairoland, 2021-02-26:
This is another similar issue with switching to a computer with Ryzen: https://github.com/espressif/esp-idf/issues/2849#issuecomment-766417975 (with different ESP chip). The solution was also to create a bridge between.