espressif/esptool #593: pico32d4 A fatal error occurred: Timed out waiting for packet content #4630

itsieber, 2020-12-28:
# Operating system
Windows 10

# Python version
3.9.1

# What Chip
ESP32-PICO-D4 (Revision 1)

# What development board or other hardware is the chip attached to
own pcb bord

# Is anything else attached to the development board, except for the serial flasher connections?
GPIO 21/22 I2C

# Are you running esptool.py from an IDE such as Arduino or Eclipse?
No IDE, Windows command prompt

# Full esptool.py command line that was run:
esptool.py -p COM4 erease_flash

# Full output from esptool.py
`esptool.py v3.1-beta1
Serial port COM4
Connecting........__
Detecting chip type... ESP32
Chip is ESP32-PICO-D4 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, Embedded Flash, VRef calibration in efuse, Coding Scheme None
Crystal is 40Mhz
MAC: 24:a1:60:57:17:70
Uploading stub...
Running stub...
Stub running...
Ereasing flash (this may take a while)...

A fatal error occurred: Timed out waiting for packet content`

# Do you have any other information from investigating this?
When i run the upload i get the error:
Warning: Could not auto-detect Flash size (FlashID=0xffffff, SizeID=0xff), defaulting to 4MB

When i run the flash_id i get:
Manufactor: ff
Device: ffff
Detected flash size Unknown

i had fixed the VDD_SDIO to 3.3.V -> no changes

my pinout:
1 +3.3
2 LNA
3 +3.3
4 +3.3
9 10k -> +3.3
19 +3.3
23 10k -> +3.3
37 +3.3
39 I2C
40 RXD
41 TXD
42 I2C
43 +3.3
46 +3.3
49 GND

dobairoland, 2020-12-28:
Hi @itsieber.

Could you please share also the output of the following command?
```
esptool.py --chip esp32 --port COM4 --trace flash_id
```

> 9 10k -> +3.3

Do you have also a capacitor between this PIN and GND as shown in Figure 4 of the [datasheet](https://www.espressif.com/sites/default/files/documentation/esp32-pico-d4_datasheet_en.pdf)?

TryingToGITthis, 2020-12-28:
I'm having the same issue. This is a custom board I thought I'd try to use for my first PCB design. My programmer
has the proper circuitry for the DST/RST pins and works with other boards. When trying to upload firmware from 
PlatformIO or Arduino, I get the following error:

> Checking size .pio\build\esp32dev\firmware.elf
> Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
> RAM:   [=         ]   5.1% (used 16580 bytes from 327680 bytes)
> Flash: [===       ]  30.0% (used 392914 bytes from 1310720 bytes)
> Configuring upload protocol...
> AVAILABLE: esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
> CURRENT: upload_protocol = esptool
> Looking for upload port...
> Use manually specified: COM50
> Uploading .pio\build\esp32dev\firmware.bin
> esptool.py v3.0
> Serial port COM50
> Connecting.......
> Chip is ESP32-PICO-D4 (revision 1)
> Features: WiFi, BT, Dual Core, 240MHz, Embedded Flash, VRef calibration in efuse, Coding Scheme None
> Crystal is 40MHz
> MAC: 24:a1:60:57:18:30
> Uploading stub...
> Running stub...
> Stub running...
> Changing baud rate to 1152000
> Changed.
> Configuring flash size...
> Warning: Could not auto-detect Flash size (FlashID=0xffffff, SizeID=0xff), defaulting to 4MB
> Compressed 15872 bytes to 10319...

A fatal error occurred: Timed out waiting for packet content
*** [upload] Error 2

Connecting to the serial port shows the following:

> rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> flash read err, 1000
> ets_main.c 371 

[Schematic_Tiny ESP32 Test_2020-12-28_14-15-38.pdf](https://github.com/espressif/esptool/files/5748614/Schematic_Tiny.ESP32.Test_2020-12-28_14-15-38.pdf)

Tims159, 2021-01-03:
I'm also facing  the same issue, but with esp8266MOD 

following is the error when i try to flash:
esptool.py v3.0
Serial port COM3
Connecting........_____....._
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 2c:f4:32:12:a9:ff
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Compressed 458752 bytes to 299358...

A fatal error occurred: Timed out waiting for packet header

following is the dump for esp8266 with trace command:

esptool.py --chip esp8266 --port COM3 --trace flash_id
esptool.py v3.0
Serial port COM3
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.001 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.106 Timed out waiting for packet header
.TRACE +0.064 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.002 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.010 Read 1 bytes: c0
TRACE +0.000 Read 35 bytes:
    0108020007071220 0000c0c001080200 | ....... ........
    070712200000c0c0 0108020007071220 | ... ...........
    0000c0                            | ...
TRACE +0.001 Received full packet: 01080200070712200000
TRACE +0.002 Received full packet: 01080200070712200000
TRACE +0.003 Received full packet: 01080200070712200000
TRACE +0.010 Read 1 bytes: c0
TRACE +0.000 Read 59 bytes:
    0108020007071220 0000c0c001080200 | ....... ........
    070712200000c0c0 0108020007071220 | ... ...........
    0000c0c001080200 070712200000c0c0 | ........... ....
    0108020007071220 0000c0           | ....... ...
TRACE +0.001 Received full packet: 01080200070712200000
TRACE +0.002 Received full packet: 01080200070712200000
TRACE +0.002 Received full packet: 01080200070712200000
TRACE +0.001 Received full packet: 01080200070712200000
TRACE +0.001 Received full packet: 01080200070712200000

TRACE +0.002 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=00100040
TRACE +0.000 Write 14 bytes: c0000a04000000000000100040c0
TRACE +0.006 Read 1 bytes: c0
TRACE +0.000 Read 11 bytes: 010a020001c1f0ff0000c0
TRACE +0.000 Received full packet: 010a020001c1f0ff0000
TRACE +0.003 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=5c00f03f
TRACE +0.001 Write 14 bytes: c0000a0400000000005c00f03fc0
TRACE +0.012 Read 1 bytes: c0
TRACE +0.000 Read 11 bytes: 010a020032f42c000000c0
[... 1211 lines cut ...]
    0000040000000400 0000040000000500 | ................
    0000050000000500 0000050000000000 | ................
    0000000000000000 0000101112000807 | ................
    09060a050b040c03 0d020e010f000101 | ................
    0000010000000400 0000c0           | ...........
TRACE +0.091 Read 1 bytes: c0
TRACE +0.000 Read 11 bytes: 0107020032f42c000000c0
TRACE +0.001 Received full packet: 0107020032f42c000000
Running stub...
TRACE +0.004 command op=0x06 data len=8 wait_response=1 timeout=0.050 data=0000000004e01040
TRACE +0.000 Write 18 bytes:
    c000060800000000 000000000004e010 | ................
    40c0                              | @.
TRACE +0.011 Read 1 bytes: c0
TRACE +0.000 Read 17 bytes:
    0106020032f42c00 0000c0c04f484149 | ....2.,.....OHAI
    c0                                | .
TRACE +0.001 Received full packet: 0106020032f42c000000
TRACE +0.001 Received full packet: 4f484149
Stub running...
TRACE +0.000 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=1c020060
TRACE +0.000 Write 14 bytes: c0000a0400000000001c020060c0
TRACE +3.009 Timed out waiting for packet header

A fatal error occurred: Timed out waiting for packet header

dobairoland, 2021-02-26:
I've finally got an ESP32-PICO-D4 (revision 1) [ESP Pico Kit V4] and tested it with the latest master branch & Windows 10. I'm sorry but I cannot reproduce the issue. Please refer to the documentation about recommendation how to use this chip on custom boards. You can be inspired also by public schematics of Espressif boards. If you still have questions then please ask on esp32.com in the hardware section.

closed 2021-02-26 by dobairoland